Koenkk/zigbee2mqtt #2191: Zigbee2mqtt crash

neographikal, 2019-10-23:
# Bug Report
Zigbee2mqtt crashed, see attached logs.
[zigbee2mqtt2-debugoutput.log](https://github.com/Koenkk/zigbee2mqtt/files/3760936/zigbee2mqtt2-debugoutput.log)
[zigbee2mqtt2.txt](https://github.com/Koenkk/zigbee2mqtt/files/3760937/zigbee2mqtt2.txt)

Also, the docker container does not restart automatically after the crash despite the --restart always:

docker run -d --name zigbee2mqtt --net=host -v /opt/zigbee2mqtt:/app/data --device=/dev/ttyACM0 -e TZ=Europe/Amsterdam --restart=always --log-driver syslog --log-opt syslog-facility=local6 --log-opt tag="{{.Name}}" koenkk/zigbee2mqtt:latest-dev

## What did you expect to happen
Not crash

## How to reproduce it (minimal and precise)
Can't, it just happens every now and then.

## Debug Info
zigbee2mqtt version: 1.6.0-dev-latest
CC253X firmware version: 20190619

neographikal, 2019-10-23:
Disabled zigbee2mqtt assistant, but it still puzzles me that the docker container didn't restart after the crash of zigbee2mqtt. Shouldn't these network scans be throttled if they cause crashes when done too often?

Koenkk, 2019-10-24:
We can add more delays but first we should check if this is the issue.

neographikal, 2019-10-26:
Crashed again, no additional software running. Logs attached.
[zigbee2mqtt2.txt](https://github.com/Koenkk/zigbee2mqtt/files/3775010/zigbee2mqtt2.txt)

Docker container was again totally down, didn't restart despite --restart=always

Koenkk, 2019-10-26:
It stops because the adapter disconnects (`#033[31mzigbee2mqtt:error#033[39m 2019-10-26T15:47:41: Adapter disconnected, stopping`)

neographikal, 2019-10-27:
kern.log sees the USB dropping as well:

```
Oct 27 07:46:31 neo-server kernel: [1323584.282277] usb 2-1.1: USB disconnect, device number 41
Oct 27 07:46:31 neo-server kernel: [1323584.283794] cdc_acm 2-1.1:1.0: failed to set dtr/rts
Oct 27 07:46:31 neo-server kernel: [1323584.523039] usb 2-1.1: new full-speed USB device number 42 using ehci-pci
Oct 27 07:46:31 neo-server kernel: [1323584.641776] usb 2-1.1: New USB device found, idVendor=0451, idProduct=16a8
Oct 27 07:46:31 neo-server kernel: [1323584.641781] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 27 07:46:31 neo-server kernel: [1323584.641783] usb 2-1.1: Product: TI CC2531 USB CDC
Oct 27 07:46:31 neo-server kernel: [1323584.641784] usb 2-1.1: Manufacturer: Texas Instruments
Oct 27 07:46:31 neo-server kernel: [1323584.641786] usb 2-1.1: SerialNumber: __0X00124B000FE8E700
Oct 27 07:46:31 neo-server kernel: [1323584.642792] cdc_acm 2-1.1:1.0: ttyACM0: USB ACM device
Oct 27 07:51:10 neo-server kernel: [1323863.320618] usb 2-1.1: USB disconnect, device number 42
Oct 27 07:51:10 neo-server kernel: [1323863.322141] cdc_acm 2-1.1:1.0: failed to set dtr/rts
Oct 27 07:51:10 neo-server kernel: [1323863.562064] usb 2-1.1: new full-speed USB device number 43 using ehci-pci
Oct 27 07:51:10 neo-server kernel: [1323863.680614] usb 2-1.1: New USB device found, idVendor=0451, idProduct=16a8
Oct 27 07:51:10 neo-server kernel: [1323863.680618] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 27 07:51:10 neo-server kernel: [1323863.680620] usb 2-1.1: Product: TI CC2531 USB CDC
Oct 27 07:51:10 neo-server kernel: [1323863.680621] usb 2-1.1: Manufacturer: Texas Instruments
Oct 27 07:51:10 neo-server kernel: [1323863.680623] usb 2-1.1: SerialNumber: __0X00124B000FE8E700
Oct 27 07:51:10 neo-server kernel: [1323863.681465] cdc_acm 2-1.1:1.0: ttyACM0: USB ACM device

```

That's strange right? I have another, I can try to exchange it, but which firmware would you suggest to use?

But the container not being restarted by docker

Koenkk, 2019-10-27:
20190619 Z-stack 1.2 source routing

neographikal, 2019-10-27:
Thank you will try

closed 2020-01-02 by stale[bot]