Koenkk/zigbee2mqtt #2192: Failed network lqi scan for device

Bastiencc, 2019-10-23:
Hi

Since few days I get this in log, what does that mean? Do I need to do something ?

```
**Oct 23 11:04:12 raspberrypi npm[498]:   zigbee2mqtt:warn 10/23/2019, 11:04:12 AM Network scan timeout, skipping outstanding lqi scans for '0x00158d0002d6f569 0x00158d0002d39c01 0x00158d0002d6f577'**
Oct 23 11:04:12 raspberrypi npm[498]:   zigbee2mqtt:warn 10/23/2019, 11:04:12 AM Network scan timeout, skipping outstanding rtg scans for ''
Oct 23 11:04:12 raspberrypi npm[498]:   zigbee2mqtt:info 10/23/2019, 11:04:12 AM MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/raw', payload '{"nodes":[{"ieeeAddr":"0x00124b001936860e","friendlyName":"0x00124b001936860e","type":"Coordinator","nwkAddr":0,"status":"online","scanfailed":[]},{"ieeeAddr":"0x0017880104af512e","friendlyName":"cuisine_switch_mur","type":"EndDevice","nwkAddr":59634,"manufName":"Philips","modelId":"RWL021","status":"offline","scanfailed":[]},{"ieeeAddr":"0x001788010408cd64","friendlyName":"salon_switch_mur","type":"EndDevice","nwkAddr":5698,"manufName":"Philips","modelId":"RWL021","status":"offline","scanfailed":[]},{"ieeeAddr":"0x00158d0002d6f569","friendlyName":"cuisine_plafond","type":"Router","nwkAddr":64741,"manufName":"LUMI","modelId":"lumi.light.aqcn02","status":"online","scanfailed":["lqi"]},{"ieeeAddr":"0x00158d0002d39c01","friendlyName":"chambre_1_plafond","type":"Router","nwkAddr":44426,"manufName":"LUMI","modelId":"lumi.light.aqcn02","status":"online","scanfailed":["lqi"]},{"ieeeAddr":"0x0017880104af5115","friendlyName":"chambre_1_switch_mur","type":"EndDevice","nwkAddr":46210,"manufName":"Philips","modelId":"RWL021","status":"offline","scanfailed":[]},{"ieeeAddr":"0x001788010408bb63","friendlyName":"chambre_1_switch_chevet","type":"EndDevice","nwkAddr":51938,"manufName":"Philips","modelId":"RWL021","status":"offline","scanfailed":[]},{"ieeeAddr":"0x00158d00030403b7","friendlyName":"cigare_capteur","type":"EndDevice","nwkAddr":9988,"manufName":"LUMI","modelId":"lumi.weather","status":"online","scanfailed":[]},{"ieeeAddr":"0x00158d000275de1d","friendlyName":"aqara_cube","type":"EndDevice","nwkAddr":31747,"manufName":"LUMI","modelId":"lumi.sensor_cube","status":"online","scanfailed":[]},{"ieeeAddr":"0x0017880104af50f3","friendlyName":"chambre_1_switch_chevet_2","type":"EndDevice","nwkAddr":496,"manufName":"Philips","modelId":"RWL021","status":"offline","scanfailed":[]},{"ieeeAddr":"0x00158d0003664f59","friendlyName":"wc_capteur","type":"EndDevice","nwkAddr":53956,"manufName":"LUMI","modelId":"lumi.sens
Oct 23 11:04:12 raspberrypi npm[498]: or_motion.aq2","status":"online","scanfailed":[]},{"ieeeAddr":"0x001788010408cda6","friendlyName":"salon_switch_canape","type":"EndDevice","nwkAddr":26201,"manufName":"Philips","modelId":"RWL021","status":"offline","scanfailed":[]},{"ieeeAddr":"0x00158d0002d6f577","friendlyName":"wc_plafond","type":"Router","nwkAddr":48812,"manufName":"LUMI","modelId":"lumi.light.aqcn02","status":"online","scanfailed":["lqi"]}],"links":[{"sourceIeeeAddr":"0x000d6ffffe1c1c67","targetIeeeAddr":"0x00124b001936860e","sourceNwkAddr":29516,"lqi":170,"depth":1,"relationship":1,"routes":[]},{"sourceIeeeAddr":"0x000b57fffebf0e33","targetIeeeAddr":"0x00124b001936860e","sourceNwkAddr":16769,"lqi":170,"depth":1,"relationship":1,"routes":[]},{"sourceIeeeAddr":"0x001788010408cda6","targetIeeeAddr":"0x00124b001936860e","sourceNwkAddr":26201,"lqi":170,"depth":1,"relationship":1,"routes":[]},{"sourceIeeeAddr":"0x00158d00030403b7","targetIeeeAddr":"0x00124b001936860e","sourceNwkAddr":9988,"lqi":107,"depth":1,"relationship":1,"routes":[]},{"sourceIeeeAddr":"0x00158d0002d39c01","targetIeeeAddr":"0x00124b001936860e","sourceNwkAddr":44426,"lqi":39,"depth":255,"relationship":3,"routes":[]},{"sourceIeeeAddr":"0x00158d0002d6f577","targetIeeeAddr":"0x00124b001936860e","sourceNwkAddr":48812,"lqi":12,"depth":255,"relationship":3,"routes":[]},{"sourceIeeeAddr":"0x00158d0002d6f569","targetIeeeAddr":"0x00124b001936860e","sourceNwkAddr":64741,"lqi":12,"depth":255,"relationship":3,"routes":[]}]}'
Oct 23 11:04:12 raspberrypi npm[498]:   zigbee2mqtt:error 10/23/2019, 11:04:12 AM Failed to ping 'cuisine_plafond'
**Oct 23 11:04:18 raspberrypi npm[498]:   zigbee2mqtt:warn 10/23/2019, 11:04:18 AM Failed network lqi scan for device: '0x00158d0002d39c01' with error: 'Error: Timed out after 10000 ms'
Oct 23 11:04:18 raspberrypi npm[498]:   zigbee2mqtt:warn 10/23/2019, 11:04:18 AM Failed network lqi scan for device: '0x00158d0002d6f577' with error: 'Error: Timed out after 10000 ms'
Oct 23 11:04:22 raspberrypi npm[498]:   zigbee2mqtt:warn 10/23/2019, 11:04:22 AM Failed network lqi scan for device: '0x00158d0002d6f569' with error: 'Error: Timed out after 10000 ms'**
```

## Debug Info
zigbee2mqtt version: 1.6.0  (commit #e26ad2a)
CC253X firmware version: 20190608

Koenkk, 2019-10-23:
Can you check if you still get it with latest dev?

Bastiencc, 2019-10-25:
I'll try and get back to you. Thanks

Bastiencc, 2019-10-26:
Hi @Koenkk 

I switched to dev version :
```Oct 26 11:07:17 raspberrypi npm[2772]: zigbee2mqtt:info  2019-10-26T10:07:17: Starting zigbee2mqtt version 1.6.0 (commit #6f92a41)
Oct 26 11:07:17 raspberrypi npm[2772]: zigbee2mqtt:info  2019-10-26T10:07:17: Starting `zigbee-herdsman...```

and I still get : 
.```Oct 26 11:08:15 raspberrypi npm[2837]: zigbee2mqtt:error 2019-10-26T10:08:15: Failed to execute LQI for 'cuisine_plafond'
Oct 26 11:08:25 raspberrypi npm[2837]: zigbee2mqtt:error 2019-10-26T10:08:25: Failed to execute LQI for 'chambre_1_plafond'
Oct 26 11:08:35 raspberrypi npm[2837]: zigbee2mqtt:error 2019-10-26T10:08:35: Failed to execute LQI for 'wc_plafond'.```

Thoses 3 devices are ZNLDP12LM - Xiaomi Aqara smart LED bulb (Router)

thanks for helping

Koenkk, 2019-10-26:
Are these all your `ZNLDP12LM`? If so, and for other devices it succeeds, then probably this device doesn't support a LQI scan. I don't have this device so I cannot confirm this.

Bastiencc, 2019-10-26:
Yes I only have 3 xiaomi bulb ZNLDP12LM
is this scan new in 1.6.0 ? I never saw this error in previous versions

Bastiencc, 2019-10-26:
I do not know if it is related to that, but ma zigbee map seems strange... 
https://imgur.com/bMoZt3e

Koenkk, 2019-10-26:
Before 1.6.0 this was not logged I think

closed 2019-10-27 by Bastiencc