Koenkk/zigbee2mqtt #2194: Cannot pair ﻿﻿﻿ZNCZ03LM Xiaomi Mi power plug ZigBee TW

tanjipon, 2019-10-23:
# Bug Report
## What happened
I cannot pair  ﻿﻿﻿ZNCZ03LM Xiaomi Mi power plug ZigBee TW . Just nothing happened.
## What did you expect to happen
Successfully pair the Xiaomi power plug  with Zigbee2Mqtt.
## How to reproduce it (minimal and precise)
1. Installing Zigbee2Mqtt on MacOS.
2. Flash CC2531 with 20190608 firmware.
3. Already tried to reset CC2531 and plug different USB port.
## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: 20190608

**Below shows the debug info:** 

chi@Chis-MacBook-Pro:/opt/zigbee2mqtt$ npm start

zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
node index.js

  zigbee2mqtt:info 10/24/2019, 2:41:17 AM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-10-24.02-41-17'
  zigbee2mqtt:info 10/24/2019, 2:41:17 AM Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)
  zigbee2mqtt:info 10/24/2019, 2:41:17 AM Starting zigbee-shepherd
  zigbee2mqtt:info 10/24/2019, 2:41:18 AM zigbee-shepherd started
  zigbee2mqtt:info 10/24/2019, 2:41:18 AM Coordinator firmware version: '20190608'
  zigbee2mqtt:info 10/24/2019, 2:41:18 AM Currently 0 devices are joined:
  zigbee2mqtt:warn 10/24/2019, 2:41:18 AM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 10/24/2019, 2:41:18 AM Allowing new devices to join.
  zigbee2mqtt:warn 10/24/2019, 2:41:18 AM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 10/24/2019, 2:41:18 AM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 10/24/2019, 2:41:18 AM Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 10/24/2019, 2:41:18 AM zigbee-shepherd ready
  zigbee2mqtt:info 10/24/2019, 2:41:18 AM Connected to MQTT server
  zigbee2mqtt:info 10/24/2019, 2:41:18 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 10/24/2019, 2:41:18 AM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"e26ad2a","coordinator":20190608,"log_level":"info","permit_join":true}'

vhristev, 2019-10-24:
It could be something wrong with USB ports > CC2531 dongle i suffer from the same issue with Pi4 just cannot pair but with pi 3 is working like a charm.

Koenkk, 2019-10-24:
I would recommend a usb extension cable. When plugged in directly interference can happen.

vhristev, 2019-10-24:
Yep i read that in the documentation and yesterday I was searching one in the holly box with cables and definitely i will try with extension.  Any recommendations of USB 2 or USB 3 ports to use on Pi 4 ? I tried all of USB 2 and 3 ports but i end up with same behaviour.

vhristev, 2019-11-02:
**Solved** with USB extender and moving away CC2531 from Pi4

Mother of god.... I put extender and move away from the CC2531 USB and now everything is working like a charm. Even I able to connect my router at first it didnt show up but after 3 power off / on ( re-pair ) it shows up.

` zigbee2mqtt:info 11/3/2019, 12:22:04 AM New device 'lumi.router' with address 0x00124b00199ddad2 connected!
`

I connect 3 devices to my CC2531 USB and they pair with a single try its incredible how this extender fixes all my issues. It seems PI4 interference of radio signals and electrical components is very strong ... stronger than Pi3. 

Thanks for the help.

closed 2019-11-03 by Koenkk