Koenkk/zigbee2mqtt #2195: Hue motion sensor. Working but with lots of timeouts in the logs.

pavon87, 2019-10-23:
Hi,

**Situation:**
I have a network of 20 devices, basically Xiaomi devices, ikea and hue lights, and two routers, CC2530 and CC2591. As a coordinator I use an official docker container (updated) in a raspberry pi with CC2530 with last firmware. Also I reproduced the same problem with another raspberry, with another container and a CC2531 antenna.

**Problem**:

 I’m trying some Hue Motion sensors, (Philips 9290012607), it sems that works properlly but I’m receiving a timeout error in the logs and some hangs after some hours working.

In summary, in the logs there are several missages like the next, with "attempt" #1, #2, #3:
	Th  zigbee2mqtt:warn 10/23/2019, 10:20:22 PM Failed to configure hue_sensor (0x0017880104b7a74b) ('Error: SRSP - ZDO - bindReq after 6000ms') (attempt #1)

And lots of other messages like:
	  zigbee2mqtt:error 10/23/2019, 9:58:35 PM Cannot get the Node Descriptor of the Device: 0x0017880104b7a74b (Error: Timed out after 10000 ms)

I have the same issue with two different Hue Motion sensors. Also I tried changing the QoS level without .

Thanks in advance

**Off_topic comment**

This is not related with the issue and is not a question, but maybe somebody is in the same situation and found better solution. Trying to obtain a High Availability system of Zigbee2mqtt, I created a “network” where there are two coordinators, one master and one slave (stopped), in two different raspberry’s, each with their antenna and each one running zigbee2mqtt container, where one is up and the other is stopped. With rsync I keep updated the devices list, and with a custom watchdog script, I control witch container is working, and I start one or the other when is needed. It is the only way I have found to make a system with some High Availability. Somebody knows a better option to make redundancy?

Koenkk, 2019-10-24:
Can you try with the latest dev branch? I expect that it fixes these issues.

pavon87, 2019-10-24:
Hi,
I just update the container to lastest-dev version, and I getting a new error (periodically every two minutes), only related with the Hue Motion:

zigbee2mqtt:info  2019-10-24T08:00:59: MQTT publish: topic 'pavon/z2m/hue_sensor', payload '{"battery":100,"linkquality":107,"occupancy":false,"illuminance":13048,"temperature":19.62}'
zigbee2mqtt:info  2019-10-24T08:00:59: Configuring 'hue_sensor'
zigbee2mqtt:info  2019-10-24T08:01:08: MQTT publish: topic 'pavon/z2m/hue_sensor', payload '{"battery":100,"linkquality":110,"occupancy":false,"illuminance":13048,"temperature":17.79}'
zigbee2mqtt:error 2019-10-24T08:01:09: Failed to configure 'hue_sensor', attempt 2 (Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))

Thanks for your support!

Koenkk, 2019-10-24:
When starting zigbee2mqtt, you need to wake up the sensor (by e.g. triggering movements or pressing to button on it) until you see `Successfully configured 'hue_sensor'`. 

If that doesn't help, try to re-pair the device.

closed 2019-12-30 by stale[bot]