Koenkk/zigbee2mqtt #2198: Error: Data request failed with error: 'No network route' (205)

VOVAiS, 2019-10-24:
# Bug Report
## What happened
I have a small ZigBee network of 4 routers and 2 end devices. But I keep getting delay in devices. QBKG12LM give errors. How can I fix this.

logs
> error 2019-10-24T14:00:46: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'No network route' (205)'
error 2019-10-24T14:06:08: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'No network route' (205)'
error 2019-10-24T14:31:11: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'No network route' (205)'
error 2019-10-24T14:40:55: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Timeout - 16111 - 1 - 52 - 11 after 15000ms'
error 2019-10-24T14:55:40: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'No network route' (205)'
error 2019-10-24T14:55:41: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'No network route' (205)'
error 2019-10-24T14:55:43: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'No network route' (205)'
error 2019-10-24T14:55:44: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'No network route' (205)'
error 2019-10-24T15:01:09: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'No network route' (205)'
error 2019-10-24T15:06:21: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'No network route' (205)'
error 2019-10-24T15:06:39: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Timeout - 16111 - 1 - 230 - 11 after 15000ms'

Maps
https://pastebin.com/638V427z
when errors occur the map changes

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.6.0 commit ec5064b 
CC253X firmware version: 20190425

VOVAiS, 2019-10-25:
log debug
https://pastebin.com/nWft7eAi

Koenkk, 2019-10-25:
Can you try with the following firmware https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

VOVAiS, 2019-10-25:
Updated the firmware to CC2531_SOURCE_ROUTING_20190619. Repaired all devices. In the logs of the error

error 2019-10-25T14:27:01: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'
error 2019-10-25T15:03:50: Publish 'set' 'state' to '0x00158d0002d7a022' failed: 'Error: Timeout - 13777 - 1 - 218 - 11 after 15000ms'

closed 2019-10-31 by VOVAiS

wojciechczyz, 2019-11-29:
I seem to have the same problem to this one:
https://github.com/Koenkk/zigbee2mqtt/issues/2437
Did you fixed it by changing the firmware?

VOVAiS, 2019-11-29:
updated firmware with routing. Reboot your device (turned off turned on the power). The network is stable

wojciechczyz, 2019-11-29:
Thank you for help. I do. Have now the latest 3.0 firmware from september. Which one you moved to? Any idea why routing devices fail?

VOVAiS, 2019-11-30:
firmware use https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

wojciechczyz, 2019-11-30:
Thank you! I luckily had another stick so flashed it with 1.2 source routing, replaced stick, rebooted and ... still similar problem.
non routing sensors work, all lights and plugs (routing) stuck and non working.
Strangely did not had to re-pair sensors.
and worse it does not pair with anything new - possibly due to limit of 5 devices.

wojciechczyz, 2019-11-30:
So I am still missing something. Possibly I need to remove all devices and start pairing again? What is the best way to remove them cleanly?
Thank you

VOVAiS, 2019-11-30:
you must first remove all devices. then add the routers and at the last add the end devices

wojciechczyz, 2019-12-01:
Thank you, I did it and it worked well!!!! I also needed to extend USB cable that was used for the stick. Have a good night and a great week!

NikolaZornic, 2019-12-26:
How did you remove all devices?

VOVAiS, 2019-12-26:
delete database.db,

NikolaZornic, 2019-12-26:
Can't re-pair devices. Log:

`zigbee2mqtt:info  2019-12-26 22:30:38: Starting interview of '0x00124b001bd841eb'
zigbee2mqtt:info  2019-12-26 22:30:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001bd841eb"}}'
zigbee2mqtt:error 2019-12-26 22:31:58: Failed to interview '0x00124b001bd841eb', device has not successfully been paired
zigbee2mqtt:info  2019-12-26 22:31:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00124b001bd841eb"}}'`

Zigbee2mqtt Version:  1.8.0 
Coordinator Version:  {'type': 'zStack12', 'meta': {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190619}}

NikolaZornic, 2019-12-27:
Also tried complete reinstall, still getting "Failed to interview..." error.

Koenkk, 2019-12-27:
@NikolaZornic with what device?

NikolaZornic, 2019-12-27:
@Koenkk just got it working :)

This is the sniffer I use https://www.aliexpress.com/item/33011691700.html (CC2531 with Antenna).
With `regular` and `source_routing` firmware only motion sensors (Xiaomi Aqara) worked (without re-pairing), bulbs (IKEA and Gledopto) and switches (IKEA and Aqara Opple) didn't work and wouldn't re-pair.

What worked is https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin - just had to re-pair all devices.

hahmann, 2020-04-26:
I was having the same issue with a Yale lock, can confirm that installing Z-Stack 3.0.x on the coordinator (CC2531) solved the issue. Repaired all the devices and they are working properly now for half a day already. I currently have 10 devices connected directly to the coordinator (1 router CC2530) and other 2 to the router.

elbueno222, 2020-05-26:
@hahmann I am having the same issue now with a Danalock V3 at random times (`Error: Data request failed with error: 'No network route' (205)`) and I am thinking to change the firmware of coordinator (CC2531) to Z-Stack 3.0.x as you did, have you experience any other issue from the moment you changed the firmware?

hahmann, 2020-05-26:
@elbueno222 I just got an issue after more than a month of the network being very stable. One thing that I noticed, last time and this time again is that I start to get errors when the device connect simultaneously to the coordinator and the router I have as you can see below, ffac is my Yale Lock.

I posted this yesterday on another thread:
[https://github.com/Koenkk/zigbee2mqtt/issues/1445]

[image]

elbueno222, 2020-06-02:
@hahmann I updated the firmware to 1.2 source route as @Koenkk mentioned as it looks like the 'No network route' (205)' error is gone, but during testing I had the same issue that you had, one of the times my device (Ikea blind) connected to both coordinator and router at the same time, and every time I tried sending a message to the device I received the error 240 MAC transaction expired. I just fixed the issue by re-pairing the device again and again until it connected just to the coordinator, but I guess the error will happen again if the device connects to both coordinator and router again at some point.

jackcui128, 2020-09-17:
@Koenkk  2 questions about your regular and source-routine firmware cc2531_20190619. 
 1.  does source routine firmware only support cc2531 as zigbee router ? no Coordinator?
 2. what is the main difference between these two firmware? 
thanks a lot in advance !

jackcui128, 2020-09-17:
@Koenkk  by the way , i use MCU ESP32 to with CC loader running in the MCU , flashing CC2531_DEFAULT_20190608.zip is ok . However ,when flashing CC2531_SOURCE_ROUTING_20190619.zip , it is not fully ok, serial port can not be recognized , during the flashing , the number only stops at 500, while flashing default version , the number stops at 512.  do i have to use cc debugger to flash the source routine firmware ?

elbueno222, 2020-09-17:
@jackcui128 the source routing firmware is a coordinator firmware, not a router firmware (I know the name is a bit confusing, I had the same misunderstanding at the beginning). 
You can find the details here: 
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator
http://cms.digi.com/resources/documentation/digidocs/90001537/references/r_large_zigbee_networks-source_routing.htm?TocPath=Working%20with%20Zigbee%7C_____14

jackcui128, 2020-09-21:
@elbueno222  thanks for your comments,  so below link is the CC2531 router firmware , right?  if i want to expand the network devices, such as one cc2531 as a coordinator and one cc2531 as a router, have you ever done this before?
https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2531/bin

elbueno222, 2020-09-21:
@jackcui128 that was my setup (CC2531 coordinator + two CC2531 routers) until last month that I upgraded my coordinator to a CC2652R, and it was working perfectly fine with 24 devices. Just be aware the source_routing coordinator firmware for CC2531 reduce the amount of devices that can be connected directly to the coordinator to just 5, so you will need to add more routers if you want to increase the amount of end devices

jackcui128, 2020-09-22:
@elbueno222  thanks for your information, so you have already used cc2652R as Z-Stack_3.x.0 and two CC2531 routers as Z-Stack_Home_1.2 (default) ? so cool.  I am just trying to use this setup (1 cc2531 as a coordinator and 1 cc2531 as a router), i have two simple questions to clarify ? the first one is the router cc2531 don't need to connect to a raspberry pi usb port which runs zigbee2mqtt , right ? it only needs to power up the router cc2531 , right ?  the second one is how to assign a enddevice to pair to the cc2531 router , not to the cc2531 coordinator ? can we set this , or just leaving the enddevice close to the router cc2531? or the paring result is random?   i am confused. tks for your guide !!

elbueno222, 2020-09-22:
@jackcui128 yes this zigbee world is really amazing when you start digging. To answer your questions, you are basically right, the routers just need to be connected to power so any usb charger will work (coordinator is the only one that needs to be connected to the raspberry pi). Regarding the connection of the device, a device will always try to connect to the closest router/coordinator so you don´t have to worry about it, just make sure you connect the router to the network first, and then when you start the pairing process in a device, the device will connect to the best available option (router or coordinator, unless any of them has reached its maximum capacity)

bfis108137, 2021-04-03:
> Thank you, I did it and it worked well!!!! I also needed to extend USB cable that was used for the stick. Have a good night and a great week!

I have to say that this worked for me too.  I was getting ready to flash my coordinator which I have no idea how to do.  A lifesaver.  Seriously.  For me it happened after I moved the computer to a new location.  It was sitting on the printer and when I put it directly on the table it started apparently.  I say apparently because I have no idea for sure when it started.