Koenkk/zigbee2mqtt #2199: Philips hue ambiant white and Color not recognized

seaside1, 2019-10-24:
Hi!

First off, awesome project, been using it for a while and this is really good stuff!

So I have a Philips Hue white and color ambiance bulb which seems to not be supported. I could go through and add it, but to me it seems weird it is not recognized, since it's commonly used by the masses, which leads me to think something else is wrong.

I'm using the CC2531 and latest dev-branch.

`10/24/2019, 7:48:04 PM - info: Zigbee: allowing new devices to join.
10/24/2019, 7:48:05 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"1189d16","coordin
ator":20190223,"log_level":"info","permit_join":true}'
10/24/2019, 7:48:37 PM - info: New device 'LCA001' with address 0x00178801061f6977 connected!
10/24/2019, 7:48:37 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x001788010
61f6977","meta":{"modelID":"LCA001"}}'
10/24/2019, 7:48:37 PM - warn: Device with modelID 'LCA001' is not supported.
`

sjorge, 2019-10-24:
It's probably similar to this one: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L1172 but a different socket, E27 perhaps?

I think adding LCA001 in that device array should work.

seaside1, 2019-10-25:
It's en E27, I'll try adding it, thanks!

filikun, 2019-10-26:
Yea I also have new Philips bulbs that are not supported. These are the new white ambiance with bluetooth added. I'm not comfortable editing the files, I did try but I don't really get it. My bulbs are LTA001.

seaside1, 2019-10-30:
I've created a pull request for this change as well:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/725

filikun, 2019-10-30:
Could you ad mine as well?

Koenkk, 2019-10-30:
@filikun can you provide a link to the product? (e.g. amazon or Philips site) Then I can add it.

filikun, 2019-10-31:
I bought them from NetOnNet in Sweden. But this is what I was trying to add to the config, I think my problems is that it is not persistent on my Hassio setup. 

https://hastebin.com/vilimezida.css

Koenkk, 2019-11-01:
@filikun thanks, added it, will be in dev branch in a few hours.