Koenkk/zigbee2mqtt #2200: Cannot add device to a group, when the friendly_name contains a '/'

sjorge, 2019-10-24:
# Bug Report
## What happened

While testing some things for #2122 I noticed when using mqtt to add a device to a group that has a '/' in the friendly_name would be ignored by zigbee2mqtt...

When adding a device to a group that has a '/' in the name, i noticed this does not get picked up at all by zigbee2mqtt.

I was using `{"topic":"zigbee2mqtt/bridge/group/bedroom/group_night_light_remote/add","payload":"bedroom/desk_lamp","_msgid":"a75513fb.bcfdc","_event":"node:bfcef621.a616f8"}
` as payload

## What did you expect to happen
The device to be added to the group.

## How to reproduce it (minimal and precise)
Have a group named 'test/group' or similar and try and add a device to it using mqtt, it will fail.

## additional information
Presumable the mqtt subscription here: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/groups.js#L31 does not match when a '/' is present in the group name.

## Debug Info
zigbee2mqtt version: `1.6.0 (commit #9b67ed5)`
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}`

Koenkk, 2019-10-25:
Should be fixed in latest dev.

sjorge, 2019-10-25:
Looks good.

closed 2019-10-25 by sjorge