Koenkk/zigbee2mqtt #2201: Allow to store MQTT credentials separate from configuration.yaml

LukasK13, 2019-10-24:
In order to use version control for the configuration file, the MQTT credentials shouldn't be store within that file. It would be great to be able to store the MQTT credentials in a separate YAML-file or in case of docker, include the credentials as environment variable.

nikolaykasyanov, 2019-10-25:
I second this! Something like Home Assistant does with `secrets.yaml`.

Koenkk, 2019-10-25:
Good idea, what should be the syntax for this?

LukasK13, 2019-10-25:
I think we could adapt the syntax from home-assistant [here](https://www.home-assistant.io/docs/configuration/secrets/) as @nikolaykasyanov mentioned, so using `password: !secret mqtt_password`. This could be detected using simple regex.
Alternatively, the syntax for splitting up devices and groups could be used, but I prefer the first solution.

Koenkk, 2019-10-25:
Implemented, documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/configuration/configuration.md#specifying-devices-and-groups-in-a-separate-file

closed 2019-10-25 by Koenkk

airdrummingfool, 2019-10-25:
@Koenkk can this be used for other configuration parameters, e.g. `network_key`? I've been wondering about a way to separate that out for a bit, and this seems like it would be a great way to do it!

LukasK13, 2019-10-26:
Wow, this was incredibly fast.
Thank you for your effort and your amazing work!

Koenkk, 2019-10-26:
It's available in latest-dev (which replaced the arm32v6 image as it is multi arch now).