Koenkk/zigbee2mqtt #2203: More actions for Xiaomi Mi/Aqara smart home cube?

segalion, 2019-10-25:
I have read from other forums more advanced info about xiaomi cube options, specifically related to face detection:

https://community.smartthings.com/t/release-xiaomi-magic-cube-controller-advanced-dth/85481 

It is supposed to send the number face for slide, tap, flip90 (face_from and face_to), and flip180.

With those info, we could make new more advanced trigger options with Home Assistant to multiply the cube actions:
- i.e. we could mark the faces (i.e. like a dice) and create 6 scenes for slide, 6 more for tap, and 6 more scenes for flip180
- i.e. mark the 4 items in every edge face ('+','-','>','<', or 'red','green','yellow','blue'), and have new 4 possibities for flip90
- i.e. inference last face and have 6 analog inputs with rotate (right and left) over 6 different faces.

Actually I have one integrated with Xiaomi Gateway, controlling a sonos to playing a mpd playlists over icecast, and it runs really good. I have discover all movements in xiaomi integration (even 'iam' and 'free-fall') except 'swing' movement, that has been reported some time but I dont know how to get it.

segalion, 2019-10-28:
Hello.
I have found some code information in the hubitat xioami driver. Probably you known this code:
https://github.com/veeceeoh/xiaomi-hubitat

Here are code where you can see faces control of xiaomi cube:

https://github.com/veeceeoh/xiaomi-hubitat/blob/master/devicedrivers/xiaomi-cube-controller-hubitat.src/xiaomi-aqara-wireless-switch.groovy

segalion, 2019-10-28:
Sorry.
I have seen your code here
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js
Seems that in 'MFKZQ01LM_action_multistate' you are sending 'side' and 'from_side' and 'to_side'

Please forget previous posts.

I only miss two harware movements:
- IAM 
- SWING
and analog rotate sending 'elapsed_time' when rotating moreover the 'angle'.

segalion, 2019-12-27:
Please. Don't close. I think we can improve a little Xiaomi cube support.

bobzer, 2020-01-05:
HI, 

I agree that there is some improvment because the side number report is not always true 

and didn't know it can recognize more stuff but hey that would be cool

tayanov, 2020-01-14:
+1.  i think its be cool

segalion, 2020-03-18:
We don't know if suggested code has been included!!!

Koenkk, 2020-03-18:
In case you want to improve the integration please create a PR, the code can be found here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L940 , see this on how to edit: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

segalion, 2020-03-19:
Ok. Thanks. Sorry. Then, you can close.

closed 2020-05-25 by stale[bot]