Koenkk/zigbee2mqtt #2205: Device names starting with number make issues in jinga2

segalion, 2019-10-25:
When I want to make templates in HA, I have seen that default device names created by Z2M are of form:
'0x00158d000xxxxxxxx'
but this cause problems in jinja2 platform (because python variables cant start with a number).

Could be possible make them start with an underscore '_'
Even better could be first code replaced by the brand:
i.e '0x00158d000' replaced by 'xiaomi_'  so
device '0x00158d00018255df' could be 'xiaomi_18255df'

sjorge, 2019-10-25:
Wouldn't setting the friendly_name solve this?

segalion, 2019-10-27:
you cannot automate and template with friendly_name. Only device_id
But I have succesfully change Name and Device id.
I changed 
'sensor.0x00158d0001dab83b_powe' 
by 
'sensor.sensor.power_1dab83b'
and now jinga2 templates work fine

closed 2020-01-02 by stale[bot]