Koenkk/zigbee2mqtt #2206: Aqara vibration sensor failing to give me a state

bart1604, 2019-10-25:
Hi all,

I have a strange issue, because nothing has changed to my configuration as far as i know.
I have a vibration sensor of Xiaomi / Aqara. I have it attached to my doorbel ( I have an old analog one, so with a bell and a ball which rings the bell)

I've tried to delete the sensor from zigbee2mqtt and rejoin it, but nothing worked. If I go to the states page of HA, ring the doorbell, I can see a very fast state of vibration, but i dissapears in 1 ms and the state is empty. Because of this, my automation does not work (which should give me a notification someone is at the door)

If i go to the logs I can see the following:
`  zigbee2mqtt:info 10/25/2019, 5:35:52 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002b1609d', payload '{"battery":51,"voltage":2915,"linkquality":115,"angle_x":6,"angle_y":-79,"angle_z":9,"angle_x_absolute":84,"angle_y_absolute":169,"angle":10,"action":"vibration"}'
  zigbee2mqtt:info 10/25/2019, 5:35:52 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002b1609d', payload '{"battery":51,"voltage":2915,"linkquality":115,"angle_x":6,"angle_y":-79,"angle_z":9,"angle_x_absolute":84,"angle_y_absolute":169,"angle":10,"action":""}'`

In here I can see that there are 2 payloads and the second is empty at the section "action", but I have never experienced this issue. So I don't know if this was the same in the past.

My configuration file is as following:
`  - platform: "mqtt"
    state_topic: "zigbee2mqtt/0x00158d0002b1609d"
    name: Deurbel trilsensor
    availability_topic: "zigbee2mqtt/bridge/state"
    icon: "mdi:gesture-double-tap"
    value_template: "{{ value_json.action }}"
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/0x00158d0002b1609d"
    name: Deurbel trilsensor battery
    availability_topic: "zigbee2mqtt/bridge/state"
    unit_of_measurement: "%"
    device_class: "battery"
    value_template: "{{ value_json.battery }}"`

Zigbee2mqtt version: 1.6.0 stable

nanobra1n, 2019-11-04:
This is how the sensor works: vibration is reported only once every minute and for that little time window you observe. Mine does the same thing.

lolorc, 2019-11-04:
yep it sends
```
Nov 04 16:38:16 zigbee2mqtt/vibration_03 {"action":"vibration","linkquality":66,"elapsed":1032,"last_seen":1572881896071}                                    
Nov 04 16:38:16 zigbee2mqtt/vibration_03 {"action":"","last_seen":1572881896071}                                                        
```
this action is immediately set to ""

I wouldn't use a vibration sensor, I don't fully trust them. For instance this quoted event is a false positive. my 3 vibration sensors send false positive several time a day... (maybe because there're on high sensibility, but I pretty sure those are false positives)

bart1604, 2019-11-04:
Thanks for your comments.

The strange thing is that it has worked for more then 9 months without any issues or false positives. I've tested some more, when I ring my doorbell no notification is send to my zigbee2mqtt logs, even after ringing the bell several seconds, nothing. When I stop ringing the doorbell the 2 messages are send right after eachother.

I my Home Assistant (I think we all use HA), at the states page I see "vibration" for a flash of a second. I think this goes to fast and therefore my automation does not work.. 

As I said, it has worked before (the automation). But if this is how the sensor is supposed to work (which I doubt), how can anyone build an automation from it?!

nanobra1n, 2019-11-04:
In my case I use it to determine when the dryer is done. Every time a vibration is reported it kicks off a timer ( of a few minutes), when the timer expires, I get a notification. At the moment I have that automation disabled because as @lolorc wrote I get too many false positives on high sensitivity and medium sensitivity is not enough. 

I only use this sensors for a few weeks, so I can't comment how they might have worked in the past.

lolorc, 2019-11-04:
@bart1604  and you can use the same method with HA, every time a vibration is reported, you set a input_boolean to true and (re)start a timer that would set the input_boolean to false after a certain time. (it could be an input_boolean or a light or whatever)

this could also be implemented in z2m, but that would require z2m to do handle the timer(s), and provide some way of configuring the delay before sending a no-more-vibration mqtt message.

bart1604, 2019-11-05:
I think I found the issue for my case. For some reason my automation has changed.
I had a delay build in and for some reason the code changed to:
delay: ['object''object']

Which did not create a error when checking the code in HA. But it stopped my automation.
I've deleted the object en did an automation.trigger and it worked again.

Now the next problem, I've deleted the vibration sensor and I haven't been able to add it again. But that is something I'll figure out myself.

Thanks for your help so far!

Kind regards,
Bart

closed 2019-11-05 by Koenkk