Koenkk/zigbee2mqtt #2207: Starting from scratch - HOW???

dedepene, 2019-10-26:
# Bug Report
## What happened
ALL sorts of issues - devices not being able to join, not being able to remove devices, no netwrok route errors, Failed to execute LQI, most of the devices do not show any connections whatsoever on the net map etc. etc.

I really need to purge the whole thing and start over, but if I just uninstall the add-on and delete the z2m folder - I can still see all the devices after a fresh install. How do I make sure everything is completely removed for a fresh start???

Also, what is currently the most STABLE combination between cc2531 fw and z2m version?
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: 20190608

timdonovanuk, 2019-10-26:
"but if I just uninstall the add-on" - what do you mean?

The database.db file contains most things. If this is gone, you shouldn't see things. Alternatively, use the remove command to tell each device to drop from the controller (the cc2531).

dedepene, 2019-10-26:
by uninstall i mean hass.io>zigbee2mqtt add-on>uninstall. Even with clean database file, old devices somehow get populated even having removed mqtt integration from HA altogether. Manually removing devices via mqtt command did not work either - at one point I was getting an error in the z2m log, then after changing firmwares and z2m versions, the error ceased but the devices remain after remove command has been issued. Isn't there like an guide or something to ultimately purge it?

timdonovanuk, 2019-10-26:
> Manually removing devices via mqtt command did not work either - at one point I was getting an error in the z2m log

Which said..?

dedepene, 2019-10-26:
can't remember, don't want to mess with it anymore and replicate after having spent hours upon hours trying to troubleshoot. Just need some pointers on how to raze the whole thing to the ground and start from a scratch.

foxy82, 2019-11-01:
When you uninstall the hass.io addin make sure the /share folder has the zigbee2mqtt stuff removed as that is persistent storage it might not remove it on an uninstall which might be why your devices are coming back.

dedepene, 2019-11-01:
> When you uninstall the hass.io addin make sure the /share folder has the zigbee2mqtt stuff removed as that is persistent storage it might not remove it on an uninstall which might be why your devices are coming back.

I'm pretty sure I had the folder removed manually to be sure. Yet the devices came back. Anyway, I'm giving up on z2m for now, just ordered a Conbee stick...

closed 2020-01-07 by stale[bot]