Koenkk/zigbee2mqtt #2213: Unable to pair Sylvania 73661 Smart+ LED strip

nanobra1n, 2019-10-27:
I am unable to pair Sylvania 73661 Smart+ LED strip. The strip itself flashes 4 times after joining is enabled (which I believe indicates that the strip is paired), but the log only reports
`Cannot get the Node Descriptor of the Device: 0x7cb03eaa00a4d9cf (Error: Timed out after 10000 ms)`

The stick (using is h4nc's antenna modded stick) is 3 feet away and I was able to pair Cree connected bulbs without problem) 

Zigbee2mqtt version 1.6.0
Coordinator version: 20190608

Koenkk, 2019-10-27:
Please check with the latest dev branch, pairing has been improved there.

nanobra1n, 2019-10-27:
No success with it either. Log is here:

https://pastebin.com/CX6XrfY3

Koenkk, 2019-10-27:
Can you run zigbee2mqtt with `DEBUG=zigbee-herdsman:controller* npm start`

nanobra1n, 2019-10-27:
Sorry, I need a bit of hand holding for this. I am running the zigbee2mqtt add-on in docker and I am just not familiar enough with docker to start the image with that command

Koenkk, 2019-10-28:
Via the hassio config, add `zigbee_herdsman_debug: true`.

nanobra1n, 2019-10-29:
Here it is. I hope it is adequate
https://pastebin.com/ArevTpzf

nanobra1n, 2019-10-30:
Here you go:

https://pastebin.com/QRE8jk6g

To rule out that the strip is defective I  would like to point out that I can pair the strip with a Wink 2 Hub. The uncertain future of Wink is the motivator to switch to z2m.

Koenkk, 2019-10-30:
For some reason it just doesn't respond to interview requests; are you using the reset procedure from: http://belkin.force.com/Articles/articles/en_US/Troubleshooting_and_Tutorials/Resetting-the-OSRAM-LIGHTIFY-Flex-RGBW?

nanobra1n, 2019-10-30:
I use a 5s on/ 5s off power cycle as described here : https://community.smartthings.com/t/how-to-re-pair-osram-lightify-flex-strip/78551

The strip cycles through its colors after this.

Koenkk, 2019-10-30:
Seems that more people are having problems: https://community.smartthings.com/t/how-to-re-pair-osram-lightify-flex-strip/78551/10

Could you sniff the traffic while pairing it with the SmartThings hub? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

nanobra1n, 2019-10-30:
I can get a reset consistently with the 5s on/off cycle.

> Could you sniff the traffic while pairing it with the Wink hub?
Not at this time. I got a preflashed stick from h4nc and while I ordered the flashing hw from Ali Express, I don't expect it to arrive for another 3-4 weeks.

nanobra1n, 2019-11-01:
So I didn't want to wait that long and got a sniffer off Amazon. I hope I did it right but here is the capture of the successful pairing with the Wink hub

https://www.dropbox.com/s/qpvyyedj1xgd7m3/nanobrain.pcapng?dl=0

Koenkk, 2019-11-01:
Thanks, can you do the same now when pairing it with zigbee2mqtt?

nanobra1n, 2019-11-02:
Well, I'll be ...

After I don't know how many tries, my strip paired while I sniffed. Just because I went to the trouble to collect it, here is the capture:

https://www.dropbox.com/s/i065x7gn157xh6u/nanobra1n_z2mpair.pcapng?dl=0

and the log is here:

https://pastebin.com/4Wz06D7G

the only thing I consciously did differently is that I restarted z2m right before pairing. I have been pairing a lot recently and it seems that pairing freshly started gave less issues.

Koenkk, 2019-11-02:
Ive compared the logs but the interview procedure is almost the same (in your case it failed at a point where they were even exactly the same). Perhaps pairing it to another hub triggered something on the bulb. I will monitor if more users encpunter this.

closed 2019-11-02 by Koenkk