Koenkk/zigbee2mqtt #2214: HA doesn't retain status of light bulbs and sensors on reboot

DerpyDoom, 2019-10-27:
Hi,

I am currently running a HA Smarthome system on a Raspberry Pi. Everything is running perfectly like I want. The only problem I've noticed is that everything that has to do with zigbee is not being retained on reboot.

I've set retain to true on every registered sensor/light, still no luck.

Koenkk, 2019-10-30:
Do you have the following in home assistant `configuration.yaml`?
```yaml
mqtt:
  discovery: true
  broker: [YOUR MQTT BROKER]  # Remove if you want to use builtin-in MQTT broker
  birth_message:
    topic: 'hass/status'
    payload: 'online'
  will_message:
    topic: 'hass/status'
    payload: 'offline'
```

DerpyDoom, 2019-10-31:
Nope, but that seems to have fixed it.

Thank you!

closed 2019-10-31 by DerpyDoom