Koenkk/zigbee2mqtt #2215: Zigbee2MQTT 1.6.0: Cannot find module 'zigbee-herdsman/dist/ziee'

IWAsGIT, 2019-10-27:
Hello,
I destroyed my zigbee2mqtt with trying to update it.
I use a RPi3 with Raspbian stretch 9.11. Zigbee2MQTT interacts (interacted) with FHEM. Here some information for the different interesting versions:
npm  = 6.9.0
node = 10.16.3
zigbee2mqtt = 1.6.0
zigbee-herdsman = 0.11.2

I installed Zigbee2MQTT last year with the current version from September 2018. Now I have to update it, because of a device which has been unknown in this old version.
So I followed the update instructions (https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html). After some troubles with the npm and node-versions I become desperate now. Because of the problems I tried the following steps but got the following errors with installing npm:

```
pi@IWA3:/opt/zigbee2mqtt $ rm -rf node_modules/
pi@IWA3:/opt/zigbee2mqtt $ npm install

> @serialport/bindings@2.0.8 install /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.16.3 runtime=node arch=arm libc= platform=linux)
make: Verzeichnis „/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build“ wird betreten
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Verzeichnis „/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build“ wird verlassen

> @serialport/bindings@2.0.8 install /opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.16.3 runtime=node arch=arm libc= platform=linux)
make: Verzeichnis „/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build“ wird betreten
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Verzeichnis „/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build“ wird verlassen

> core-js-pure@3.1.4 postinstall /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/core-js-pure
> node scripts/postinstall || echo "ignore"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon: 
> https://opencollective.com/core-js 
> https://www.patreon.com/zloirock 

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

> core-js-pure@3.1.4 postinstall /opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/core-js-pure
> node scripts/postinstall || echo "ignore"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon: 
> https://opencollective.com/core-js 
> https://www.patreon.com/zloirock 

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 2097 packages from 594 contributors and audited 877679 packages in 323.171s
found 3 vulnerabilities (2 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details
```

After npm audit fix there's still one last vulnerability:
Moderate - Memory Exposure, Package bl, patched in >=0.9.5 <1.0.0 || >=1.0.1
With npm view package bl version I found out, that my version is 4.0.0.
So I don't know what's the problem.

So my npm start also fails:
```
pi@IWA3:/opt/zigbee2mqtt $ sudo npm start

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'zigbee-herdsman/dist/deprecated/ziee'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/zapp/cie.js:6:14)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-10-27T12_02_38_973Z-debug.log
```

I spent many hours about this problem in the last days and tried many solutions from the different issues (e.g. https://github.com/Koenkk/zigbee2mqtt/issues/1892). Can somebody help me with my problem?

sjorge, 2019-10-27:
1.6.0 still uses zigbee-shepard AFAIK, it's only dev that for now is using zigbee-herdsman.

Have you tried deleting your node_modules folder and running npm install again?

IWAsGIT, 2019-10-27:
Thank you for your answer!
Yes I already tried it for more times. It leads to exactly this result.
Maybe it's a good idea to do back to a 1.5.x version. But I don't know how to do it. How can I downgrade to version 1.5.x?

sjorge, 2019-10-27:
I'm not sure how you can go back to 1.5.0... you could try and go to dev though, just checkout the dev branch instead of master.

IWAsGIT, 2019-10-27:
Hm, I'm not really fit in git. I tried now to get the version 1.5.1 (https://github.com/Koenkk/zigbee2mqtt/tree/1.5.1) with the commit ac3b924.
I tried the following steps like the original instruction, but with the commit instead of the head. But it doesn't work, I'm still in 1.6.0:

```
# Stop zigbee2mqtt and go to directory
sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt

# Backup configuration
cp -R data data-backup

# Update
git checkout ac3b924 -- npm-shrinkwrap.json
# git pull # tried with and without this line
rm -rf node_modules
npm install

# Restore configuration
cp -R data-backup/* data
rm -rf data-backup

# Start zigbee2mqtt
sudo systemctl start zigbee2mqtt
```
What's wrong for changing the version to the older one?

IWAsGIT, 2019-10-28:
I resetted zigbee2mqtt from 1.6.0 to 1.5.1 now and it works with the following steps:

```
# Stop zigbee2mqtt and go to directory
sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt

# Backup configuration
cp -R data data-backup

# Update
git reset --hard ac3b924
rm -rf node_modules
npm install

# Restore configuration
cp -R data-backup/* data
rm -rf data-backup

# Start zigbee2mqtt
sudo systemctl start zigbee2mqtt
```

So I haven't solved the beginning problem 'zigbee-herdsman/dist/deprecated/ziee' for 1.6.0 on my system but my system can work again.

closed 2019-10-28 by IWAsGIT