Koenkk/zigbee2mqtt #2216: Osram Plug unavailable in HA but responding via mqtt messages

Bruceforce, 2019-10-27:
# Bug Report
## What happened
One of my two osram plugs
```
  '0x7cb03eaa0a00ed53':
    friendly_name: osram_plug_02
```

becomes unavailable in home assistant but it still receives command via mqtt directly. I'm using mosquitto as mqtt server and I'm running everyting inside docker containers.

## What did you expect to happen
The Osram plug shouldn't be unavailable in home assistant since it still seems to process mqtt messages.

## How to reproduce it (minimal and precise)
1. start up zigbee2mqtt docker container
1. wait until osram plug becomes unavaible in home assistant
1. send direct mqtt message `zigbee2mqtt/0x7cb03eaa0a00ed53/set` with payload `{"state": "ON"}` via MQTT.fq

## Debug Info
zigbee2mqtt version: 1.6.0 (commit #e26ad2a)
CC253X firmware version: 20190608

Debug logs are here: https://gist.github.com/Bruceforce/c91af88522876fc3bd32e237062395a0

Koenkk, 2019-10-28:
Can you try with the latest dev  branch?

Bruceforce, 2019-11-03:
I used the `:latest-dev` tag of the docker image. Now all my devices seems to be unpaired.

```
info  2019-11-03T18:57:21: Logging to directory: '/app/data/log/2019-11-03.19-57-21'
debug 2019-11-03T18:57:21: Removing old log directory '/app/data/log/2019-10-29.10-52-46'
debug 2019-11-03T18:57:21: Loaded state from file /app/data/state.json
info  2019-11-03T18:57:21: Starting zigbee2mqtt version 1.6.0 (commit #8a5f39f)
info  2019-11-03T18:57:21: Starting zigbee-herdsman...
debug 2019-11-03T18:57:21: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[25],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","backupPath":"/ap
p/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2019-11-03T18:57:22: zigbee-herdsman started
info  2019-11-03T18:57:22: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
debug 2019-11-03T18:57:22: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":25}
info  2019-11-03T18:57:22: Currently 0 devices are joined:
info  2019-11-03T18:57:22: Zigbee: disabling joining new devices.
info  2019-11-03T18:57:22: Connecting to MQTT server at mqtt://mosquitto:1883
error 2019-11-03T18:57:22: Cannot find '0x000d6ffffe529eda' of group 'living_room_ceiling'
error 2019-11-03T18:57:22: Cannot find '0x90fd9ffffe32015e' of group 'living_room_ceiling'
error 2019-11-03T18:57:22: Cannot find '0xd0cf5efffed828c3' of group 'living_room_ceiling'
error 2019-11-03T18:57:22: Cannot find '0x000d6ffffe4c1e88' of group 'bedroom_ceiling'
error 2019-11-03T18:57:22: Cannot find '0x000d6ffffe62696f' of group 'bedroom_ceiling'
error 2019-11-03T18:57:22: Cannot find '0x000d6ffffe63c485' of group 'bedroom_ceiling'
info  2019-11-03T18:57:22: Connected to MQTT server
info  2019-11-03T18:57:22: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2019-11-03T18:57:22: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"8a5f39f","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revi
sion":20190608}},"log_level":"debug","permit_join":false}'
debug 2019-11-03T18:57:58: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
info  2019-11-03T18:57:58: Zigbee: allowing new devices to join.
info  2019-11-03T18:57:58: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"8a5f39f","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revi
sion":20190608}},"log_level":"debug","permit_join":true}'
debug 2019-11-03T18:59:59: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'false'
info  2019-11-03T18:59:59: Zigbee: disabling joining new devices.
info  2019-11-03T18:59:59: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"8a5f39f","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revi
sion":20190608}},"log_level":"debug","permit_join":false}'
```

Is this expected?

Koenkk, 2019-11-05:
This is not expected, zigbee-herdsman should only clear your database when changing the panid or channel (assuming you didnt do this).

Unfortunately you have to re-pair now. I will make sure that zigbee-herdsman will make a backup before removing it.

Bruceforce, 2019-11-05:
No I didn't change anything in configuration.yaml. I have a daily backup of `database.db` and tried to restore it but it was cleared at every restart of zigbee2mqtt. I can try to find it in the Backup again and share it with you if that helps.

However I can't really test if the original issue is resolved by the new zigbee2mqtt version because it gets stuck every few days showing all devices unavailable. I have to Reset the CC2531 every time this happens. This wasn't the case in the previous version. I think I should open a new issue for that?

Koenkk, 2019-11-05:
Can you restore the database backup and provide the log when starting zigbee2mqtt with `DEBUG=* npm start`

Bruceforce, 2019-11-05:
I tried this and now it is working somehow 🙄  (at least the db is not cleared anymore) but devices show unavailable with the database backup. I assume this is, because I re-paired them?

I still can't test properly with the `latest-dev` tag because of #2280

Bruceforce, 2019-11-17:
It's working for now with the latest dev branch and source routing firmware on the CC2531.

Bruceforce, 2019-12-15:
@Koenkk 
I reopen this, because it started to happen again with source routing firmware and latest-dev docker container of zigbee2mqtt. It happens to "random" devices. Once it was a Osram plug and now it is a tradfri bulb. 

Do you need me to collect logs with `Debug=* npm start` again or should I try something else?

Koenkk, 2019-12-15:
I'm afraid we cannot do anything further from the zigbee2mqtt side. When sniffing the traffic you will see that the device doesn't respond making zigbee2mqtt think that the device is offline. This seems to happen often with Osram devices (probably a bug in the Osram firmware). You can add it to the `availability_blacklist` https://www.zigbee2mqtt.io/information/availability.html.

Bruceforce, 2019-12-15:
Okay got it. If I add it to the blacklist, would I still be able to control it via home Assistant if it doesn't respond to zigbee2mqtt?

Just want to add that you are doing an awesome job with this project and with answering to all this reports in here.

Koenkk, 2019-12-16:
When added to the blacklist it will never be marked as unavailable. But still if the connection really fails it cannot be controlled.

closed 2020-02-14 by Koenkk