Koenkk/zigbee2mqtt #2217: Failed to call 'DeviceReceive' 'onZigbeeEvent'

ksuclipse, 2019-10-27:
# Bug Report
## What happened

Version: (commit #8d97262)
 
Error during join of device:

zigbee2mqtt:error 2019-10-27T15:34:54: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at mappedDevice.fromZigbee.filter (/app/lib/extension/deviceReceive.js:83:42)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/app/lib/extension/deviceReceive.js:82:52)
    at Controller.callExtensionMethod (/app/lib/controller.js:320:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:228:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/app/lib/zigbee.js:56:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:412:22)
    at Generator.next (<anonymous>))

## What did you expect to happen
device to join
## How to reproduce it (minimal and precise)
try to join certain devices like sylvania plug or body motion sensor xiaomi

## Debug Info
zigbee2mqtt version:
CC253X firmware version:
Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

Koenkk, 2019-10-28:
- Does it still occur with the latest dev branch?
- When this error happens; can you provide the output when running zigbee2mqtt with `DEBUG=zigbee-herdsman:controller* npm start`.

marcutzo84, 2019-11-13:
If could help I had same issue with 1.7.1 and I solved disabling this in the config:

 "zigbee_shepherd_devices": false

before was true and the issue occured. With false now everything works again

Marco.

closed 2020-01-20 by stale[bot]

mihadoo, 2020-02-07:
I have the same problem. Here is the log:

zigbee2mqtt:debug 2020-02-07 20:03:04: Received Zigbee message from 'atticswitch1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:error 2020-02-07 20:03:04: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at mappedDevice.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:106:42)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:105:52)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:327:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:228:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/opt/zigbee2mqtt/lib/zigbee.js:57:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:447:22)
    at Generator.next (<anonymous>))

This is what I have added to devices.js:

    // Ksentry Clone from AliExpress
    {
        zigbeeModel: ['SA-003-Zigbee'],
        model: 'SA-003-Zigbee',
        vendor: 'Ksentry Electronics China Clone',
        description: '[Zigbee OnOff Controller](http://ksentry.manufacturer.globalsources.com/si/6008837134660'+
                     '/pdtl/ZigBee-module/1162731630/zigbee-on-off-controller-modules.htm)',
        supports: 'on/off',
        fromZigbee: [fz.state_change, fz.state],
        toZigbee: [tz.on_off],
    },