Koenkk/zigbee2mqtt #2220: Presence of `transition` parameter influences brightness in ON command

Otah, 2019-10-28:
# Bug Report
## What happened
When a light is on and has e.g. `brightness: 50`, sending a command `{"state": "ON", "transition": 1}` sets the brightness to maximum (255). Whereas if you send just `{"state": "ON"}`, nothing happens (= the light stays on with brightness at level 50).

It behaves the same way no matter what the actual value of `transition` is. Just its presence changes the behavior. The same situation is also if the default `transition` is set in the device-specific options.
## What did you expect to happen
I expect if sending an ON command to a light which is already on, nothing will happen. I definitely not expect that it should change the brightness and especially I don't expect that presence of `transition` parameter should change how it behaves.
## How to reproduce it (minimal and precise)
Given a Philips White Ambiance bulb (I only have Philips Hue bulbs, no other manufacturer to test) without default `transition` parameter in device-specific options, publishing one by one to its `/set` topic:
```
{"state": "ON", "brightness": 50} // turns on the light and sets the brightness to 50
{"state": "ON"}                   // correctly does nothing, as the light is already on
{"state": "ON", "transition": 1}  // incorrectly increases the brightness to max
```
## Debug Info
zigbee2mqtt version: commit `26ad2a` (version 1.6.0), was also present on `927c4db`
CC253X firmware version: _I think this is a software issue, but I can provide if necessary_

From the log, I'd guess it's connected to the fact that the `transition` parameter changes the Zigbee command from:
`Zigbee publish to device 'ambianceBulb3', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - 11`
to:
`Zigbee publish to device 'ambianceBulb3', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":10} - {"manufSpec":0,"disDefaultRsp":0} - 11`

The topic sounds similar to the #1260 but not the issue itself.

Koenkk, 2019-10-28:
Should be fixed in latest zigbee2mqtt dev.

pepp86, 2019-11-03:
Experiencing the same issue when switching lamps off with transition, brightness is set to 0 which makes some trouble in my installation.

Koenkk, 2019-11-03:
@pepp86 I guess it depends on the bulb, why dont you leave out the transition?

pepp86, 2019-11-04:
@Koenkk  Having this issue on all devices supporting transition.

I want to use transition, so I can't leave it out!?

Koenkk, 2019-11-05:
@copystring @pepp86 your issue should be fixed in latest dev, please check and confirm.

Koenkk, 2019-11-07:
Thanks, assuming this can be closed now.

closed 2019-11-07 by Koenkk