Koenkk/zigbee2mqtt #2221: Support for lumi.curtain.aq2

Dennisd80, 2019-10-28:
I would like to have the aqara roller motor to be supported.  

It's a roller blind motor by aqara/xiaomi. I can't seems to add it to devices,js (using hassio) probally because i'm on zigbee2mqtt edge. Would somebody be so kind to add this device?

Koenkk, 2019-10-30:
I think this can easily be done by changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L675 to `zigbeeModel: ['lumi.curtain', 'lumi.curtain.aq2'],`

Dennisd80, 2019-11-01:
Unfortunately after changing devices.js i get the following error. zigbee2mqtt:warn 11/1/2019, 9:01:47 PM Device with modelID 'lumi.curtain.aq2' is not supported. (hassio addon)

Koenkk, 2019-11-01:
Did you set `zigbee_shepherd_devices: true`?

Dennisd80, 2019-11-02:
Yes, i did. It also recognizes the custom devices.js. But i saw i had a error that i't could copy the custom device.js  to node_module/zigbee2mqtt/herdsman-converter. (was on stable zigbee2mqtt), then i decided to try on zigbee2mqttedge and now it seems to work. Thanks!

closed 2019-11-02 by Dennisd80

Koenkk, 2019-11-02:
So after this the device fully works?

Dennisd80, 2019-11-02:
Yes 100%

Koenkk, 2019-11-03:
Thanks, will be supported out of the box in the next zigbee2mqtt release.