Koenkk/zigbee2mqtt #2222: Failed to whitelist device

fatemeh76m, 2019-10-28:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

fatemeh76m, 2019-10-28:
Using topic zigbee2mqtt/bridge/config/whitelist and sending device friendly_name as payload always results in "failed to whitelist <friendly_name>", but in the configuration.yaml file the friendly_name can be seen in the whitelist.
what is the possible problem?

[image]
Is the device added to the whitelist or not? If it is added, why other devices which are not in whitelist can join to network?

Koenkk, 2019-10-28:
Can you try with latest zigbee2mqtt dev?

fatemeh76m, 2019-11-02:
The code I'm working on it, is for one month ago. Have you changed anything in whitelist since then?

Koenkk, 2019-11-03:
I guess the whitelisting fails because it is already whitelisted. can you share your `configuration.yaml`?

Arminhosseini, 2019-11-09:
Hi, I have this problem too but if we assume you are right white list does not work as I expected because in zigbee2mqtt topics you mentioned when a device whitelisted other devices will be removed but this not happened for me. and in your code, there is an if condition that when a white list is not empty if we connecting any other devices that are not whitelisted coordinator should remove it but it's not happening
because however my whitelisted device in conf.yaml is not empty consider as null and continue to add device.

Arminhosseini, 2019-11-09:
another thing is that when I whitelist a new device that I didn't whitelist this error will show again.
and what is the latest dev?
thank's for your reply.

Koenkk, 2019-11-10:
@Arminhosseini please provide debug logs and your `configuration.yaml` when this problems occurs.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

closed 2020-01-09 by Koenkk