Koenkk/zigbee2mqtt #2224: Hue dimmer switch not pairing

HA-TB303, 2019-10-29:
# Bug Report
## What happened
Hue FDimmer switch: 324131092621 not pairing

## What did you expect to happen
For it to pair correctly, I get:
```
zigbee2mqtt:info  2019-10-29T11:30:13: Accepting joining non-banned device '0x00178801103f0ccb'
zigbee2mqtt:info  2019-10-29T11:30:13: Accepting joining non-banned device '0x00178801103f0ccb'
zigbee2mqtt:debug 2019-10-29T11:30:13: Device '0x00178801103f0ccb' announced itself
```
but nothing more.

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: Edge branch on Hassio
Coordinator: CC2530 + CC2591 version: 20190608
````
zigbee2mqtt:info  2019-10-29T11:31:40: Starting zigbee2mqtt version 1.6.0 (commit #36c7a27e6c632e0450db18279fc46a1d01365579)
zigbee2mqtt:info  2019-10-29T11:31:40: Starting zigbee-herdsman...
zigbee2mqtt:info  2019-10-29T11:31:42: zigbee-herdsman started
zigbee2mqtt:info  2019-10-29T11:31:42: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:info  2019-10-29T11:31:42: Currently 26 devices are joined:
````

james-fry, 2019-10-29:
I know this is not helping you, but I paired a couple of hue dimmers yesterday, on version:
zigbee2mqtt version 1.6.0 (commit #cb3ec45) or cc1352p-2 with latest zstack 3.x.0 f/w

```
info  2019-10-28T20:47:16: Device '0x0017880104092eb5' joined
info  2019-10-28T20:47:16: MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0017880104092eb5"}}'
info  2019-10-28T20:47:16: Starting interview of '0x0017880104092eb5'
info  2019-10-28T20:47:16: MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0017880104092eb5"}}'
debug 2019-10-28T20:47:17: Device '0x0017880104092eb5' announced itself
debug 2019-10-28T20:47:29: Received Zigbee message from '0x0017880104092eb5', type 'readResponse', cluster 'genBasic', data '{"modelId":"RWL021","manufacturerName":"Philips","powerSource":3}' from endpoint 1 with groupID 0
debug 2019-10-28T20:47:29: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"RWL021","manufacturerName":"Philips","powerSource":3}'
debug 2019-10-28T20:47:29: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
info  2019-10-28T20:47:29: MQTT publish: topic 'homeassistant/sensor/0x0017880104092eb5/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt_cc1352p/0x0017880104092eb5","json_attributes_topic":"zigbee2mqtt_cc1352p/0x0017880104092eb5","name":"0x0017880104092eb5_action","unique_id":"0x0017880104092eb5_action_zigbee2mqtt_cc1352p","device":{"identifiers":["zigbee2mqtt_0x0017880104092eb5"],"name":"0x0017880104092eb5","sw_version":"Zigbee2mqtt 1.6.0","model":"Hue dimmer switch (324131092621)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt_cc1352p/bridge/state"}'
info  2019-10-28T20:47:29: MQTT publish: topic 'homeassistant/sensor/0x0017880104092eb5/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt_cc1352p/0x0017880104092eb5","json_attributes_topic":"zigbee2mqtt_cc1352p/0x0017880104092eb5","name":"0x0017880104092eb5_battery","unique_id":"0x0017880104092eb5_battery_zigbee2mqtt_cc1352p","device":{"identifiers":["zigbee2mqtt_0x0017880104092eb5"],"name":"0x0017880104092eb5","sw_version":"Zigbee2mqtt 1.6.0","model":"Hue dimmer switch (324131092621)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt_cc1352p/bridge/state"}'
info  2019-10-28T20:47:29: MQTT publish: topic 'homeassistant/sensor/0x0017880104092eb5/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt_cc1352p/0x0017880104092eb5","json_attributes_topic":"zigbee2mqtt_cc1352p/0x0017880104092eb5","name":"0x0017880104092eb5_linkquality","unique_id":"0x0017880104092eb5_linkquality_zigbee2mqtt_cc1352p","device":{"identifiers":["zigbee2mqtt_0x0017880104092eb5"],"name":"0x0017880104092eb5","sw_version":"Zigbee2mqtt 1.6.0","model":"Hue dimmer switch (324131092621)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt_cc1352p/bridge/state"}'
debug 2019-10-28T20:47:29: Received Zigbee message from '0x0017880104092eb5', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":2,"stackVersion":1}' from endpoint 1 with groupID 0
debug 2019-10-28T20:47:29: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1,"appVersion":2,"stackVersion":1}'
debug 2019-10-28T20:47:29: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2019-10-28T20:47:30: Received Zigbee message from '0x0017880104092eb5', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20160302","swBuildId":"5.45.1.17846"}' from endpoint 1 with groupID 0
debug 2019-10-28T20:47:30: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20160302","swBuildId":"5.45.1.17846"}'
debug 2019-10-28T20:47:30: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
info  2019-10-28T20:47:30: Successfully interviewed '0x0017880104092eb5', device has successfully been paired
info  2019-10-28T20:47:30: Device '0x0017880104092eb5' is supported, identified as: Philips Hue dimmer switch (324131092621)
info  2019-10-28T20:47:30: MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x0017880104092eb5","model":"324131092621","vendor":"Philips","description":"Hue dimmer switch","supported":true}}'
info  2019-10-28T20:47:30: Configuring '0x0017880104092eb5'
info  2019-10-28T20:47:33: Succesfully configured '0x0017880104092eb5'
debug 2019-10-28T20:47:36: Received Zigbee message from '0x0017880104092eb5', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"unknown1":3145728,"type":0,"unknown2":0,"time":0}' from endpoint 2 with groupID 0
debug 2019-10-28T20:47:36: Received Zigbee message from '0x0017880104092eb5', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
[... cut ...]
debug 2019-10-28T20:47:37: Received Zigbee message from '0x0017880104092eb5', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"unknown1":3145728,"type":2,"unknown2":0,"time":1}' from endpoint 2 with groupID 0
info  2019-10-28T20:47:37: MQTT publish: topic 'zigbee2mqtt_cc1352p/0x0017880104092eb5', payload '{"counter":1,"linkquality":72,"last_seen":1572295657680,"action":"off-press","duration":0}'
info  2019-10-28T20:47:37: MQTT publish: topic 'zigbee2mqtt_cc1352p/0x0017880104092eb5', payload '{"counter":1,"linkquality":72,"last_seen":1572295657680,"action":""}'
```

Can you reset the dimmer and try again?

HA-TB303, 2019-10-29:
I have tried numerous times, I can't remove and can't succesfully pair and use.

HA-TB303, 2019-10-29:
I however use the zstack firmware 1.2. I am reluctant to update to 3.0 since pairing all my zigbee devices is a few hours work :(

Koenkk, 2019-10-29:
Can you share your database.db?

HA-TB303, 2019-10-29:
Is it safe to share publicly?

Koenkk, 2019-10-29:
Yes; no security critical information is inside.

HA-TB303, 2019-10-29:
blub..

[database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/3783574/database.db.zip)

Koenkk, 2019-10-29:
It looks that the device has been paired correctly (`0x00178801103f0ccb`), but I guess the actual problem is that you don't see button clicks?

If that is the case, can you try to re-trigger a configure, stop zigbee2mqtt, change `"configured":1` to `"configured":0` for this device in `database.db` and start zigbee2mqtt; while starting zigbee2mqtt press buttons on the remote (to keep it awake) until you see `Succesfully configured` for this device.

HA-TB303, 2019-10-29:
```
zigbee2mqtt:error 2019-10-29T13:08:17: Failed to configure '0x00178801103f0ccb', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
```

HA-TB303, 2019-10-29:
It works again. I finally **reset** the device, restarted zigbee2mqtt and kept pressing the buttons again and now it works ;)

Koenkk, 2019-10-29:
Great!

closed 2019-10-29 by Koenkk

Garfonso, 2019-12-11:
@HA-TB303 
Hi, I think I might have the same problem like you. Might I ask, how and when did you exactly reset the switch in order for it to react to the configuration attempt? I always get the timeout.