Koenkk/zigbee2mqtt #2225: Make zigbee2mqtt more robust against errors in DB.

schwabe, 2019-10-29:
# Bug Report
## What happened
Wrong parsing of device information leads to crash on startup. 

/opt/zigbee2mqtt/lib/controller.js:262
        Object.keys(payload).forEach((key) => {
               ^

TypeError: Cannot convert undefined or null to object
    at Function.keys (<anonymous>)
    at Controller.iteratePayloadForAttrOutput (/opt/zigbee2mqtt/lib/controller.js:262:16)
    at Object.keys.forEach (/opt/zigbee2mqtt/lib/controller.js:278:29)
    at Array.forEach (<anonymous>)
    at Controller.iteratePayloadForAttrOutput (/opt/zigbee2mqtt/lib/controller.js:262:30)
    at Controller.publishEntityState (/opt/zigbee2mqtt/lib/controller.js:256:22)
    at zigbee.getAllClients.forEach (/opt/zigbee2mqtt/lib/controller.js:221:22)
    at Array.forEach (<anonymous>)
    at Controller.sendAllCachedStates (/opt/zigbee2mqtt/lib/controller.js:219:37)
    at Controller.onMQTTConnected (/opt/zigbee2mqtt/lib/controller.js:87:14)

## What did you expect to happen
It starts but logs warnings and discards invalid data:

## How to reproduce it (minimal and precise)
I added logging information to the controller.js

```
diff --git a/lib/controller.js b/lib/controller.js
index 8c74844..ca69ae6 100644
--- a/lib/controller.js
+++ b/lib/controller.js
@@ -259,6 +259,9 @@ class Controller {
     }
 
     iteratePayloadForAttrOutput(topicRoot, payload, options) {
+       console.log(payload)    
+       if (payload === null)
+               return;
         Object.keys(payload).forEach((key) => {
             let subPayload = payload[key];
             let message;
```

This makes zigbee2mqtt startup. It also logs the following:

```
{ linkquality: 65, battery: 'NaN', voltage: null }
  zigbee2mqtt:info 10/29/2019, 1:53:02 PM MQTT publish: topic 'zigbee2mqtt/0x000d6f000feab1cc/linkquality', payload '65'
  zigbee2mqtt:info 10/29/2019, 1:53:02 PM MQTT publish: topic 'zigbee2mqtt/0x000d6f000feab1cc/battery', payload 'NaN'
null
```
The line from `database.db`:

```{"id":16,"type":"EndDevice","ieeeAddr":"0x000d6f000feab1cc","nwkAddr":13418,"manufId":4364,"manufName":"OSRAM","powerSource":"Battery","modelId":"Lightify Switch Mini\u0000","epList":[1,2,3],"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"swBuildId":"e.1.9.0M","status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":2064,"inClusterList":[0,1,32,4096,64768],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Lightify Switch Mini\u0000","dateCode":"201711151316150","powerSource":3,"swBuildId":"e.1.9.0M"}},"genPowerCfg":{"dir":{"value":1},"attrs":{"batteryVoltage":31,"batteryVoltMinThres":24,"batteryVoltThres1":26,"batteryVoltThres2":27,"batteryVoltThres3":28,"batteryAlarmState":0}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{"checkinInterval":0,"longPollInterval":187,"shortPollInterval":4,"fastPollTimeout":720}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Lightify Switch Mini\u0000","dateCode":"201711151316150","powerSource":3,"swBuildId":"e.1.9.0M"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":2064,"inClusterList":[0,4096,64768],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"64768":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":0,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Lightify Switch Mini\u0000","dateCode":"201711151316150","powerSource":3,"swBuildId":"e.1.9.0M"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}}},"dateCode":"201711151316150","_id":"K8RF6azfBuTzDgv7"}```

## Debug Info
zigbee2mqtt version: current master

My workaround is probably not the best but I do not know the code base well enough to figure out a real fix.

Koenkk, 2019-10-29:
Does it still occur with zigbee2mqtt latest-dev?

schwabe, 2019-10-29:
Sorry, for not realising that `master` was the stable branch and not the development branch, most projects to do it different. 

But it looks that it started at least. And this *might* be the problematic device:

```
zigbee2mqtt:error 2019-10-29T13:15:43: Failed to call 'DeviceConfigure' 'onZigbeeStarted' (TypeError: Cannot read property 'friendlyName' of null
    at DeviceConfigure.configure (/opt/zigbee2mqtt/lib/extension/deviceConfigure.js:64:52)
    at DeviceConfigure.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/deviceConfigure.js:41:28)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44))
```

Closing this for now and will reopen this when I see a new crash related to this.

closed 2019-10-29 by schwabe

CodeFinder2, 2019-11-11:
> Sorry, for not realising that `master` was the stable branch and not the development branch, most projects to do it different.
>
> But it looks that it started at least. And this _might_ be the [...]

I am getting the same exception and just based on the output, I am not able to identify the device that is causing this error. (Clearly, it may also be related to an invalid configuration file but I was also not able to find any errors just by grep'ping through all my configs files, especially, database.db and state.json. They appeared okay to me.)

@Koenkk Can you add/include more information in the output to point us to the faulty device when this message is shown? I've already taken a look at lib/controller.js:320 but I am somewhat lost in the code (don't know how to modify it but I am also not really a JS guy). I would like to get rid of this error. Using (latest) v1.7, by the way.

Koenkk, 2019-11-11:
@CodeFinder2 can you check if all devices that are in `database.db` are in the `configuration.yaml`? Search for ieeeAddr.

CodeFinder2, 2019-11-11:
@Koenkk I came up with writing this little Python script to do the check for me (maybe helpful for other running into this issue):

```python
#!/usr/bin/env python

import yaml
import json

device_file = '[]'
database_file = 'database.db'
addr_key = 'ieeeAddr'

with open(device_file, 'r') as stream:  # YAML
    try:
        devs = [ d for d in yaml.safe_load(stream) ]
        with open(database_file, 'r') as json_file:  # JSON per line
            db = [json.loads(l) for l in json_file.readlines()]
            addrs = [ e[addr_key] for e in db ]
            print("INFO: Found {} device(s) ('{}' keys) in '{}' and {} device(s) in '{}'.".format(len(addrs), addr_key, database_file, len(devs), device_file))
            for addr in addrs:
                if addr not in devs:
                    print("WARNING: Unable to find {} in '{}'".format(addr, device_file))
    except yaml.YAMLError as exc:
        print(exc)
```
And it gave me this output:

> INFO: Found 33 device(s) ('ieeeAddr' keys) in 'database.db' and 25 device(s) in '[]'.
WARNING: Unable to find 0x00124b0018e1ea13 in '[]'
WARNING: Unable to find 0x7cb03eaa00a94f54 in '[]'
WARNING: Unable to find 0x00158d0002f8d805 in '[]'
WARNING: Unable to find 0x00158d000313c990 in '[]'
WARNING: Unable to find 0x00158d000316d4fb in '[]'
WARNING: Unable to find joinTime in '[]'
WARNING: Unable to find 0xd0cf5efffefeec6c in '[]'
WARNING: Unable to find 0x7cb03eaa0a00eaa7 in '[]'

Seems...not so good? :smile:  Is this what you were looking/askin for?

For example, I don't know why "joinTime" is given as the data for "ieeeAddr". What should I do with these entries? Is it safe to simply delete them in my database.db?

Thanks for all your hard work, really appreciated! :+1:

Koenkk, 2019-11-12:
Jointime is an old attribute of pre zigbee2mqtt 1.7. you should add the missing entries to configuration.yaml

CodeFinder2, 2019-11-12:
> Jointime is an old attribute of pre zigbee2mqtt 1.7. you should add the missing entries to configuration.yaml

I solved the issue(s) by deleting the line containing ieeeAddr="joinTime" in database.db and adding the missing entries to my devices.yaml (referenced from configuration.yaml).

Thanks!