Koenkk/zigbee2mqtt #2226: An "Invalid checksum" blocks zigbee2mqtt

demanu21, 2019-10-29:
### **What happened**
Every once in a while I get an Invalid checksum, which blocks all future messaging.

**The debug trace:**
2019-10-29T17:18:03.213Z serialport/bindings/poller received "readable"
2019-10-29T17:18:03.213Z serialport/binding-abstract read
2019-10-29T17:18:03.213Z serialport/bindings/unixRead Starting read
2019-10-29T17:18:03.213Z serialport/bindings/unixRead Finished read { bytesRead: 3,
  buffer:
   <Buffer fe 0e 61 02 02 00 02 06 03 90 15 34 01 02 00 00 00 00 da fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes
2019-10-29T17:18:03.214Z serialport/stream binding.read finished { bytesRead: 3 }
2019-10-29T17:18:03.214Z zigbee-herdsman:zStack:unpi:parser <-- [254,28,68]
2019-10-29T17:18:03.214Z zigbee-herdsman:zStack:unpi:parser --- parseNext [254,7,254,28,68]
2019-10-29T17:18:03.214Z serialport/stream _read reading { start: 1731, toRead: 63805 }
2019-10-29T17:18:03.214Z serialport/binding-abstract read
2019-10-29T17:18:03.214Z serialport/bindings/unixRead Starting read
2019-10-29T17:18:03.214Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
2019-10-29T17:18:03.214Z serialport/bindings/poller Polling for "readable"
2019-10-29T17:18:04.518Z serialport/bindings/poller received "readable"
2019-10-29T17:18:04.518Z serialport/binding-abstract read
2019-10-29T17:18:04.518Z serialport/bindings/unixRead Starting read
2019-10-29T17:18:04.518Z serialport/bindings/unixRead Finished read { bytesRead: 105,
  buffer:
   <Buffer fe 0e 61 02 02 00 02 06 03 90 15 34 01 02 00 00 00 00 da fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes
2019-10-29T17:18:04.518Z serialport/stream binding.read finished { bytesRead: 105 }
2019-10-29T17:18:04.518Z zigbee-herdsman:zStack:unpi:parser <-- [129,0,0,2,4,51,96,1,1,0,39,0,50,156,15,0,0,8,24,33,10,0,0,41,156,6,51,96,29,204,254,28,68,129,0,0,5,4,51,96,1,1,0,39,0,64,156,15,0,0,8,24,34,10,0,0,33,68,27,51,96,29,119,254,37,68,129,0,0,3,4,51,96,1,1,0,39,0,76,156,15,0,0,17,24,35,10,0,0,41,255,3,20,0,40,255,16,0,41,253,39,51,96,29,215]
2019-10-29T17:18:04.518Z zigbee-herdsman:zStack:unpi:parser --- parseNext [254,7,254,28,68,129,0,0,2,4,51,96,1,1,0,39,0,50,156,15,0,0,8,24,33,10,0,0,41,156,6,51,96,29,204,254,28,68,129,0,0,5,4,51,96,1,1,0,39,0,64,156,15,0,0,8,24,34,10,0,0,33,68,27,51,96,29,119,254,37,68,129,0,0,3,4,51,96,1,1,0,39,0,76,156,15,0,0,17,24,35,10,0,0,41,255,3,20,0,40,255,16,0,41,253,39,51,96,29,215]
2019-10-29T17:18:04.519Z zigbee-herdsman:zStack:unpi:parser --> error Error: Invalid checksum
    at Function.fromBuffer (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/frame.js:33:19)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:47:51)
    at Parser._transform (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:35:14)
    at Parser.Transform._read (_stream_transform.js:190:10)
    at Parser.Transform._write (_stream_transform.js:178:12)
    at doWrite (_stream_writable.js:415:12)
    at writeOrBuffer (_stream_writable.js:399:5)
    at Parser.Writable.write (_stream_writable.js:299:11)
    at SerialPort.ondata (_stream_readable.js:708:20)
    at SerialPort.emit (events.js:198:13)
2019-10-29T17:18:04.519Z zigbee-herdsman:zStack:znp:error Got unpi error Error: Invalid checksum
2019-10-29T17:18:04.519Z zigbee-herdsman:zStack:unpi:parser --- parseNext [1,1,0,39,0,50,156,15,0,0,8,24,33,10,0,0,41,156,6,51,96,29,204,254,28,68,129,0,0,5,4,51,96,1,1,0,39,0,64,156,15,0,0,8,24,34,10,0,0,33,68,27,51,96,29,119,254,37,68,129,0,0,3,4,51,96,1,1,0,39,0,76,156,15,0,0,17,24,35,10,0,0,41,255,3,20,0,40,255,16,0,41,253,39,51,96,29,215]
2019-10-29T17:18:04.520Z serialport/stream _read reading { start: 1836, toRead: 63700 }
2019-10-29T17:18:04.520Z serialport/binding-abstract read
2019-10-29T17:18:04.520Z serialport/bindings/unixRead Starting read
2019-10-29T17:18:04.520Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
2019-10-29T17:18:04.520Z serialport/bindings/poller Polling for "readable"

**### What did you expect to happen**
Survive...

### **How to reproduce it (minimal and precise)**
Nothing special to reproduce, just leave it running for a while. I have a [CC2530 coordinator with espeasy](https://www.zigbee2mqtt.io/how_tos/how_to_esp8266_with_cc2530.html), 3 "E1603/E1702 tradfri control outlets" (primarily used as routers), a bunch of xiaomi WSDCGQ11LM (temp/hum/press sensor) and a couple of xiaomi MCCGQ01LM (door/window sensors).
Only way I can restore communication is restart the docker container

### **Debug Info**
zigbee2mqtt version: I am running [latest-dev with socat in a docker](https://hub.docker.com/r/forepe/zigbee2mqtt-socat/) 
_=> Starting zigbee2mqtt version **1.6.0 (commit #cb3ec45)**_
CC253X firmware version: _**"softwareBuildID":"zStack12","dateCode":"20190608"**_

Koenkk, 2019-10-30:
Should be fixed in the latest dev branch, can you confirm?

demanu21, 2019-10-31:
I pulled the latest latest-dev docker (Starting zigbee2mqtt version 1.6.0 (commit #cb3ec45), I think it still occurs. Not sure if it's related, but now the invalid checksum is followed by an unhandled promise rejection (although I saw the unhandled promise somewhere else in de debug log as well).

`2019-10-31T09:45:33.876Z serialport/bindings/unixRead Finished read { bytesRead: 53,
  buffer:                                                              
   <Buffer fe 0e 61 02 02 00 02 06 03 90 15 34 01 02 00 00 00 00 da fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes
2019-10-31T09:45:33.876Z serialport/stream binding.read finished { bytesRead: 53 }
2019-10-31T09:45:33.876Z zigbee-herdsman:zStack:unpi:parser <-- [18,0,0,0,4,2,0,158,221,221,221,221,221,221,221,221,2,65,250,254,255,111,13,0,34,72,37,2,15,131,221,221,221,221,221,221,221,221,42,75,249,254,255,111,13,0,76,84,37,2,15,
2019-10-31T09:45:33.876Z zigbee-herdsman:zStack:unpi:parser --- parseNext [254,56,69,178,34,72,0,254,1,101,49,0,85,254,3,69,196,38,111,0,203,254,72,69,177,38,111,0,3,0,3,221,221,221,221,221,221,221,221,112,233,157,25,0,75,18,0,0,0,4,
2019-10-31T09:45:33.877Z zigbee-herdsman:zStack:unpi:parser --> error Error: Invalid checksum                                            
    at Function.fromBuffer (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/frame.js:33:19)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:47:51)                                                                                                  
    at Parser._transform (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:35:14)
    at Parser.Transform._read (_stream_transform.js:190:10)               
    at Parser.Transform._write (_stream_transform.js:178:12)
    at doWrite (_stream_writable.js:415:12)                                                                                                                                                            
    at writeOrBuffer (_stream_writable.js:399:5)
    at Parser.Writable.write (_stream_writable.js:299:11)                                                                                     
    at SerialPort.ondata (_stream_readable.js:708:20)
    at SerialPort.emit (events.js:198:13)                                          
2019-10-31T09:45:33.877Z zigbee-herdsman:zStack:znp:error Got unpi error Error: Invalid checksum
2019-10-31T09:45:33.877Z zigbee-herdsman:zStack:unpi:parser --- parseNext [2,65,250,254,255,111,13,0,34,72,37,2,15,131,221,221,221,221,221,221,221,221,42,75,249,254,255,111,13,0,76,84,37,2,15,111,150]
2019-10-31T09:45:33.877Z serialport/stream _read reading { start: 851, toRead: 64685 }
2019-10-31T09:45:33.877Z serialport/binding-abstract read                                                                                                              
2019-10-31T09:45:33.877Z serialport/bindings/unixRead Starting read
2019-10-31T09:45:33.877Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN                                     
2019-10-31T09:45:33.877Z serialport/bindings/poller Polling for "readable"
(node:40) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtLqiReq after 6000ms                                                                          
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)                                                                                                      
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)                                                                                                                            
    at Timer.processTimers (timers.js:223:10)
(node:40) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (reje
2019-10-31T09:45:43.579Z zigbee-herdsman:zStack:znp:SREQ --> ZDO - mgmtRtgReq - {"dstaddr":28454,"startindex":0}
2019-10-31T09:45:43.580Z zigbee-herdsman:zStack:unpi:writer --> frame [254,3,37,50,38,111,0,93]                                                                         
2019-10-31T09:45:43.580Z serialport/stream _write 8 bytes of data
`

Koenkk, 2019-10-31:
But after this error does it still continue?

demanu21, 2019-11-01:
Now I am more sure then before: It does not (that is: from my point of view, we could have a different definition of "continue"). Also: the unhandled promise I mentioned earlier is not related.

I see a invalid checksum in the debug log: 
- the "debug log" continues to produce 'bytesRead's (so yes, this one continues, but that was never the problem)
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/3797454/z2m.log)

- the "mqtt log" shows no more activity (so no, this on still halts).
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3797455/log.txt)

=> no data is published to my mqtt broker, no more data is inserted in my influxdb
=> restart of docker solves the issue. This is why I have a cronjob running which restarts the docker container every hour (which I disabled for now)
=> The auto soft reset also does not resolve the problem. (I had that on on default for some time, tried 3600 for a couple of weeks, now I have it at 5400. I am fine with a docker restart every hour, but that is merely a workaround)

Koenkk, 2019-11-01:
The fix is not in cb3ec45, please update to at least 7669682716f2f9d97b4670eb926d5f2ec903e470

demanu21, 2019-11-01:
Above comment still applies, after updating to https://github.com/Koenkk/zigbee2mqtt/commit/7669682716f2f9d97b4670eb926d5f2ec903e470

```
zigbee2mqtt:info  2019-11-01T11:03:15: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"7669682","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":false}'
```

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3797850/log.txt)
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/3797851/z2m.log)

Koenkk, 2019-11-01:
Turns out there was a second bug, also fixed in latest dev branch.

demanu21, 2019-11-02:
And now I think is is fixed => 3 Invald checksum, nothing blocked.

For completeness:
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/3800473/z2m.log)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3800474/log.txt)

Koenkk, 2019-11-02:
Awesome!

closed 2019-11-02 by Koenkk

demanu21, 2019-11-02:
You are awesome ;-)