Koenkk/zigbee2mqtt #2227: No network route (205) to LLKZMK11LM

LCerebo, 2019-10-29:
# Bug Report
## What happened
As mentioned in #2147 after updating the zigbee2mqtt container to the latest-dev version all my Xiaomi Relay (LLKZMK11LM) stop responding with the message 'Error: Data request failed with error: 'No network route' (205)'.
As required you can find the container log and the sniffed traffic here:

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3785534/log.txt)
[Traffic.zip](https://github.com/Koenkk/zigbee2mqtt/files/3785540/Traffic.zip)

Let me know if you need further information

## Debug Info
zigbee2mqtt version: 1.6.0 (commit #344cd60)
CC253X firmware version: 20190608

Koenkk, 2019-10-30:
When receiving '205 no network route' errors with the default firmware it can help to switch to the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing.

demanu21, 2019-10-30:
Same occasional behavior with "E1603/E1702 tradfri control outlets"

zigbee2mqtt version 1.6.0 (commit #cb3ec45)
CC253X firmware version: 20190608

pcb1962, 2019-11-06:
Switching to source routing firmware worked for me. I lost access to my IKEA outlets after upgrading to dev version, source routing firmware fixed that.

LCerebo, 2019-11-25:
I've switched to latest stable release (1.7.0) and source routing firmware as suggested and now it seems to work fine.

closed 2019-11-25 by LCerebo