Koenkk/zigbee2mqtt #2236: Not Only Mi Aquara Door Sensor - Missed updates when contact toggled quickly

alexsahka, 2019-10-30:
My setup: Latest Dev koenkk/zigbee2mqtt DOCKER, CC2531 with the coordinator/Z-Stack_3.0.x/bin/CC2531_20190425, tried before with the same result: (coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608)

I'm new to Z2M, and been playing with it since Feb 2019.
I used the Xiaomi/Aquara gateway with Node-red for a long time without any problem.

As soon as I switched to Z2M, I start noticing that I'm missing some events from Aquara window/Door Sensors, randomly (more often, when contact toggled quickly).
I'm tracking every contact open/close with MQTT/NodeRed/InfluxDB/Grafana for testing purposes.

I can't figure out the problem (Even tried this solution: https://github.com/Koenkk/zigbee2mqtt/issues/1511) I decided to switch back to Xiaomi gateway/Node-red/MQTT route and Immediately no more missing events, everything workout smoothly as before.
So I decided to keep doors/windows sensors this way.

Around Summertime I bought Yale Z2M not supported lock and with Koen Kanters help it is working fine with the Z2M now.
But again I start noticing missing events.
I checked the Z2M docker log file for missing events and these events are missing in the log file as well.

How can I start troubleshooting this problem?

Koenkk, 2019-10-30:
The first thing to check if the devices actually send the message. This can be done using sniffing: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html.

alexsahka, 2019-10-30:
This happened randomly, and Aquara Windows/Doors Sensors working without losing any events 
 when I move it form Z2M/MQTT/NodeRed/InfluxDB/Grafana to Xiaomi_gateway/NodeRed/InfluxDB/Grafana for last 5 month.

Ton1965, 2019-10-31:
I used to have a similar problem with Xiaomi door sensor when some events got lost with z2m. Turned out to be too low link quality on connection with the coordinator. Added a router and it is stable for several months already without a single event missed. The link quality shown was ~30 but sometimes fell to ~15. That time I thought it was enough. But it looks like the door sensor somehow lowers the power when transmitting several messages in quick succession. 
My Xiaomi gateway was much closer to the sensor so it did not miss any events. Another thing is the gateway probably has better sensitivity due to its size.

alexsahka, 2019-10-31:
I added CC2530 router after pairing Yale lock with CC2531 broker and the network map is showing that Yale Lock is still connected to the broker directly. How can I force Yale Lock to connect to the CC2530 router? 
[image]

Koenkk, 2019-10-31:
Try to re-pair it, also do you use a usb extension cable for your cc2531? This also helps to increase linkquality

alexsahka, 2019-10-31:
Yes, I'm using 8' USB extension cable for the cc2531, and it is up in the air 8' above floor level.

alexsahka, 2019-10-31:
After re-pair the lock is still missing updates when locked/unlocked quickly (under 4-sec)
If I do UNLOCK  wait for at least 4-sec then LOCK it is reporting properly -OFF -ON
If the wait time between lock state change is under 4 sec, the lock is always missing state.
[image]

Koenkk, 2019-10-31:
So can you sniff what happens in that case? (https://github.com/Koenkk/zigbee2mqtt/issues/2236#issuecomment-548147910).

alexsahka, 2019-10-31:
This is will be a learning curve, I will try this later. I don't know yet but probably I need to buy one more CC2531 for this.

Koenkk, 2019-10-31:
If only the router is offline, everything that is still in range of the CC2531 should work.

closed 2020-01-06 by stale[bot]