Koenkk/zigbee2mqtt #2237: OSRAM Smart+ (AB3257001NJ) button does not signal state

Bartole, 2019-10-30:
# Bug Report
## What happened
I registered two OSRAM Smart+ devices without any problem and can switch them as expected using Domoticz. Using the button on the devices does not change the state in Domoticz and it does not produce a message in the log of Zigbee2MQTT.

## What did you expect to happen
I expected the state to change in Domoticz accordingly as I press the button on the device and see a message appearing in the Zigbee2MQTT log.

## How to reproduce it (minimal and precise)
Press the button on the Smart+ device

## Debug Info
zigbee2mqtt version: version 1.6.0 (commit #e26ad2a)
CC253X firmware version: 20190608

Bartole, 2019-10-30:
Second reboot fixed the issue. I don't understand why, but it now works both ways.

Thx

Koenkk, 2019-10-30:
Probably because it now configured it correctly, this should be improved in the zigbee2mqtt latest dev (soon zigbee2mqtt 1.7).

closed 2019-10-30 by Koenkk