Koenkk/zigbee2mqtt #2238: Aqara Devices Problems

soloam, 2019-10-31:
I have several devices in my network, from xiaomi, aqara, Ikea, osram. All work ok, except the aqara! I keep gettin devices from aqara offline, the PIR sensors, the Temperature sensors and the siwtches! I only have this problem in the aqara ones! I never had one single problem with the old xiaomi switches, the aqara versions I have to repair them once in a while! The PIR sensor keep losing the pairing, and the temperature sensors the same.

Out of nowhere I get a offline, and I have to repair them, most of them are some meters away from the coordinator. This only happens to me on the aqara version of the devices.

Any one with similar problem? What can I do to try do debug this?

`info  2019-10-31T13:06:54: Logging to directory: 'data/log'
info  2019-10-31T13:06:54: Starting zigbee2mqtt version 1.6.0 (commit #4164f01)
info  2019-10-31T13:06:54: Starting zigbee-herdsman...
info  2019-10-31T13:06:58: zigbee-herdsman started
info  2019-10-31T13:06:58: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}'
info  2019-10-31T13:06:58: Currently 36 devices are joined:`

Thank you

bochelork, 2019-11-01:
Same here with Mijia, but no need to repair. Sometimes a door sensor is offline for half a day, then it suddenly returns. Sometimes I need to press the pair-button om a temp/humid sensor once and it's back in the network. 

Do you see any messages in zigbee2mqtt when you just press the pair-button/pinhole shortly?

soloam, 2019-11-01:
Yes, sometimes it only needs to press the repair button once! O don't know if this is a problem with the sensor or with the zigbee2mqtt, but for the time that I used the xiaomi gateway, not long, I never needed to do this!

Koenkk, 2019-11-01:
are you running the zigbee2mqtt latest dev?

soloam, 2019-11-01:
version 1.6.0 (commit #4164f01) not the latest! I can try to update it to see if I see any improvement! Any work done regarding this in the latest builds @Koenkk?

Thank you for your work on this

Koenkk, 2019-11-02:
Yes, see #1888

closed 2020-01-08 by stale[bot]