Koenkk/zigbee2mqtt #2240: Salus SP600 not detected at all for pairing

gausie, 2019-10-31:
# Bug Report
## What happened

Using hassio-zigbee2mqtt (edge), while other Zigbee devices will connect, the SP600 doesn't register in any way and will not pair. I have tried the various reset states for the SP600, holding the button as you plug it in to trigger the three colour reset, and holding the button down for 10 seconds to unpair.

## What did you expect to happen

The SP600 to pair with zigbee2mqtt!

## How to reproduce it (minimal and precise)

Set `permit_join` to true and restart zigbee2mqtt, plug in an SP600 next to the CC2531. Wait!

## Debug Info
```
Debug information
-----
-----
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": true,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://core-mosquitto:1883",
    "user": "mqtt",
    "password": "mqtt"
  },
  "serial": {
    "port": "/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE8A09E-if00"
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1.0,
    "initialdelay": 1.0,
    "options": "-d -d",
    "log": false
  }
}Debug information end
./run.sh: line 24: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2019-10-31T15:57:04: PM2 log: Launching in no daemon mode
2019-10-31T15:57:04: PM2 log: App [npm:0] starting in -fork mode-
2019-10-31T15:57:05: PM2 log: App [npm:0] online
> zigbee2mqtt@1.6.0 start /app
> node index.js
zigbee2mqtt:info  2019-10-31T15:57:15: Logging to directory: '/share/zigbee2mqtt/log/2019-10-31.15-57-14'
zigbee2mqtt:info  2019-10-31T15:57:16: Starting zigbee2mqtt version 1.6.0 (commit #36c7a27e6c632e0450db18279fc46a1d01365579)
zigbee2mqtt:info  2019-10-31T15:57:16: Starting zigbee-herdsman...
zigbee2mqtt:info  2019-10-31T15:57:19: zigbee-herdsman started
zigbee2mqtt:info  2019-10-31T15:57:19: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:info  2019-10-31T15:57:19: Currently 1 devices are joined:
zigbee2mqtt:info  2019-10-31T15:57:19: 0x00158d0004458653 (0x00158d0004458653): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
zigbee2mqtt:warn  2019-10-31T15:57:19: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2019-10-31T15:57:19: Allowing new devices to join.
zigbee2mqtt:warn  2019-10-31T15:57:19: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2019-10-31T15:57:19: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2019-10-31T15:57:19: Connecting to MQTT server at mqtt://core-mosquitto:1883
zigbee2mqtt:info  2019-10-31T15:57:21: Connected to MQTT server
zigbee2mqtt:info  2019-10-31T15:57:21: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2019-10-31T15:57:21: MQTT publish: topic 'zigbee2mqtt/0x00158d0004458653', payload '{"contact":true,"linkquality":0,"battery":100,"voltage":3125}'
zigbee2mqtt:info  2019-10-31T15:57:21: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"36c7a27e6c632e0450db18279fc46a1d01365579","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
```

jameswarner62, 2019-10-31:
In case it helps, my SP600 was unpaired from factory, and joined instantly without being at all close to the coordinator. I've not tried a reset condition. Are you sure you don't have any other nearby zigbee networks accepting it?

gausie, 2019-10-31:
The only thing that possibly could have accepted it is the Hue bridge, but I disconnected it at the mains and repeated the process with no luck

jameswarner62, 2019-10-31:
Just re-paired one of mine to check (held down button for 10 seconds) and it rejoined fine. Have you just got the one to test? can it pair with the hue hub?

gausie, 2019-10-31:
I've got two and they both seem to be doing the same thing. I did try to
pair with the Hue hub as well but it doesn't see them. I might just have to
return them and get the Hive ones instead I guess!

gausie, 2019-11-03:
Ok the Hive plug has arrived now and also isn't detected by my setup. What could cause some ZigBee devices to be added no problem but for others to not show up at all in the logs, as if they didn't exist?

Koenkk, 2019-11-03:
- Is your CC2531 connected via an extension cable? (to rule out interference)
- When trying to pair, can you share the log when running zigbee2mqtt with `DEBUG=zigbee-herdsman:controller* npm start`?

gausie, 2019-11-03:
No extension cord and I'm running it through hassio, so I can't do that
super easily.

I'll test it out with an extension and from my laptop so I can run
zigbee2mqtt directly

Thanks!

Koenkk, 2019-11-03:
You can, for hassio edge addon just set `"zigbee_herdsman_debug": true` in the config.

gausie, 2019-11-04:
Ah, well got it set up on my laptop before I saw that! Anyway, no appreciable difference in the logs as far as I can see with the debug environment variable set

```

zigbee2mqtt:info  2019-11-04T11:04:43: Logging to directory: '/media/gausie/Shared (ext4)/Projects/zigbee2mqtt/data/log/2019-11-04.11-04-43'
zigbee2mqtt:info  2019-11-04T11:04:43: Starting zigbee2mqtt version 1.6.0 (commit #2fb4e20)
zigbee2mqtt:info  2019-11-04T11:04:43: Starting zigbee-herdsman...
zigbee2mqtt:info  2019-11-04T11:04:45: zigbee-herdsman started
zigbee2mqtt:info  2019-11-04T11:04:45: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:info  2019-11-04T11:04:45: Currently 1 devices are joined:
zigbee2mqtt:info  2019-11-04T11:04:45: 0x00158d0004458653 (0x00158d0004458653): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
zigbee2mqtt:warn  2019-11-04T11:04:45: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2019-11-04T11:04:45: Allowing new devices to join.
zigbee2mqtt:warn  2019-11-04T11:04:45: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2019-11-04T11:04:45: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2019-11-04T11:04:45: Connecting to MQTT server at mqtt://hassio.local:1883
zigbee2mqtt:info  2019-11-04T11:04:45: Connected to MQTT server
zigbee2mqtt:info  2019-11-04T11:04:45: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2019-11-04T11:04:45: MQTT publish: topic 'zigbee2mqtt/0x00158d0004458653', payload '{"battery":100,"voltage":3085,"contact":false,"linkquality":70}'
zigbee2mqtt:info  2019-11-04T11:04:45: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"2fb4e20","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
```

gausie, 2019-11-04:
I restarted everything and expanded the `debug` namespace to all of zigbee-herdsman and got this. Could it be relevant?

```
  zigbee-herdsman:controller:log Permit joining +3m
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228] +3m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82] +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [] - 82 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [] - 240 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:controller:log Permit joining +3m
(node:14043) UnhandledPromiseRejectionWarning: Error: Cannot request when znp has not been initialized yet
    at Znp.request (/media/gausie/Shared (ext4)/Projects/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:173:19)
    at ZStackAdapter.<anonymous> (/media/gausie/Shared (ext4)/Projects/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:124:32)
    at Generator.next (<anonymous>)
    at /media/gausie/Shared (ext4)/Projects/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/media/gausie/Shared (ext4)/Projects/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:4:12)
    at Object.queue.execute [as func] (/media/gausie/Shared (ext4)/Projects/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:122:44)
    at Queue.<anonymous> (/media/gausie/Shared (ext4)/Projects/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/queue.js:29:46)
    at Generator.next (<anonymous>)
    at /media/gausie/Shared (ext4)/Projects/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/queue.js:8:71
(node:14043) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 54)
```

gausie, 2019-11-05:
Ran it again today and I no longer get that error. I really have no idea why this type of device just doesn't work for me. Could it be an issue with my CC2531?

Koenkk, 2019-11-05:
Could it be that it pairs to another network? Are any other zigbee devices on which are in a different network?

gausie, 2019-11-05:
It's still in pairing mode (flashing orange button) so I don't think that's the case. But I unplugged the Hue bridge and started again with no luck.

Koenkk, 2019-11-05:
I have no clue what is going on honestly. If they also dont pair with the hue bridge it would indicate that something else (really strange) is going on.

gausie, 2019-11-05:
It was two separate SP600 and two separate Hive Active Plugs so it seems to
not just be specific faulty hardware. So weird!

Koenkk, 2019-11-05:
A last thing to try would be to pair it at a different location (e.g. different house), perhaps there is some heavy interference.

gausie, 2019-11-19:
I ended up buying a tasmota compatible smart plug and abandoning doing it through zigbee

closed 2019-11-19 by gausie

elmopl, 2019-11-29:
I seem to have the same problem.
I have CC2531 connected to Raspberry Pi with "Z-Stack_Home_1.2 (default)" (from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator).

Today I received my Salus SP600 Smart Plug and when I plug it in it seems to makes no trace on output from zigbee2mqtt.

I have paired Xiaomi button, and Ikea motion sensor so far without any problems and they work fine.

I run zigbee2mqtt through systemctl (not that it would make much difference) and I have also tried it with:

    [Service]
    Environment=DEBUG=zigbee-herdsman:*

which did add more levels of debugging, however there is no trace related to plugging in my plug.

I have no other coordinators at home other the one running on Raspberry Pi.

The plug seems to go through 3 stages of flashing when plugged in:

 - flashing green
 - flashing 3 x red -> pause -> 3 x red -> pause ...
 - after a while it changes to continuous flashing of red

Do you have any suggestions on how I could try checking what is going on with that plug? Is there some program (and firmware?) that would use that CC2531 to scan and debug devices trying to pair? (I have a second CC2531 dongle which I could easily flash with something else or use in parallel to the one used by zigbee2mqtt).

elmopl, 2019-11-30:
I did more checks. I used CC2531 with firmware that comes with packet sniffer from Texas Instrument.

I moved my laptop with dongle and my Salus SP600 plug to a furthest point from all other devices to be able to reduce amount of noise in traffic.

As far as I can see I was consistently getting a packets (which seemed to be also very much in sync with flashing of the plug, every time a sequence of 3 red flashes started a packet appeared):

```
+----------------------------------------------------+----------------- - - -
|     Packet sniffer frame header                    |
+----+-------------+-------------------------+-------+
|info| Packet nbr. | Time stamp              | Length|  Packet data
+----+-------------+-------------------------+-------+----------------- - - -
| 13 | 08 00 00 00 | F0 B6 F8 B2 01 00 00 00 | 0B 00 | 0A 03 08 B8 FF FF FF FF 07 34 BC
+----+-------------+-------------------------+-------+----------------- - - -
| 13 | 09 00 00 00 | D0 E3 84 BC 01 00 00 00 | 0B 00 | 0A 03 08 C8 FF FF FF FF 07 34 BC
+----+-------------+-------------------------+-------+----------------- - - -
| 13 | 0B 00 00 00 | 6C 9A 0E C6 01 00 00 00 | 0B 00 | 0A 03 08 D8 FF FF FF FF 07 34 BC
+----+-------------+-------------------------+-------+----------------- - - -
| 13 | 0C 00 00 00 | AC 9F 9A CF 01 00 00 00 | 0B 00 | 0A 03 08 E8 FF FF FF FF 07 3C BC
+----+-------------+-------------------------+-------+----------------- - - -
```

Unfortunately I know nothing about ZigBee so it tells me nothing.
I guess next step for me is to read up on what those packets mean.

elmopl, 2019-11-30:
More investigation.
Turning the CC2531 acting as controller I can see that Beacon request is not followed by anything.
That would suggest that CC2531/zigbee2mqtt is actually responding to beacon request.

I am not sure what is missing from the response that stops SP600 Smart Plug from connecting.