Koenkk/zigbee2mqtt #2243: OSRAM Smart + Crazy

jasimancas, 2019-10-31:
I have a Osram Smart + plug and randomd turn off/on/off/on.

This is the log:

`zigbee2mqtt:debug 2019-10-31T22:50:32: Received Zigbee message from 'Termo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 3 with groupID 0
,zigbee2mqtt:debug 2019-10-31T22:50:28: Device 'Termo' announced itself
,zigbee2mqtt:info  2019-10-31T22:50:32: MQTT publish: topic 'zigbee2mqtt/Termo', payload '{"state":"OFF","linkquality":42,"last_seen":1572562232686,"device":{"friendlyName":"Termo","ieeeAddr":"0x7cb03eaa0a0129ec","networkAddress":51488,"type":"Router","manufacturerID":48042,"manufacturerName":"OSRAM","powerSource":"Mains (single phase)","applicationVersion":12,"stackVersion":2,"zclVersion":1,"hardwareVersion":1,"dateCode":"20140331DEOS****","softwareBuildID":"V1.04.12"}}'
,zigbee2mqtt:debug 2019-10-31T22:50:58: Device 'Termo' announced itself
,zigbee2mqtt:info  2019-10-31T22:50:32: MQTT publish: topic 'zigbee2mqtt/Termo', payload '{"state":"OFF","linkquality":39,"last_seen":1572562232692,"device":{"friendlyName":"Termo","ieeeAddr":"0x7cb03eaa0a0129ec","networkAddress":51488,"type":"Router","manufacturerID":48042,"manufacturerName":"OSRAM","powerSource":"Mains (single phase)","applicationVersion":12,"stackVersion":2,"zclVersion":1,"hardwareVersion":1,"dateCode":"20140331DEOS****","softwareBuildID":"V1.04.12"}}'
,zigbee2mqtt:info  2019-10-31T22:51:05: MQTT publish: topic 'zigbee2mqtt/Termo', payload '{"state":"OFF","linkquality":39,"last_seen":1572562265635,"device":{"friendlyName":"Termo","ieeeAddr":"0x7cb03eaa0a0129ec","networkAddress":51488,"type":"Router","manufacturerID":48042,"manufacturerName":"OSRAM","powerSource":"Mains (single phase)","applicationVersion":12,"stackVersion":2,"zclVersion":1,"hardwareVersion":1,"dateCode":"20140331DEOS****","softwareBuildID":"V1.04.12"}}'
,zigbee2mqtt:debug 2019-10-31T22:51:05: Received Zigbee message from 'Termo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 3 with groupID 0
,zigbee2mqtt:debug 2019-10-31T22:51:28: Device 'Termo' announced itself
,zigbee2mqtt:info  2019-10-31T22:51:34: MQTT publish: topic 'zigbee2mqtt/Termo', payload '{"state":"OFF","linkquality":42,"last_seen":1572562294523,"device":{"friendlyName":"Termo","ieeeAddr":"0x7cb03eaa0a0129ec","networkAddress":51488,"type":"Router","manufacturerID":48042,"manufacturerName":"OSRAM","powerSource":"Mains (single phase)","applicationVersion":12,"stackVersion":2,"zclVersion":1,"hardwareVersion":1,"dateCode":"20140331DEOS****","softwareBuildID":"V1.04.12"}}'
,zigbee2mqtt:info  2019-10-31T22:51:05: MQTT publish: topic 'zigbee2mqtt/Termo', payload '{"state":"OFF","linkquality":42,"last_seen":1572562265653,"device":{"friendlyName":"Termo","ieeeAddr":"0x7cb03eaa0a0129ec","networkAddress":51488,"type":"Router","manufacturerID":48042,"manufacturerName":"OSRAM","powerSource":"Mains (single phase)","applicationVersion":12,"stackVersion":2,"zclVersion":1,"hardwareVersion":1,"dateCode":"20140331DEOS****","softwareBuildID":"V1.04.12"}}'
,zigbee2mqtt:debug 2019-10-31T22:51:34: Received Zigbee message from 'Termo', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 3 with groupID 0
,zigbee2mqtt:info  2019-10-31T22:51:34: MQTT publish: topic 'zigbee2mqtt/Termo', payload '{"state":"OFF","linkquality":39,"last_seen":1572562294538,"device":{"friendlyName":"Termo","ieeeAddr":"0x7cb03eaa0a0129ec","networkAddress":51488,"type":"Router","manufacturerID":48042,"manufacturerName":"OSRAM","powerSource":"Mains (single phase)","applicationVersion":12,"stackVersion":2,"zclVersion":1,"hardwareVersion":1,"dateCode":"20140331DEOS****","softwareBuildID":"V1.04.12"}}'
,zigbee2mqtt:debug 2019-10-31T22:52:28: Device 'Termo' announced itself
`

## Debug Info
zigbee2mqtt version: latest dev
CC253X firmware version: latest source_routing

Koenkk, 2019-11-01:
Probably an other device in your network is controlling it, via sniffing you can find out which (see zigbee2mqtt docs).

jasimancas, 2019-11-01:
Ok, i buy another 2531, flash with sniffer firm and try to see what happend.

jasimancas, 2019-11-08:
Hi again, I have the Sniffer USB ON, but i cant obtain the network key.

Last year I changed the network key, but when I restart the numbers I have in configuration.yaml are not the ones I set.

How can I get it?

Koenkk, 2019-11-08:
The network key can also be sniffed: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html#3-sniffing-traffic

jasimancas, 2019-11-08:
Yes, i repairing a device, but i dont see Device Announcement….. in Info

closed 2019-11-23 by jasimancas