Koenkk/zigbee2mqtt #2244: Can't set Hue and Saturation on Philips LightStrip LST002

TilmanK, 2019-11-01:
# Bug Report
I connected a LightStrip LST002 to my Zigbee network. The interview was successful as soon as I moved the device to the Coordinator (it was not trough my routers). 

This is the device in my db-file:
`{"id":15,"type":"Router","ieeeAddr":"0x00178801044a5ca1","nwkAddr":44994,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LST002","epList":[11,242],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LST002","manufacturerName":"Philips","powerSource":1,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170323","swBuildId":"1.24.0_r20239"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170323","swBuildId":"1.24.0_r20239","zclVersion":1,"interviewCompleted":true,"meta":{}}`

I'm now trying to set hue and saturation, both fails.

**Hue**
I try to set Hue by sending the following message via MQTT:
`{
"hue": 100
}`

On the logs I'm receiving the following error message

`Nov  1 08:34:07 NAS-Serenity npm[6537]: #033[31mzigbee2mqtt:error#033[39m 2019-11-01T07:34:07: No converter available for 'hue' (100)`

I don't get anything back on MQTT.

**Saturation**
The message to MQTT is as follows:
`{
"color": {"saturation": 100}
}`

I'm getting the same message in logs as well as on MQTT:

`{
  "type" : "zigbee_publish_error",
  "message" : "Publish 'set' 'color' to 'Bett links' failed: 'TypeError: Cannot read property 'color' of null'",
  "meta" : {
    "friendly_name" : "Bett links"
  }
}`

## Debug Info
zigbee2mqtt version:   "version" : "1.6.0", "commit" : "7669682"
CC2531 firmware version: latest source routing

Thanks again, for your help!

Cheers,
Tilman

Koenkk, 2019-11-01:
Setting hue and saturation is not supported yet so I consider this as a feature request.

TilmanK, 2020-01-04:
I was wondering since this issue didn't get any attention by others, is there any other way setting the color? Is no one trying to color their lights? :)

I think that x and y are currently supported, but AFAIK it's pretty complicated to convert hue and saturation to x and y. I'd definitely need hue and saturation because of the HomeKit integration in node red.

Koenkk, 2020-01-07:
This should now be possible in the latest dev branch, see `hue` and `saturation` in https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameset

closed 2020-01-07 by Koenkk

TilmanK, 2020-01-07:
You made an already awesome day perfect. Thanks! ;)