Koenkk/zigbee2mqtt #2245: There isn't a way to reach zigbee2mqtt.io home page

RyuzakiKK, 2019-11-01:
Issues are disabled in the zigbee2mqtt.io repo, so I'm posting this here.

Problem:
If you open a page in the site zigbee2mqtt.io you are unable to reach its home page because it's not linked anywhere.

Steps to reproduce:
Go to https://www.zigbee2mqtt.io/how_tos/how_to_secure_network.html

Expected result:
A link to https://www.zigbee2mqtt.io somewhere in that page.
In the header the big `zigbee2mqtt.io` title is not clickable and `View on Github` points to https://github.com/Koenkk/zigbee2mqtt.io
In the footer the clickable `zigbee2mqtt.io` points instead to https://github.com/Koenkk/zigbee2mqtt.io

carldebilly, 2019-11-01:
I agree it's a problem I often have while navigating on this site.

Koenkk, 2019-11-01:
It would be great if someone with Jekyll knowledge can help improving the site.

LukeHandle, 2019-11-04:
Never touched Jekyll, but appears to be a case of overriding the the theme - copy in the original template and update the needed bits:
https://github.com/pages-themes/cayman#layouts

Koenkk, 2019-11-09:
Fixed, you can now click on *zigbeem2qtt.io* to navigate back to the home page.

closed 2019-11-09 by Koenkk