Koenkk/zigbee2mqtt #2246: Device '' does not exist - bulbs not working

moelski, 2019-11-01:
# Bug Report
## What happened
I connected 3 bulbs to my CC2531: 
1x LED1624G9 - IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (Router)
2x LED1837R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)

It seems that they are connected but there is no response from the bulbs. No messages coming in and the mqtt Server didn´t get any message, too. 

This is the pairing part of one bulb: 
```
zigbee2mqtt:info  2019-11-01T10:16:08: Starting interview of '0x14b457fffe4178e5'
zigbee2mqtt:info  2019-11-01T10:16:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x14b457fffe4178e5"}}'
zigbee2mqtt:info  2019-11-01T10:16:18: Successfully interviewed '0x14b457fffe3e959b', device has successfully been paired
zigbee2mqtt:info  2019-11-01T10:16:18: Device '0x14b457fffe3e959b' is supported, identified as: IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1837R5)
zigbee2mqtt:info  2019-11-01T10:16:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x14b457fffe3e959b","model":"LED1837R5","vendor":"IKEA","description":"TRADFRI LED bulb GU10 400 lumen, dimmable","supported":true}}'

```

Is the pairing done or is there anything missing? 

The only thing I got as a response is this: 
`zigbee2mqtt:error 2019-11-01T10:20:39: Device '' does not exist
`

I´m not sure if this belongs to my bulbs or if it is another problem. 

## What did you expect to happen
See some reports from the bulbs and some states in MQTT...

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: latest DEV Version
CC253X firmware version: latest BIN

moelski, 2019-11-01:
The only thing I ever see from the bulbs is this log message: 
`zigbee2mqtt:debug 2019-11-01T11:18:54: Device '0x000d6ffffe07d478' announced itself
`

Koenkk, 2019-11-01:
Looks that the device has bee paired successfully, normally bulbs don't send any status update. Try controlling it, documentation: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttdevice_idset

Closing this as it doesn't seem to be an issue.

closed 2019-11-01 by Koenkk

moelski, 2019-11-02:
Hi Koenkk, 
thx for your feedback. But it doesn´t work. 
First of all there is no entry created at the MQTT Server if the bulb is paired. Is this the default behaviour? 
And even if I send data to the mqtt server there is no effect on the lamp: 
```
mosquitto_pub -h 192.168.30.6 -p 1883 -t "zigbee2mqtt/Lampe_RGB" -m '{"state": "ON",  "brightness": 25}'
mosquitto_pub -h 192.168.30.6 -p 1883 -t "zigbee2mqtt/Lampe_RGB/state" -m 'OFF'
mosquitto_pub -h 192.168.30.6 -p 1883 -t "zigbee2mqtt/Lampe_RGB" -m '{"state": "ON",  "brightness": 25, "hue": 360}'
```
These are my MQTT Test messages, but nothing happens. And zigbee2mqtt didn´t see any messages from Lampe_RGB, too. 

Did z2m subscribe to all topics from zigbee2mqtt at the mqtt server? 

Take a look at the logfile: 
```
zigbee2mqtt:info  2019-11-02T05:27:52: Currently 3 devices are joined:
zigbee2mqtt:info  2019-11-02T05:27:52: Cube (0x00158d00029aca6e): MFKZQ01LM - Xiaomi Mi/Aqara smart home cube (EndDevice)
zigbee2mqtt:info  2019-11-02T05:27:52: Lampe_RGB (0x000d6ffffe07d478): LED1624G9 - IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (Router)
zigbee2mqtt:info  2019-11-02T05:27:52: 0x000d6ffffe193cf5 (0x000d6ffffe193cf5): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
zigbee2mqtt:warn  2019-11-02T05:27:52: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2019-11-02T05:27:52: Allowing new devices to join.
zigbee2mqtt:warn  2019-11-02T05:27:52: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2019-11-02T05:27:52: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2019-11-02T05:27:52: Connecting to MQTT server at mqtt://192.168.30.6
zigbee2mqtt:info  2019-11-02T05:27:52: Connected to MQTT server
zigbee2mqtt:info  2019-11-02T05:27:52: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2019-11-02T05:27:52: MQTT publish: topic 'zigbee2mqtt/Cube/battery', payload '80'
zigbee2mqtt:info  2019-11-02T05:27:52: MQTT publish: topic 'zigbee2mqtt/Cube/voltage', payload '2965'
zigbee2mqtt:info  2019-11-02T05:27:52: MQTT publish: topic 'zigbee2mqtt/Cube/linkquality', payload '89'
zigbee2mqtt:info  2019-11-02T05:27:52: MQTT publish: topic 'zigbee2mqtt/Cube/side', payload '0'
zigbee2mqtt:info  2019-11-02T05:27:52: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"7669682","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
zigbee2mqtt:debug 2019-11-02T05:28:01: Received MQTT message on 'zigbee2mqtt/bridge/networkmap' with data ''
zigbee2mqtt:debug 2019-11-02T05:28:01: Received MQTT message on 'zigbee2mqtt/bridge/networkmap/routes' with data ''
zigbee2mqtt:debug 2019-11-02T05:28:23: Received MQTT message on 'zigbee2mqtt/bridge/group/remove_all' with data ''
zigbee2mqtt:error 2019-11-02T05:28:23: Device '' does not exist
zigbee2mqtt:debug 2019-11-02T05:28:48: Received MQTT message on 'zigbee2mqtt/bridge/networkmap' with data ''
zigbee2mqtt:debug 2019-11-02T05:28:48: Received MQTT message on 'zigbee2mqtt/bridge/networkmap/routes' with data ''
zigbee2mqtt:debug 2019-11-02T05:28:48: Received MQTT message on 'zigbee2mqtt/bridge/group/remove_all' with data ''
zigbee2mqtt:error 2019-11-02T05:28:48: Device '' does not exist
zigbee2mqtt:debug 2019-11-02T05:32:50: Saving state to file /opt/zigbee2mqtt/data/state.json
```
There is no message for the Lampe_RGB.

Koenkk, 2019-11-02:
You are using the wrong topic ,should be `zigbee2mqtt/Lampe_RGB/set`

moelski, 2019-11-02:
You made my day. THX alot :-)