Koenkk/zigbee2mqtt #2248: home assistant integration: only json

bartplessers, 2019-11-01:
# Bug Report
## What happened
According to documentation, you can use 2 methods to integrate zigbee2mqtt in home assistant
https://www.zigbee2mqtt.io/integration/home_assistant.html

However:
I made 2 automations:

this does NOT work
```
- id: '0xd0cf5efffe307960_toggle_hass'
  alias: (R01) Remote IKEA Toggle button (HASS)
  trigger:
  - entity_id: sensor.0xd0cf5efffe307960_click
    platform: state
    to: toggle
  condition: []
  action:
  - data:
      entity_id: light.0xd0cf5efffe0bb71a_light
    service: light.toggle
```

But his this DOES work correctly:

```
- id: '0xd0cf5efffe307960_toggle_mqtt'
  alias: (R01) Remote IKEA  Toggle button (MQTT)
  trigger:
  - platform: mqtt
    topic: zigbee2mqtt/0xd0cf5efffe307960
  condition:
  - condition: template
    value_template: '{{ "toggle" == trigger.payload_json.action}}'
  action:
  - data:
      entity_id: light.0xd0cf5efffe0bb71a_light
    service: light.toggle
```
So for some reason, I have to use the json notation, which is OK, but more complex and prone to errors…

## What did you expect to happen
Both automations should work

## Debug Info
Zigbee2mqtt version 1.6.0
on bare-metal RPI3
CC2531_USB-stick (using an USB extension cable)
firmware version: CC2531_DEFAULT_20190608.zip

Koenkk, 2019-11-01:
` - entity_id: sensor.0xd0cf5efffe307960_click` should be ` - entity_id: sensor.0xd0cf5efffe307960_action`

closed 2019-11-01 by Koenkk

bartplessers, 2019-11-02:
thanx again @Koenkk !

Xavi Moreno pointed me also in the good direction at
https://community.home-assistant.io/t/please-help-automation-for-ikea-tradfri-remote-control-e1524-e1810/142707/3