Koenkk/zigbee2mqtt #2250: Updating bindings for devices with a '/' in the name doesn't work.

sjorge, 2019-11-01:
https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/deviceBind.js#L24

First issue seems to be the mqtt subscription, as it currently accepts + or +/+ wild cards so it will match a device or device/endpoint, if the device name has a '/' it, it will no longer work.

Not sure we can get this to work though, as there is also getIDAndPostfix which strips of the last chunk as the endpoint name (I think).

if this is indeed causing a lot of issues, it might be time to revist the allowing '/' in device and group names. I personally use it a lot because I can easiliy subscript to all devices in a room, but it does cause a lot of issues frequently.

sjorge, 2019-11-01:
Anyway, still having issues with a E1743 I borrowed from a friend for testing.

Even if the source and target do not have a '/' in the name, I am unable to bind it. It also doesn't even log that it received the mqtt topic :s

Koenkk, 2019-11-01:
Fixed in latest dev.

Multiple `/` doesn't seem to be used a lot, therefore you can encounter some bugs (but issues like this will help to resolve them).

For the E1743 I think you first need to unbind it from group 99 or 0, and maybe it can even be bound to a group and not directly to a device (not sure though).

Closing this as the OP issue has been fixed.

closed 2019-11-01 by Koenkk

sjorge, 2019-11-01:
I'll see if I can borrow the E1743 again at some point to see if I can unbind it from the group... I tried unbinding it from the coordinator with no success either.

For a group... do I just send it the group_id ?

Koenkk, 2019-11-01:
yes

sjorge, 2019-11-02:
Doesn't look fixed yet, I now get an assertion:

```
zigbee2mqtt:error 2019-11-02T10:48:16: Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Target is unknown
    at DeviceBind.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/deviceBind.js:32:9)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))
```

sjorge, 2019-11-02:
nvm, somehow a non printable char snuk in :|