Koenkk/zigbee2mqtt #2251: gledopto GU10 spots remove themself from database.db

trekker25, 2019-11-01:
Have 6 gledopto GU10 spots.

All paired in a group.

this is a correct one in database.db:
{"id":63,"type":"Router","ieeeAddr":"0x00124b001c42f6a8","nwkAddr":50706,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-S-003Z","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{}},"genIdentify":{"dir":{"value":1},"attributes":{}},"genGroups":{"dir":{"value":1},"attributes":{}},"genScenes":{"dir":{"value":1},"attributes":{}},"genOnOff":{"dir":{"value":1},"attributes":{}},"genLevelCtrl":{"dir":{"value":1},"attributes":{}},"lightingColorCtrl":{"dir":{"value":1},"attributes":{}}},"binds":[]},"13":{"profId":49246,"epId":13,"devId":528,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attributes":{}}},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.7","zclVersion":1,"meta":{}}

all 6 lights where working for months but suddenly 3 lights are not responding. I checked the database.db and suddenly i find this for the none working lights:

{"id":55,"ieeeAddr":"0x00124b001b6b433a","nwkAddr":46125,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{}}
{"id":56,"ieeeAddr":"0x00124b001c4324cd","nwkAddr":17820,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{}}

Removing them from database.db and performing a restart of docker just brings them back the same way in database.db

This has happened several times before, but these times i just tried a reset of the GU10 spot. 

But now i want to find out why, because mylady get's annoyed. 
How can i prevent this?

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

trekker25, 2019-11-01:
seems they are really kicked out the network:

error 2019-11-01T18:58:22: Publish 'set' 'state' to 'spot_badkamer_4' failed: 'Error: Data request failed with error: 'No network route' (205)'
info  2019-11-01T18:58:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'spot_badkamer_4' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"spot_badkamer_4"}}'

Think it happens when the kids by accident turn of the "real"power of the whole group of GU10 spots. After several times the lights get confused i guess.

Probably only thing that will help is a repair..

Koenkk, 2019-11-01:
What CC2531 firmware are you using? If not already, please try https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing.

If after a power cycle of the device it doesn't respond anymore, it looks that something goes wrong inside the firmware of the bulb.

trekker25, 2019-11-04:
got the latest source_routing firmware.

I needed to repair the 3 bulbs. I assume the cheap Gledoptos are just different in quality as real Hue lights. But the price is waaaay lower. 17 euro vs 50 euro

closed 2019-11-04 by trekker25