Koenkk/zigbee2mqtt #2253: Allow setting TX power for CC1352P-2

Koenkk, 2019-11-01:
The default TX power for the CC1352P-2  is 5dbm, however the device supports up to 25dbm. We should allow the user to control this.

fredrikgk, 2019-11-28:
The radio supports certain power settings in the range from -20 dBm to the maximum level for a specific device. I am not sure exactly where in the stack to find the power settings, but you can see them in SmartRF Studio. The SW will select the closest RF setting to the number provided, so if you for example set transmit_power: 6 for CC2652R, 5 dBm will be used.

There is no way to destroy the HW through this setting.

Note that the CC1352P do not have any available settings between 5 dBm and 14 dBm. Personally, I would simply go with the maximum setting anyway. For a single device for personal use, no one will ever be affected by that.

XRyu, 2019-11-28:
So, when using the CC1352P-2, setting this to 19 will get the most power and using the PA?
I have only read (but not understand) this: http://dev.ti.com/tirex/content/simplelink_cc13x2_26x2_sdk_2_40_00_81/docs/proprietary-rf/proprietary-rf-users-guide/proprietary-rf/tx-power.html#use-the-20-dbm-power-amplifier

fredrikgk, 2019-11-28:
Correct!

You do not really have to care about any of the documentation you linked to, that is handled by the SW.

Koenkk, 2019-11-28:
@fredrikgk thanks for your response, instead of setting a number, would it make sense to just set a level? e.g. `low`= -22, `medium` = 0 , `high` = 19

Alfiegerner, 2019-11-28:
Apologies for butting in - can anyone give me a sense of the range of the CC1352P-2 on max setting vs CC2530 + CC2591 (or GBAN in my case)?

fredrikgk, 2019-11-28:
@Koenkk, I like the flexibility of being able to set whatever. I would keep the configuration setting as-is.

The only change I would recommend is setting the default level for CC2652R to 5.

fredrikgk, 2019-11-28:
@Alfiegerner , the performance is very similar. I would not expect you to see any noticeable difference in range.

dmitrybabeshko, 2019-11-28:
@Koenkk 
Hi! Thank you for implementing this! Could you please also update hassio docker zigbee2mqtt-edge addon to allow setting this new experimental section with TX power?

didiht, 2019-11-28:
Will setting the transmit power also work with CC2562RB?

I know this device is not officially supported but it is managing my zigbee network.

fredrikgk, 2019-11-28:
Zigbee herdsman does not differentiate between CC2652R and CC2652RB.

Alfiegerner, 2019-11-28:
> @Alfiegerner , the performance is very similar. I would not expect you to see any noticeable difference in range.

Thank you.

dmitrybabeshko, 2019-12-12:
@danielwelch 
Hello, could you please extend zigbee2mqtt hassio docker addon to allow specifying this experimental/tx_power option? Thanks!

danielwelch, 2019-12-13:
I’d be happy to add this to the edge version. Are there other experimental options that should also be added?

dmitrybabeshko, 2019-12-13:
@danielwelch 
I know only about transmit_power.

TASSDevon, 2020-01-02:
I just got a CC1352P-2 and was wondering when this will be integrated into master. I'd very much like to bump up the TX power and be on mainline rather than dev branch. Thanks @Koenkk for your huge investment of time and work in this project!

sfstar, 2020-01-03:
Please be aware that setting the experimental transmit power too high seems to cause issues with the network stability if the coordinator looses power / is restarted.
I've set the transmit power to different settings over the last few weeks.
Whereby, I noticed that the link quality almost doubled (one way towards the receiving device) when setting the transmit power to the max.
However, when my coordinator got restarted, I suddenly got a lot of NO network route 205 issues as well as some end devices not checking in anymore.

I do not think It's a bug, but rather a side effect that comes with setting high transmit power.
I hypothesize that the CC coordinators determine routes upon starting.
Therefore, setting a high transmit power after pairing doesn't negatively affect the network.
However, when the coordinator is power cycled, it established new routes and picks routes based on the high transmit linkquality.
Thereby, creating an issue where the routes favored by the coordinator are routes that are directly connected to the coordinator.
Whereby, the receiving devices aren't capable of sending a strong enough signal to be picked up by the coordinator.

Can somebody confirm or reject my hypothesis based on their knowledge of zigbee2mqtt and the z-stack?

skandragon, 2020-01-07:
From my understanding, the mesh forms by end devices tracking who they are connected to, and using that endpoint.  This implies a two-way path, and the self-healing nature of the network will cause a coordinator with poor receive but high transmit to be unfavorable compared to a router with a good receive and transmit, from any given endpoint's POV.

1igeza, 2020-02-28:
Is this experimental function only for coordinator or is it working also with CC2530 routers?

Koenkk, 2020-02-28:
@1igeza it only works on the coordinator. This value cannot be set over the air via  a zigbee command.

gev, 2020-03-24:
The cc2538+cc2592 [datasheet](http://www.ti.com/lit/an/swra447/swra447.pdf) contains next tx power values table:
[image]
As I see for 20dbm need to pass 0xC5 into setTransmitPower(), but it is impossible because setTransmitPower() get only -128...127 values

closed 2020-05-30 by stale[bot]

basmeyer, 2022-02-01:
> Please be aware that setting the experimental transmit power too high seems to cause issues with the network stability if the coordinator looses power / is restarted. I've set the transmit power to [...]

I cannot contribute to the explanation of your theory but I do confirm the exact same experience with the TX power setting. I actually strongly discourage people to just tamper with it in order to try to get some better LQI values in a (big) live situation and rather try it on a test setup instead, because it unfortunately generated a lot of work for me. Immediately the coordinator, a Sonoff ZBDongle Plus with the firmware 20211217, got limped straight after I had put it on 20 (corresponding with +20 dBm I presume) and more then half of the devices were on either LQI N/A or 0. Tried dozen of things to undo what I did and I ended up with a NVRAM reset of the stick. That finally solved it. Luckily I did not cook it, that was my first thought.

For anyone that needs to know: [](https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md)
In my case: `python -m zigpy_znp.tools.nvram_reset /dev/ttyUSB0`
Note: re-pairing everything or using a backup of the Zigbee network mandatory.
Still not sure what really caused this behaviour but I am very interested to know. For now I leave it to default, network OK, way better coverage than with Conbee II anyway, "do not fix what ain't broken" in mind.
Definitely for the developers worth to have a second look at! Or at least to put some comment next to the setting to warn.