Koenkk/zigbee2mqtt #2254: How to acquire device firmware version?

electrofloat, 2019-11-02:
This is more of a question then anything, but I could not figure it out how to acquire a zigbee device's firmware version.
I have an issue with a Gledopto bulb and I have to figure it out what firmware it has.

Maybe I missed something in the documentation?

Koenkk, 2019-11-02:
AFAIK gledopto doesnt provide firmware updates for their device, for Hue and TRADFRI it might be possible in the feature: https://github.com/Koenkk/zigbee-herdsman/issues/60

closed 2019-11-02 by Koenkk

electrofloat, 2019-11-02:
I see. But which property is the firmware version?
```
{"id":2,"type":"Router","ieeeAddr":"0x00124b001ea82419","nwkAddr":33606,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-B-008Z","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-B-008Z","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.0"}},"genOnOff":{"attributes":{"onOff":1}},"lightingColorCtrl":{"attributes":{"currentX":38076,"currentY":16056,"colorTemperature":500}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[]},"13":{"profId":49246,"epId":13,"devId":528,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.0","zclVersion":1,"interviewCompleted":true,"meta":{}}
```
appVersion? stackVersion? hwVersion? dateCode? swBuildId? zclVersion?
Or all of these together?

Also would be nice to have an mqtt topic to query the device's firmware version.

Koenkk, 2019-11-03:
This is included in https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigdevicesget.

CodeFinder2, 2020-01-14:
> This is more of a question then anything, but I could not figure it out how to acquire a zigbee device's firmware version.
> I have an issue with a Gledopto bulb and I have to figure it out what firmware it has.
> 
> Maybe I missed something in the documentation?

I was also wondering about this question. It should be the `swBuildId` attribute stored in the `database.db` file. Just verfied this for OSRAM smart plugs and it seems to apply for other vendors / devices as well, although the format (of the string/data) seems to differ across vendors.

OSRAM smart plug is reported to be `"swBuildId":"V1.04.90"` on my devices and I've read 1.04.12 is also out there (sometimes referred to as 'V12').

HTH!