Koenkk/zigbee2mqtt #2255: Unable to add/remove groups via MQTT

sjorge, 2019-11-02:
It doesn't seem possible to add/remove actual groups over mqtt,

It would be nice to be able to add a new group by setting the friendly name for a group_id, something like

zigbee2mqtt/bridge/group/add/123 and payload 'test_group' will create test_group with id 123
zigbee2mqtt/bridge/group/remove/123 any payload will remove group with id 123 from the config
zigbee2mqtt/bridge/group/remove/test_group any payload, same as above but based on friendly_name

Koenkk, 2019-11-03:
This is already possible with: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqtt/bridge/config/add_group

closed 2019-11-03 by Koenkk

sjorge, 2019-11-03:
That doesn't seem to have a way to specify the group id though, e.g. if you want to create a specific one like 99 for the on/off ikea remote or random generated one from the ikea stearing device remote.

Koenkk, 2019-11-03:
That's indeed not possible, for that you need to use the `configuration.yaml`.

sjorge, 2019-11-03:
Just had a look, looks like something I might be able to add to add_group. Will chuck it on my TODO.