Koenkk/zigbee2mqtt #2256: Add device name to "failed to ping" message

copystring, 2019-11-02:
To be a lot more useful when checking the log would be possible to add the name of the device which failed to be pinged?

Maybe:
`zigbee2mqtt:error 2019-11-02T17:22:43: Failed to ping '0x7cb03eaa00b23e45' (WeihnachtsbeleuchtungKueche)
`

instead of:
`zigbee2mqtt:error 2019-11-02T17:22:43: Failed to ping '0x7cb03eaa00b23e45'`

Koenkk, 2019-11-03:
Implemented.

closed 2019-11-03 by Koenkk

copystring, 2019-11-03:
Thanks for the update. This works fine for pings which succeed. However failed pings still only show the address.

Koenkk, 2019-11-03:
Can you share your log and `configuration.yaml`?

copystring, 2019-11-03:
My bad. I just happen to have 2 devices which I did not rename and it happened to be those two appearing in the log. Sorry about that. It works good. Thanks.