Koenkk/zigbee2mqtt #2257: Restarting Home Assistant makes the dev branch crash

x99percent, 2019-11-02:
# Bug Report
## What happened
zigbee2mqtt crashes when HA is restarted

## What did you expect to happen
No crash

## How to reproduce it (minimal and precise)
Restart Home Assistant

## Debug Info
zigbee2mqtt version: docker koenkk/zigbee2mqtt:latest-dev
CC253X firmware version: 20190608, I think

```
/app/lib/extension/homeassistant.js:1104
                this.zigbee.getAllClients().forEach((device) => {
                            ^

TypeError: this.zigbee.getAllClients is not a function
    at Timeout.setTimeout [as _onTimeout] (/app/lib/extension/homeassistant.js:1104:29)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-11-02T18_27_49_319Z-debug.log
```

This is the error shown in the docker log. The error doesn't show in the log file.

Innanazmova83, 2019-11-02:
> <h1>Bug Report</h1>
> <h2>What happened</h2>
> <p>zigbee2mqtt crashes when HA is restarted</p>
> <h2>What did you expect to happen</h2>
> <p>No crash</p>
> <h2>How to reproduce it (minimal and precise)</h2>
> <p>Restart Home Assistant</p>
> <h2>Debug Info</h2>
> <p>zigbee2mqtt version: docker koenkk/zigbee2mqtt:latest-dev<br>
> CC253X firmware version: 20190608, I think</p>
> <pre><code>/app/lib/extension/homeassistant.js:1104
>                this.zigbee.getAllClients().forEach((device) =&gt; {
> </code></pre>
> <p>This is the error shown in the docker log. The error doesn't show in
> the log file.</p>
> style="font-size:small;-webkit-text-size-adjust:none;color:#666;">&mdash;<br
> href="https://github.com/Koenkk/zigbee2mqtt/issues/2257?email_source=notifications&amp;email_token=AM32YJSJVCBD36YSDKQBDELQRXCAFA5CNFSM4JIHKA42YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4HWL6MYQ">view
> it on GitHub</a>, or <a
> height="1" width="1" alt="" /></p>
> <script type="application/ld+json">[
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target":
> "name": "View Issue"
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",

Хуй

Получить TypeApp для Android

п>

Koenkk, 2019-11-03:
Did you overwite homeassistant.js via a docker volume? (Probably to add support for new devices), if so remove this, the file you are using is outdated and not compatible.

x99percent, 2019-11-06:
I was unaware that the homeassistant.js file was included in the dev container. The instructions indicate that you just have to switch to the "-dev" label, and that's it.

closed 2019-11-06 by Koenkk