Koenkk/zigbee2mqtt #2258: Xiaomi vibration sensor doesn't show the strength value (vibration curve)

kitopopo, 2019-11-02:
Hello everyone,
I am interested in knowing if xiaomi vibration sensor could show the information called "strenght" inside of "vibration curve" in the device menu". This information appears correctly displayed in the application "mi home" of xiaomi, but after linking the vibration sensor to zigbe2mqtt I can not see this information, only vibration, tilt, angle etc ...
It would be very interesting for me and other users to be able to use the "strenght value" information to be able to close my awnings depends on the vibration produced by the wind. This information in the xiaomi application is shown with 3 digits ranging from value 0 to value 999. I would be very grateful if you could implement this value. Thanks in advance and best regards

# Bug Report
## What happened
Xiaomi vibration sensor don´t show "strenght value" 
## What did you expect to happen
I thought that strenght value appears in the info sensor
## How to reproduce it (minimal and precise)
moving the sensor i can see all values
## Debug Info
zigbee2mqtt version: 20190608
CC253X firmware version: CC2531

[image]

kitopopo, 2019-11-25:
Good morning,

Is there any progress in the inclusion of the "strenght value" of the xiaomi vibration sensor?

in the following link a user comments that with the zigate plugin he can collect the strenght value information. Would it be possible to add this value to zigbee2mqtt? 

https://github.com/domoticz/domoticz/issues/3300

thanks in advance. Best regards.

kitopopo, 2020-01-12:
The vibration sensor nowadays shows:

`'zigbee2mqtt/0x00158d0002axxxxx', payload '{"battery":100,"voltage":3005,"linkquality":18,"angle":5,"angle_x":-2,"angle_y":-55,"angle_z":35,"angle_x_absolute":92,"angle_y_absolute":145,"action":"vibration"}'`

Other information provided by this sensor on the Xiaomi Gateway is the value of strenght.
Are you working to show the strenght value? I would be very grateful if this sensor showed the strenght value, depending on this value from 0 to 999 units I could close my awnings when the strenght value exceeded for example 88 strenght units, I think it would be really useful

The zigate shows correctly this info:

[image]

Thanks in advanced again.

kitopopo, 2020-01-23:
Hi dear @Koenkk ,

What would it look like to you if I link my vibration sensor to the xiaomi gateway and sniff the traffic causing several vibrations with different intensity? would this be enough for you to be able to implement the "strenght" value that the xiaomi gateway shows? It would be very useful for me. I would be eternally grateful for the implementation. I await your news. and thank you very much for your work, dedication and your wonderful project. I hope you have a good day. A cordial greeting.

Koenkk, 2020-01-23:
@kitopopo yes if you could sniff the traffic that can be done, at the time you sniff the traffic also write down the actual values you see in the xiaomi app.

kitopopo, 2020-01-23:
wonderful. I am installing everything to smell the traffic on windows in a few minutes. Let's see if i get it. very friendly

kitopopo, 2020-01-23:
Dear @Koenkk , 

With the 25 channel i have the data of the sniffer between vibration sensor and the xiaomi gateway. I have added the vibration sensor to the start of the sniffer. after i have make 3 vibrations. ( in the 153 seconds wireshark log , 214.85 seconds log , 296.24 seconds log and 362.79 secons log ) . 

in the lasts updates of the "mi home app" xiaomi has dissabled the vibration curve, this information is not show inside option of vibration sensor. But, i have installed a old version for make the sniffer. In this "my home" version appears the vibration curve and the strenght value but the info no is filled in At least now. Maybe later this information appears. A few months ago sometimes this information was filled in and sometimes not. I wish you could see the vibration values ​​in the sniffer file. 

Maybe have you old sniffer files of this sensor or maybe you have a contact with 
some zigate developer due to this value in zigate is showed.

I have not more ideas. I have sended the file telegram way.

Thank you very much in advance and thanks again for your collaboration. I am very grateful for your sup

kitopopo, 2020-01-24:
Dear @Koenkk ,

I have the measurements in the vibration curve of xiaomi vibration sensor:
- Strenght 53 ( today at 00:36 )
- Strenght 189 ( Today 01:04 )
- Strenght 3 ( Today 01:22 )
- Strenght 4 ( Today 01:27 )
- Strenght 240 ( Today 01:33 )

Yesterday I made a last measurement of vibration. This vibration should be the last (240) and I think it should be included in this sniffer file. This second sniffer file of vibration sensor only has the join of vibration sensor and 1 vibration measure.

Koenkk, 2020-01-25:
Added, will be supported in the dev branch in a few hours, thanks!

closed 2020-01-25 by Koenkk

kitopopo, 2020-05-04:
Dear friend @Koenkk  , I hope you are okay.

Do you remember this post? I was speaking this morning with @waaren from domoticz forum about that i we can use the 'strenght' value in domoticz to be able close my awinds with the wind force It would be very interesting. Mr @waaen comment this:

> The point is that domoticz only create these devices because it is instructed to do so by the zigbee2mqtt module. So if you want a 4th device to be created the zigbee2mqtt module must be updated to add that request.

Currently when adding xiaomi vibration sensor, domoticz generates these 3 devices:

```
4xx	zigbe2mqtt	0x00158d000xxxxx_signal	54	zigbe2mqtt - 0x00158d000xxxxx (Link Quality)	General	Custom Sensor	115

4xx	zigbe2mqtt	0x00158d000xxxxx_cell	60	zigbe2mqtt - 0x00158d000xxxxx (Battery Voltage)	General	Voltage	3.025 V

4xx	zigbe2mqtt	0x00158d000xxxxx_action	64	sensor vibra	Light/Switch	Selector Switch	Off
```

I would be very grateful if you could include a fourth device with the value of streenght, is this possible? I and other users of the domoticz telegram group would be eternally grateful to you. to be 

Thank you very much in advance.

Koenkk, 2020-05-04:
I'm not sure how devices are discovered in Domitcz, but I guess it has to be added here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin (so please open an issue there to add the strenght sensor for DJT11LM)

kitopopo, 2020-05-04:
Yes, I have thist plugin installed, I guess I should tell the problem to @stas-demydiuk
, if you cannot solve it and from domoticz neither then I will write to mr stas-demydiuk, thanks @Koenkk . best regards