Koenkk/zigbee2mqtt #2260: Blaupunkt SCM-S1 not recognized (Unsupported Device)

snoopyCode, 2019-11-03:
# Bug Report

## What happened
I wired a Blaupunkt SCM-S1 and physical buttons with my shutters. Controlling the shutters with the buttons works.

When I start pairing the SCM-S1 it says 'Device with modelID 'SCM_00.00.03.14TC' is not supported.'

## What did you expect to happen

Digging into zigbee converters 10.1.4 (which is used in 1.5.1) reveals that Blaupunkt SCM-S1 should be the model 'SCM-R_00.00.03.15TC' and 'SCM_00.00.03.14TC'. Also newer versions of the converters does not contain 'SCM_00.00.03.14TC'.

A bit of research suggests that SCM_00.00.03.14TC is not that wrong, e.g. here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1229

I don't think I can update the SCM-S1 device.

## How to reproduce it (minimal and precise)

Pair the device by holding the reset button for more than 10 sec until the red light blinks.

## Debug Info
zigbee2mqtt version: 1.5.1 (1.6.0 didn't work for me. Maybe because I updated from 1.2.1)
cc2531 Coordinator firmware version: 20190223

Koenkk, 2019-11-03:
It's definitely not wrong, some devices use multiple identifiers. Added it, will be supported in the latest dev branch in a few hours.

closed 2019-11-03 by Koenkk

snoopyCode, 2019-11-03:
For a quick solution I justed added
`
    {
        zigbeeModel: ['SCM_00.00.03.14TC'],
        model: 'SCM-S1',
        vendor: 'Blaupunkt',
        description: 'Roller shutter',
        supports: 'open/close',
        fromZigbee: [fz.cover_position_via_brightness, fz.cover_state_via_onoff],
        toZigbee: [tz.cover_position_via_brightness, tz.cover_open_close_via_brightness],
    },
`
in the converters device file and it works great. May you add the model in the next version. That would be great. Thanks.

Koenkk, 2019-11-03:
I've already done that, available in latest dev.