Koenkk/zigbee2mqtt #2261: IKEA LEPTITER Recessed spot light (T1820) does not connect

tkiced, 2019-11-03:
# Bug Report
## What happened

IKEA LEPTITER Recessed spot light (T1820) does not connect with Hass.io add-on of zigbee2mqtt but it is in the supported devices list.

I got this message in zigbee2mqtt logs:
"zigbee2mqtt:warn 11/3/2019, 1:19:19 PM Device with modelID 'LEPTITER Recessed spot light' is not supported.
  zigbee2mqtt:warn 11/3/2019, 1:19:19 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html"

I checked the file "devices.js" in the docker container of the add-on at location "/zigbee2mqtt-1.6.0/node_modules/zigbee-shepherd-converters/" and i did not find this device in the list.

## What did you expect to happen

How can we do to make it working ?

## How to reproduce it (minimal and precise)

I have done a reset factory of the IKEA LEPTITER as describe in the documentation to connect the  spot light.

## Debug Info
zigbee2mqtt version: version: 1.6.0
CC253X firmware version: 20190608

Koenkk, 2019-11-03:
This device is only supported in the dev branch yet (edge for hassio). 

Anyway, it shouldn't be on the supported devices page as it is not supported in a release version yet, therefore I removed it for now.

closed 2019-11-03 by Koenkk

tkiced, 2019-11-03:
thank you for your prompt reply