Koenkk/zigbee2mqtt #2262: Installation on raspberry4 with buster lite sucks

fellnase, 2019-11-03:
- Raspberry-Pi 4 -2GB
- Raspbian Buster Lite ( 2019-09-26) (Fresh installation)
- Follow installation from: https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

After installing "npm install" I got following error

added 2097 packages from 594 contributors and audited 877679 packages in 257.751s
found 3 vulnerabilities (2 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details

Ignoring these error try to start "zigbee2mqtt" with " npm start"
I got following error message:

```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'zigbee-herdsman/dist/deprecated/ziee'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/zapp/cie.js:6:14)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-11-03T14_17_58_752Z-debug.log

```

harry126, 2019-11-03:
I have the same problem after zigbee2mqtt-update.

[image]

Koenkk, 2019-11-03:
Duplicate of #1892, possible solution: https://github.com/Koenkk/zigbee2mqtt/issues/1892#issuecomment-527837064

If this doesn't work use the latest dev branch.

closed 2019-11-03 by Koenkk