Koenkk/zigbee2mqtt #2265: Calling add_group, remove_group, add_group results in error

sjorge, 2019-11-03:
# Bug Report
## What happened
Added a test group by publishing `test` to `zigbee2mqtt/bridge/config/add_group`,  remove it afterwards by publishing `test` to `zigbee2mqtt/bridge/config/remove_group` and add another group `test` to `zigbee2mqtt/bridge/config/add_group` ...

Results in the following error:

```
zigbee2mqtt:error 2019-11-03T19:45:26: Failed to call 'BridgeConfig' 'onMQTTMessage' (Error: Group with groupID '1' already exists
    at Function.create (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/group.js:78:19)
    at Controller.createGroup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:227:32)
    at Zigbee.createGroup (/opt/zigbee2mqtt/lib/zigbee.js:214:30)
    at BridgeConfig.addGroup (/opt/zigbee2mqtt/lib/extension/bridgeConfig.js:229:21)
    at BridgeConfig.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/bridgeConfig.js:309:44)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))
```

This makes sense as 'retain' seems to be a required property for groups.

## What did you expect to happen
The next free ID (in this case 1) to be succesfully reused.

## How to reproduce it (minimal and precise)
By adding, removing and adding a group via MQTT

## Debug Info
zigbee2mqtt version: 1.6.0 @ dev from 20191103
CC253X firmware version: n/a

sjorge, 2019-11-03:
I found the issue, not sure how to fix myself.

After a remove_group... the group still exists in database.db!

```
{"id":18,"type":"Group","groupID":1,"members":[],"meta":{}}
{"id":19,"type":"Group","groupID":99,"members":[],"meta":{}}
{"id":20,"type":"Group","groupID":2,"members":[],"meta":{}}
{"id":21,"type":"Group","groupID":100,"members":[],"meta":{}}
```
99 and 100 were from tests with my providing a groupID, 1 and 2 were without and were auto created.

sjorge, 2019-11-05:
I'm not actually sure what the proper fix here is... 
1. check the config file only and succeed in the add_group even if the group exists in the db, we should then somehow get the members from the db and write them to the config?
2. have remove_group do a remove of all the group members before removing the group and fail the remove if removing a member fails? if we succeed we can then remove the group from the db too because it won't have any member so it should no longer be in use.

Personally 2 sounds better than 1, but seems way more complex to implement and we'll have to deal with device that are 'dead' that will never respond to the remove from group message... or with sleeping device that missed the request to leave the group.

Koenkk, 2019-11-08:
I will opt for 2 here, I will make the required changes in herdsman.

sjorge, 2019-11-08:
With the new timeout stuff... can we have herdman maybe keep trying to remove group members for a maybe like 1h+ ? Most battery device seem to checkin every so often, I know the aqara sensors do so every hour. I

Koenkk, 2019-11-08:
We shouldn't worry to much about the end devices, these are not the type of devices that you put in groups. I'm thinking about something similar that we currently have for devices:
- `remove`: first removes all devices from the group, if all succeeded, removes the group from `database.db`
- `force_remove`: tries to remove all devices from the group, no matter if failed or succeeded, removes the group from `database.db`.

sjorge, 2019-11-08:
QQ about something I thought about while stock in traffic...

Do we keep track of bindings in the db? I’d so we may also want to unbind the group if it is still bound to a device

Koenkk, 2019-11-08:
Yes good point! Bindings are also stored (only when done after zigbee2mqtt 1.6)

alex-savin, 2019-11-15:
I have the same error with CC1352P-2
```
zigbee2mqtt:info  2019-11-15T19:36:24: Zigbee: allowing new devices to join.
zigbee2mqtt:error 2019-11-15T19:36:30: Failed to call 'BridgeConfig' 'onMQTTMessage' (Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
zigbee2mqtt:info  2019-11-15T19:41:26: Zigbee: disabling joining new devices.
zigbee2mqtt:error 2019-11-15T19:41:32: Failed to call 'BridgeConfig' 'onMQTTMessage' (Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
```

sjorge, 2019-12-18:
Ran into this again with removing a group today :(

sjorge, 2020-02-16:
This is still an issue, be gone stale label!

Koenkk, 2020-02-22:
Fixed in latest dev branch, thanks!

closed 2020-02-22 by Koenkk