Koenkk/zigbee2mqtt #2267: After putting IKEA Tradfri remote in a group with power supply there are no brightness updates.

cbullo, 2019-11-03:
# Bug Report
## What happened
I grouped my IKEA Tradfri remote with the Tradfli 30W power supply as described at the bottom of the this link https://www.zigbee2mqtt.io/information/binding.html
It worked fine I can smoothly control the light and I can control power supply from my MQTT broker. My issue is that when using the remote I'm not getting brightness changed notifications on my MQTT from power supply. However, I'm getting state notifications.

## What did you expect to happen
When I press brightness button I'm expecting to get a new message with updated brightness on my MQTT broker.

## How to reproduce it (minimal and precise)
Set up IKEA Tradfri with Tradfri power supply as described at the bottom here : https://www.zigbee2mqtt.io/information/binding.html
Turn on lights and press brightness up/down button on the remote. Observe lack of message on the broker
Press the toggle button on the remote. Observe that there is message one the broker.
## Debug Info
zigbee2mqtt version:
latest
CC253X firmware version:
latest

sjorge, 2019-11-04:
@Koenkk maybe another device that only report state? If so the same trick as with the hue bulbs.

Koenkk, 2019-11-05:
To receive state changes you have to use the report function: https://www.zigbee2mqtt.io/information/report.html

cbullo, 2019-11-05:
Oh, nice. I haven't noticed this feature before. I'll try it later, thanks.

cbullo, 2019-11-05:
@Koenkk Looks like report works fine for what I need, maybe a bit spammy. Out of curiosity, why is `state (ON/OFF)` reported even without setting `report` to true and brightness is not?

Koenkk, 2019-11-06:
It depends on the device, some report it by default and some dont.

closed 2019-11-06 by Koenkk