Koenkk/zigbee2mqtt #2268: Eurotronic Spirit SPZB0001 is not pairing with zigbee2mqtt

omitpi, 2019-11-03:
# Bug Report
## What happened
I wanted to pair a Eurotronic Spirit zigbee thermostat to my zigbee2mqtt network. But it doesn't pair. I reset the device and put it in Join Mode (showing "Jin" on the display) 
OSRAM-Smart Plugs are well connected for example. 
Some Specs: {"version":"1.6.0","commit":"e26ad2a","coordinator":20190608,"log_level":"info","permit_join":true}
The R-Pi 3B is running openhabian as OS. 

## What did you expect to happen
That it just connects like my OSRAM-Smart Plugs already did. Seeing it to appear in the MQTT Explorer.

## How to reproduce it (minimal and precise)
It happens all the time.

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: 20190608

Koenkk, 2019-11-04:
It might be worth trying the dev branch as it has improved pairing behaviour.

omitpi, 2019-11-04:
@Koenkk Thanks for your answer! :-)
Do you mean the "developement firmware version" for flashing the CC253X Dongle?

Koenkk, 2019-11-05:
No, see https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

omitpi, 2019-11-05:
Okay, I did this and it says 
{"version":"1.6.0","commit":"00bc4de","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}
but no pairing either...I already switched the Eurotronic Spirit SPZB0001 but same issue...

Do you have any more advices?

Koenkk, 2019-11-06:
- Did you pair any other devices to zigbee2mqtt?
- Is another zigbee hub running? (so that it could pair to a different network)
- Can you share your `database.db`.

bredzim, 2019-11-07:
I have the same problem, using coordinator firmware revision: 20190425 and using dev branch, no success with Eurotronic SPZB0001 pairing (no any message seen in debug mode log).

Other devices (Aqara, Ikea Tradfri) are working fine.

omitpi, 2019-11-07:
> * Did you pair any other devices to zigbee2mqtt?
> * Is another zigbee hub running? (so that it could pair to a different network)
> * Can you share your `database.db`.

1) Yes two OSRAM Smart Plugs.
2) No, just one hub.
3) Like this?
 {"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e1b2a9","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1572817679,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList"$
{"id":2,"type":"Router","ieeeAddr":"0x7cb03eaa00b1f0f8","nwkAddr":13324,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01","epList":[3],"endpoints":{"3":{"profId":49246,"epId":3,"devId":16,"inCl$
{"id":4,"type":"Router","ieeeAddr":"0x7cb03eaa0a025803","nwkAddr":25459,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01","epList":[3],"endpoints":{"3":{"profId":49246,"epId":3,"devId":16,"inCl$

Koenkk, 2019-11-08:
I found some interesting information here from @allofmex 

> What you may also try is holding all three button, then a counter starts and Spz resets.

Can you try this?

omitpi, 2019-11-08:
I tried this several times already....No success!

Is there maybe a mistake in the pairing specifications?

bredzim, 2019-11-08:
In my case this reset solved the problem - pairing done - many thanks Koen for the hint!

omitpi, 2019-11-08:
What did you do exactly, bredzim?

bredzim, 2019-11-08:
I pressed three buttons (boost, minus and plus simultaneously) until the counter on the LCD has reached 10 and LCD displayed "rES". After that thermostat started to pair and after few seconds finished with success.

Try this way, maybe it is also solution for you.

allofmex, 2019-11-08:
And please have the CC253X fresh plugged. Depending on firmware, pairing may work within the first minutes after startup only!

rob1n, 2019-11-08:
Resetting the device (by holding all three buttons) worked for me as well, even tho I'm not on `edge`.
Thanks y'all :)

omitpi, 2019-11-12:
Alrigthy! I am travelling this whole week, will try it when I am home next week. Thank guys, let's see if this helps! :-)

omitpi, 2019-11-17:
Okay Guys, got this finally working with the reset...you need to stay on the three buttons till the display is turned off. :-D

Is anyone of you using this with openhab? If yes how did you define your channels? 
The "Incoming value transformation" and JSONPATH?

Thanks.

danergo, 2019-11-26:
> And please have the CC253X fresh plugged. Depending on firmware, pairing may work within the first minutes after startup only!

Do you have any idea on what changes after some minutes in the firmware that prevents the connection of this (and maybe other) kind of devices?

allofmex, 2019-11-27:
> ...what changes after some minutes in the firmware that prevents the connection 
I did not found a firmware parameter to configure this timeout.

To clearify: this means, only **pairing of new** (factory default) devices works for first minutes only (Stack 1.2 firmware). 
If the device was paired before and just power cycled (example Eurotronic battery replaced), it rejoins any time.

danergo, 2019-11-27:
> I did not found a firmware parameter to configure this timeout.

Have you sniffed the ZB traffic between Eurotronic and CC253x?
Do you know what is the request from the Eurotronic that goes through after freshly replugged CC253x, but stuck if you don't replug it?

> To clearify: this means, only **pairing of new** (factory default) devices works for first minutes only (Stack 1.2 firmware).
> If the device was paired before and just power cycled (example Eurotronic battery replaced), it rejoins any time.

That's fine yes, my intention is to improve the stack 1.2 fw so we don't need to unplug/replug on new device pairing situations.

allofmex, 2019-11-27:
> Have you sniffed the ZB traffic between Eurotronic and CC253x?

I have this with all devices. I guess it is a security function in 1.2 z-stack. The coordinator publishes network key only at beginning. 
Would like to have this changed too, but not critical. Just need to be known to users.

s00500, 2019-12-17:
Yes cool, just had the same after I changed my network key and then it would not show up anymore, holding the 3 buttons till the rest solved the issue immediately, needs to go into the docs, then this can be closed I guess :-)

closed 2019-12-17 by Koenkk