Koenkk/zigbee2mqtt #2273: ZigBee device types

kaelsaes, 2019-11-04:
Hi,

what are the device types CHILD_FFD_RXI and PARENT?

Thx

alwashe, 2019-11-09:
A child is an end device without routing.
A parent is a device wich can be a router or coordinator.

https://www.zigbee2mqtt.io/information/zigbee_network.html
```
End Device

End devices do not route traffic. They may also sleep, which makes end devices a suitable choice for battery operated devices. An end device only has one parent, either the coordinator or a router, generally the closest device when it was paired. All communications to and from the end device is via their parent. If a parent router goes offline all traffic to its children will cease until those end devices time out and attempt to find a new parent. Some models of end device, notably Xiaomi, don’t attempt to find a new parent so will remain isolated until re-paired with the network.
```

kaelsaes, 2019-11-10:
Thanks a lot.

closed 2019-11-10 by kaelsaes