Koenkk/zigbee2mqtt #2274: Xiaomi Motion Sensor RTCGQ11LM not reporting occupancy:true the second (like#1498)

e-t-h, 2019-11-04:
# Bug Report

I have the same issue like described in the closed issue #1498: sensors don't report occupancy:true.
I have two RTCGQ11LM sensors, ran for a long time at an Aqara Hub, without any issues. Since I am working with zigbee2mqtt I've noticed that they shows the same behavior @zerofruchtshake desribed in issue #1498: the occupancy value is not set properly. I am currently not able to get a message with "true".

## What happened

Two sensors:
Sensor 1 (friendly name Bew_unten) is hardware modified like discribed in the documentation to shorten the time the sensor ignores any movements.

Sensor 2 (friendly name Bew_oben) is not modified.
Both have the same issue.

## What did you expect to happen

sensors value "occupancy" should be set to "true"

## How to reproduce it (minimal and precise)

It happens every time with every sensor; sensors were resetted, 

## Debug Info
zigbee2mqtt version: 1.6.0,
CC253X firmware version:20190608

I've collect some debug messages with the option DEBUG=zigbee-shepherd*-option, might be that helps? Currently I am not able to sniff with a second CC253x..

sensor 1

> zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg +114ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1024,"srcaddr":33408,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":34,"securityuse":0,"timestamp":8829962,"transseqnumber":0,"len":8,"data":{"0":24,"1":47,"2":10,"3":0,"4":0,"5":33,"6":18,"7":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":47,"cmdId":"report","payload":[{"attrId":0,"dataType":33,"attrData":18}]}} +92ms
  zigbee2mqtt:debug 11/4/2019, 12:27:34 PM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":18}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b48ee2) of endpoint 1
  zigbee2mqtt:info 11/4/2019, 12:27:34 PM MQTT publish: topic 'zigbee2mqtt/Bew_oben', payload '{"illuminance":18,"linkquality":34,"occupancy":false,"battery":100,"voltage":3035}'
  zigbee2mqtt:debug 11/4/2019, 12:27:34 PM Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":18}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b48ee2) of endpoint 1
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +31ms 

sensor 2

> zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg +8ms
 zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1024,"srcaddr":40372,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":31,"securityuse":0,"timestamp":8830101,"transseqnumber":0,"len":8,"data":{"0":24,"1":24,"2":10,"3":0,"4":0,"5":33,"6":44,"7":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":24,"cmdId":"report","payload":[{"attrId":0,"dataType":33,"attrData":44}]}} +9ms
  zigbee2mqtt:debug 11/4/2019, 12:27:34 PM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":44}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b42bd0) of endpoint 1
  zigbee2mqtt:info 11/4/2019, 12:27:34 PM MQTT publish: topic 'zigbee2mqtt/Bew_unten', payload '{"illuminance":44,"linkquality":31,"battery":100,"voltage":3035,"occupancy":false}'
  zigbee2mqtt:debug 11/4/2019, 12:27:34 PM Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":44}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b42bd0) of endpoint 1

Any ideas? What can I additionaly provide? Does this debug output help to determine what the stick have received? I assume that the data item in the incoming message stores this information? Am I able to figure out how the software translate the data into the payload (if so, where I can find it)?

data 1 {"0":24,"1":47,"2":10,"3":0,"4":0,"5":33,"6":18,"7":0}
data 2 {"0":24,"1":24,"2":10,"3":0,"4":0,"5":33,"6":44,"7":0}
payload 1  {"illuminance":18,"linkquality":34,"occupancy":false,"battery":100,"voltage":3035}'
payload 2  {"illuminance":44,"linkquality":31,"battery":100,"voltage":3035,"occupancy":false}

Sincerly..

Koenkk, 2019-11-05:
In the log I see that only the illuminance level is reported and no occupancy, can you check again with the xiaomi hub if they work there?

e-t-h, 2019-11-05:
I've just did it, and it works. I get the message "Motion detected, Illumination is: xx Lux.

Today I've observed the sensors and found messages with occupancy = true. Here a corresponding log:

> zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1024,"srcaddr":40372,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":105,"securityuse":0,"timestamp":5300358,"transseqnumber":0,"len":8,"data":{"0":24,"1":11,"2":10,"3":0,"4":0,"5":33,"6":36,"7":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":11,"cmdId":"report","payload":[{"attrId":0,"dataType":33,"attrData":36}]}} +9ms
  zigbee2mqtt:debug 11/4/2019, 10:13:34 PM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":36}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b42bd0) of endpoint 1
  zigbee2mqtt:info 11/4/2019, 10:13:34 PM MQTT publish: topic 'zigbee2mqtt/Bew_unten', payload '{"illuminance":36,"linkquality":105,"battery":100,"voltage":3035,"occupancy":true}'
  zigbee2mqtt:debug 11/4/2019, 10:13:34 PM Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":36}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b42bd0) of endpoint 1
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1030,"srcaddr":40372,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":105,"securityuse":0,"timestamp":5300370,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,12,10,0,0,24,1]}} +22ms

Where can you see that no occupancy is reported? For me it looks quite similar..

 Thx

Koenkk, 2019-11-06:
If occupancy is reported you will find `msOccupancySensing` in the log, you should find it above the log you currently posted (otherwise `occupancy` won't be `true`).

guardiande, 2019-11-18:
I just posted #2364.

I wanted to let you know that I once again was able to get the faulty sensor to work again. I re-paired it several times and suddenly it started working again.

Normally it works for a few weeks or months and then it starts failing again. The procedure to get it cooperate again is not exactly reproducible and simply restarting zigbee2mqtt can make it fail again.

Sepultura22, 2019-11-20:
I have the same problem

e-t-h, 2019-11-20:
Now we are three of us ;-) 
In the meanwhile I've tested both of my sensors at the Xiaomi Hub with no errors. The sensors reports the illuminance from time to time and also every event of movement correctly. Next step in my case is to update the Zigbee2MQTT software to the newest version, reconnect the sensors to it and test again (with the knowledge of what guardiande wrote). 
I'll be back ;-)

Sepultura22, 2019-11-21:
I have the latest version of zigbee2mqtt@1.7.1 and the latest firmware in CC2531 (20190608), but the problem persists.

Koenkk, 2019-11-21:
To further debug this issue, could you sniff the traffic while the sensor should detect occupancy when connected with zigbee2mqtt (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)?

e-t-h, 2019-11-21:
I suppose I need another CC2531? I will order one and come back with the results. It's gonna take some time.

Koenkk, 2019-11-21:
Yes you need another one

Sepultura22, 2019-11-21:
I will wait patiently for e-t-h to do the test, because I don't have another dongle either. :)

e-t-h, 2019-12-16:
Hi Koenkk,
I've got my second receiver, flashed it, installed Wireshark and ZBOSS on my Windows Laptop like described in your doc. I can start the system, the green LED is flashing. But I don't see any traffic although there is something. 
I am not sure which Zigbee-page I have to select, I've tried all of them. 
Are there known pitfalls?

e-t-h, 2019-12-16:
Okay, thank you for your support, surprisingly it is now working! Please ignore the posting before ;-)

e-t-h, 2019-12-17:
Hi Koenkk,
amusingly the sensors do now what I expect they have to do. 
Did you change anything in the software? 
I've done what you've suggested regarding the sniffing and additionally I've updated the zigbee2mqtt to the latest version 1.8.0. After that I've begun the tests. Everything works well with the motion sensors now. They report the status "occupancy=true" and after a time with no movement the status is set to "false". 

Might be you've changed something? Or: I've been running  the sensors together with the XIAOMI-hub meanwhile I've been waiting for the USB stick. Is it possible that an update of the firmware of the sensor have happened in this period?
I'll observe the behavior and will report here if something different happens. 

Is there anything I can read to be able to understand the data packets I receive with my brand new sniffer what you can recommend? I would appreciate any advice. 

By the way, is it possible to initiate a transmit of the illumination value by sending a request from the stick?

Last but not least: Thank you for your work!

Ekkehard

guardiande, 2019-12-17:
Hi @e-t-h,

I'm afraid that you most likely experience the same symptoms in the future again. I'm having this problem for many months now and the problem normally persists only for a week or something and then the sensor stops working again.

Sometimes it happened right after just restarting z2m and I think it also happened during normal operation. Then it takes me many re-pairing to get it work again and then it works stable for some time.

It could be that the behaviour is more likely to happen if the sensor connects via a router but I'm not sure.

Jan

e-t-h, 2019-12-17:
Hi Jan,
I am also not very convinced that the issue simply disappeared. Thats why I've asked for hints how to work with the sniffing tool. Unfortunately I'm not experienced with Zigbee messages. 
I'll observe the behavior and when it'll changes I'll come back to this post. Currently I don't see much more opportunities. I'm waiting for the reply of Koenkk also.
Did you upgrade to 1.8.0? This update is only just two days old. If not it would be helpful if you can check if it includes the solution. I had the 1.6.0 running when the issue appeared.

Ekkehard

Koenkk, 2019-12-17:
Xiaomi devices can drop off the network when the linkquality is low, this can be improved: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

guardiande, 2019-12-17:
Hi @Koenkk,

The sensors don't completely drop off the network since they're still reporting luminance and battery levels every hour our so (as you already saw from analyzing the logs a few posts above).

Jan

Koenkk, 2019-12-17:
@guardiande that's even more interesting. When this happens, could you sniff the zigbee traffic when you expect occupancy being detected? In this way we can determine if it's an issue of zigbee2mqtt or the sensor itself. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

guardiande, 2019-12-17:
@Koenkk I just got myself a second CC2530 for that but (un-)fortunately it's working now. Will have to way for the next failure :)

e-t-h, 2020-01-21:
> Hi @e-t-h,
>
> I'm afraid that you most likely experience the same symptoms in the future again. I'm having this problem for many months now and the problem normally persists only for a week or [...]

Hi Jan,
you are right, one of my sensors shows this behavior again, the other one works fine. After that happens I've updated zigbee2mqtt to 1.9.0 and removed the battery from the faulty sensor - without success. I do not use repeaters, the distance is maximum 5 meters. Link quality is reported by the sensors as 73 and 55. I'll sniff it in the next few days and report it here.
See you..

kluszczyn, 2020-01-21:
I'm having the same issue. Some of Aqara PIRs working fine, others not reporting occupancy change (but reporting other parameters the same time, and when movement present in front of sensor). I sniffed traffic from that sensor (and it happened that during recording it reported once occupancy). As this was my first time with wireshark, can someone please advise how (and in what format) export data for sharing here (I would like to keep security key private)?

essera, 2020-01-24:
Hi all,
i have the same issue too with my Aquara PIR´s. The description of kluszczyn is very similar to my issue.
6 PIR are working right but 2 sends never an occupancy true. When they discovered a movement they send an occupancy 'false' instead of 'true'. i was on Software level 1.8.0 and made an update to 1.9.0 but the behavoir did´nt changed.
I did an repairing an now they are working as expected. 
But some is strange. When movement is detected the PIR send an occupancy 'false' and e very little time later it send an occupancy 'true'

Event Listing:
2020-01-24 17:45:05.125 MQTT2_DEVICE MQTT2_zigbee_akz illuminance: 7
2020-01-24 17:45:05.125 MQTT2_DEVICE MQTT2_zigbee_akz occupancy: false
2020-01-24 17:45:05.125 MQTT2_DEVICE MQTT2_zigbee_akz linkquality: 0
2020-01-24 17:45:05.125 MQTT2_DEVICE MQTT2_zigbee_akz battery: 86
2020-01-24 17:45:05.125 MQTT2_DEVICE MQTT2_zigbee_akz voltage: 2975
2020-01-24 17:45:05.186 MQTT2_DEVICE MQTT2_DVES_5BAB7D on
2020-01-24 17:45:05.196 DOIF MQTT2_zigbee_akz_DOIF_1 cmd_nr: 1
2020-01-24 17:45:05.196 DOIF MQTT2_zigbee_akz_DOIF_1 cmd: 1
2020-01-24 17:45:05.196 DOIF MQTT2_zigbee_akz_DOIF_1 cmd_event: MQTT2_zigbee_akz
2020-01-24 17:45:05.196 DOIF MQTT2_zigbee_akz_DOIF_1 cmd_1
2020-01-24 17:45:05.199 MQTT2_DEVICE MQTT2_zigbee_akz battery: 86
2020-01-24 17:45:05.199 MQTT2_DEVICE MQTT2_zigbee_akz voltage: 2975
2020-01-24 17:45:05.199 MQTT2_DEVICE MQTT2_zigbee_akz illuminance: 7
2020-01-24 17:45:05.199 MQTT2_DEVICE MQTT2_zigbee_akz occupancy: true
2020-01-24 17:45:05.199 MQTT2_DEVICE MQTT2_zigbee_akz linkquality: 0

Very strange.
Any idea´s ?

e-t-h, 2020-01-24:
Hi essera,
what does repair mean exactly?

essera, 2020-01-24:
Oh sorry - repair means i did an "repairing" with any device with the this issue. now there are working. I cannot say how long it takes til the device have the same issue again. But then i will post it here.

mprice43, 2020-02-11:
I too am having similar issues with the RTCGQ11LM. Movement is triggering messages in the Zigbee2MQTT log but message shows occupancy=false. Also shows illuminance. 

zigbee2mqtt:info  2020-02-11 19:40:32: MQTT publish: topic 'zigbee2mqtt/laundry_ceiling_light', payload '{"state":"OFF","brightness":255,"color_temp":250,"linkquality":107,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_motion_sensor', payload '{"illuminance":9,"linkquality":105,"occupancy":true,"battery":100,"voltage":3045,"illuminance_lux":9}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":107,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":107,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":107,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":107,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:36: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":107,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:40:39: MQTT publish: topic 'zigbee2mqtt/lounge_rm_lamps_pwr_plug', payload '{"state":"OFF","power":0,"voltage":null,"consumption":84.48,"temperature":31,"linkquality":105}'
zigbee2mqtt:info  2020-02-11 19:41:36: MQTT publish: topic 'zigbee2mqtt/office_motion_sensor', payload '{"illuminance":9,"linkquality":105,"occupancy":false,"battery":100,"voltage":3045,"illuminance_lux":9}'
zigbee2mqtt:info  2020-02-11 19:41:37: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"ON","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:43:06: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"OFF","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:43:06: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"OFF","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
zigbee2mqtt:info  2020-02-11 19:43:07: MQTT publish: topic 'zigbee2mqtt/office_ceiling_light', payload '{"state":"OFF","brightness":255,"color_temp":250,"linkquality":105,"color":{"x":0.383,"y":0.382}}'
[... cut ...]
The first time the sensor is triggering correctly and reporting occupancy = true.
1 minute later is then reports occupancy = false which is expected as the timeout is set to 60 seconds in devices.yaml. The 3rd and 4th message contain occupance=false despite obviously detecting 
my movement. The sensor / or the zigbee2mqtt interpretation of the message seems to be haphazard, reporting false more times than true to a detection of motion. 
Has anyone done a network trace (sniff) yet or do I need to buy and second C2531 board to do this?

phplego, 2020-02-15:
I have the same issue for ~1 year. Somtimes sensors stop to send `occupancy: true` and instead of that send `occupancy: false`. Repairing fixes the issue for some time (days or even weeks). 

I tried to narrow the scope by doing these experiments:

* Change disstance to sensor. It does not have affect. No meter is it close to router as 50 cm or 10 meters.
* Change version of zigbee2mqtt to 1.6.0, 1.8.0 and 1.9.0. Same issue appears on any of them in some time.
* Change sensors, buy new ones. It did not help.

Very often the issue disappears just by itself, without repairing. So reproduce the issue isn't an easy task. How can I help the investigation? Thanks

hdo, 2020-02-17:
I noticed the same behaviour with one of my PIR sensors (5 in total). 
However additional to the missing occupancy message i'm also missing the correct illuminance value. I always got illuminance 0 (this happens with 3 of my sensors):

`
zigbee2mqtt:info  2020-02-17 06:09:21: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":18,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:10:56: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":12,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:12:14: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":0,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:13:42: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":0,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:15:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":0,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:19:08: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":6,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:24:37: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":0,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:26:00: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":0,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:27:00: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":15,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:30:30: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":0,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:37:45: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":0,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:38:47: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":15,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:39:59: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":15,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:42:08: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":0,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:44:10: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":42,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:45:15: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":0,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:47:28: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":72,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:48:35: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":72,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:49:36: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":0,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 06:52:42: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":72,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 07:15:51: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":78,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 07:19:50: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":9,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 07:21:52: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":18,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 07:25:57: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":3,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 07:28:22: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":12,"occupancy":false,"illuminance_lux":0}'
[... cut ...]
zigbee2mqtt:info  2020-02-17 08:15:01: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":9,"occupancy":false,"illuminance_lux":0}'
zigbee2mqtt:info  2020-02-17 08:15:01: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":3,"linkquality":9,"occupancy":false,"illuminance_lux":3}'
zigbee2mqtt:info  2020-02-17 08:16:46: MQTT publish: topic 'zigbee2mqtt/0x00158d0001e50d78', payload '{"battery":100,"voltage":3175,"illuminance":0,"linkquality":12,"occupancy":false,"illuminance_lux":0}'`

essera, 2020-02-17:
Hi all,

i promised in my last post to give you an update if the issue happens again - it happens again :-(

It happend to my test PIR on my desk and one in my cellar.

Meanwhile i tried the same Things to get the PIR back to work.

1. PIR is not working so i tried
- Reinstalling the Batterie - doesn´t work
- install a new Batterie - doesn´t work
- Rebooting zigbee2mqtt - doesn´t work
- updating from earlier Version of zigbee2mqtt - doesn´t work
- RePairing the PIR works for a while - 

One PIR stops working only one time - after ReParing , he is working since 5 weeks.
One PIR works after rePairing only a few day´s , after a while he started working with no action from my side.
In the time he was not working he Shows occupany: false and illuminance 0 (same as hdo postet).

i was on Holiday for a week and in that time all PIR´s worked  OK, now.

@Koenkk 
do you have any idea´s what we can do to solve the issue or help you to fix it ?

Koenkk, 2020-02-18:
@essera check https://github.com/Koenkk/zigbee2mqtt/issues/2274#issuecomment-557133101

phplego, 2020-02-18:
> To further debug this issue, could you sniff the traffic while the sensor should detect occupancy when connected with zigbee2mqtt (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)?

I finally started to sniff, and it works!! :)

Does anyone know how to identify messages from the particular device? Thanks

Koenkk, 2020-02-18:
@phplego the cluster should be `Occupancy sensing`

phplego, 2020-02-18:
[image]

But zigbee2mqtt sends

```
2020-02-18T13:49:18.075580141Z zigbee2mqtt:info  2020-02-18 13:49:18: MQTT publish: topic 'zigbee2mqtt/closet_ms', payload '{"illuminance":0,"linkquality":168,"occupancy":false,"battery":100,"voltage":3025}'
```
@Koenkk

Koenkk, 2020-02-18:
@phplego very interesting, this indicates that the sensor itself is not the problem. Therefore we need to look further into the whole zigbee2mqtt/zigbee-herdsman/adapter stack. While this happens, can you capture the logging when running zigbee2mqtt with `DEBUG=zigbee-herdsman* npm start` (https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging)

phplego, 2020-02-18:
> @phplego very interesting, this indicates that the sensor itself is not the problem. Therefore we need to look further into the whole zigbee2mqtt/zigbee-herdsman/adapter stack. While this happens, [...]

[image]

The only thing, that appears in the log is that ^
Looks like it reports only illumination, right?

phplego, 2020-02-18:
For comparision, this is log records in case, when occupancy reports properly:
[image]

phplego, 2020-02-18:
I have one possible idea to ivestigate. I noticed, that in the database the record, related to this sensor is much shorter, than others:

```json
{ 
   "id":11,
   "type":"EndDevice",
   "ieeeAddr":"0x00158d000302f85e",
   "nwkAddr":18389,
   "manufId":4151,
   "manufName":"LUMI",
   "powerSource":"Battery",
   "modelId":"lumi.sensor_motion.aq2",
   "epList":[ 
      1
   ],
   "endpoints":{ 
      "1":{ 
         "epId":1,
         "inClusterList":[ 

         ],
         "outClusterList":[ 

         ],
         "clusters":{ 
            "msIlluminanceMeasurement":{ 
               "attributes":{ 
                  "measuredValue":0
               }
            },
            "genBasic":{ 
               "attributes":{ 
                  "65281":{ 
                     "1":3025,
                     "3":30,
                     "4":12712,
                     "5":22,
                     "6":[ 
                        0,
                        3
                     ],
                     "10":7656,
                     "11":7,
                     "100":0
                  },
                  "modelId":"lumi.sensor_motion.aq2",
                  "appVersion":5
               }
            },
            "msOccupancySensing":{ 
               "attributes":{ 
                  "occupancy":1
               }
            }
         },
         "binds":[ 

         ]
      }
   },
   "appVersion":5,
   "stackVersion":2,
   "hwVersion":1,
   "dateCode":"20170627",
   "swBuildId":"3000-0001",
   "zclVersion":1,
   "interviewCompleted":true,
   "meta":{ 

   },
   "lastSeen":1582035024556
}
```

In the same time, the record for working sensor is a bit different:
```json
{ 
   "id":20,
   "type":"EndDevice",
   "ieeeAddr":"0x00158d000313a331",
   "nwkAddr":53568,
   "manufId":4151,
   "manufName":"LUMI",
   "powerSource":"Battery",
   "modelId":"lumi.sensor_motion.aq2",
   "epList":[ 
      1
   ],
   "endpoints":{ 
      "1":{ 
         "profId":260,
         "epId":1,
         "devId":263,
         "inClusterList":[ 
            0,
            65535,
            1030,
            1024,
            1280,
            1,
            3
         ],
         "outClusterList":[ 
            0,
            25
         ],
         "clusters":{ 
            "genBasic":{ 
               "dir":{ 
                  "value":3
               },
               "attributes":{ 
                  "65281":{ 
                     "1":3045,
                     "3":32,
                     "4":5032,
                     "5":69,
                     "6":[ 
                        0,
                        1
                     ],
                     "10":50856,
                     "11":6,
                     "100":0
                  }
               }
            },
            "genPowerCfg":{ 
               "dir":{ 
                  "value":1
               },
               "attributes":{ 

               }
            },
            "genIdentify":{ 
               "dir":{ 
                  "value":1
               },
               "attributes":{ 

               }
            },
            "genOta":{ 
               "dir":{ 
                  "value":2
               },
               "attributes":{ 

               }
            },
            "msIlluminanceMeasurement":{ 
               "dir":{ 
                  "value":1
               },
               "attributes":{ 
                  "measuredValue":46
               }
            },
            "msOccupancySensing":{ 
               "dir":{ 
                  "value":1
               },
               "attributes":{ 
                  "occupancy":1
               }
            },
            "ssIasZone":{ 
               "dir":{ 
                  "value":1
               },
               "attributes":{ 

               }
            },
            "manuSpecificCluster":{ 
               "dir":{ 
                  "value":1
               },
               "attributes":{ 

               }
            }
         },
         "binds":[ 

         ]
      }
   },
   "appVersion":5,
   "stackVersion":2,
   "hwVersion":1,
   "dateCode":"20170627",
   "swBuildId":"3000-0001",
   "zclVersion":1,
   "meta":{ 

   },
   "lastSeen":1582037125743
}
```

Koenkk, 2020-02-20:
@phplego the incomplete database entry is probably not related, do you see anything more when the occupancy is send (checked via sniffing) and the debug output in zigbee2mqtt?

phplego, 2020-02-21:
@Koenkk 

[image]

I tried to compare bad packet sequence (when I get `occupancy:false`) with good sequence (when `occupancy:false`). They are pretty the same, have no difference in the packet content. Only order of messages may differ (illuminance before occupancy and vise versa).

The only thing that I have noticed is that "bad sequence" happens only after some long time of silence. Lets say I did not touch sensor for several hours. Then, after that peace, first time I entered in the sensor field  may become "bad". After "bad" heppened then next time always working good

> do you see anything more when the occupancy is send (checked via sniffing) and the debug output in zigbee2mqtt?

Could you tell what we are looking for? What attributes of debug log should I pay attention? 
Thank you!

And, by the way, If you have any ideas what may went wrong (in theherdsman, usb dongle firmware and etc), please share, I will try to do my best to prove/disprove that hypothesises. For example I can edit JS code and put debugging "printlns" to interest places. Thanks

phplego, 2020-02-21:
I have noticed that in failed packet there is different Destanation
[image]

Success packet:
[image]

Koenkk, 2020-02-21:
@phplego good finding, it means that the sensor will send the message via a router, the router is responsible for sending it to the coordinator. What kind of device is this? Search for the `50856` networkaddress in your database.db.

phplego, 2020-02-21:
@Koenkk 
> What kind of device is this?

It is my *Aqara double key wired wall switch* (with neutral line) https://www.zigbee2mqtt.io/devices/QBKG12LM.html

phplego, 2020-02-22:
@Koenkk I have 8 routers in my house (aqara wall switches) and I think some of them may not work properly. Is it possible to disable zigbee device to operate as router? Or maybe you know some another way to debug the network?
[image]
Thank you.

Koenkk, 2020-02-23:
@phplego it's not possible to disable the routing functionality on a device.

Can you try pairing the RTCGQ11LM close to the coordinator (this should make the coordinator the parent) and then move the RTCGQ11LM to the desired location? Does it keep working?

bongiozzo, 2020-03-22:
Hello!

Do you have some updates on this problem?

It seems I have the same problem now.
I have 3 motion sensors 
- one Xiaomi RTCGQ01LM
0x00158d0002b48e80_occupancy in hallway
- two Aqara Xiaomi RTCGQ11LM 
0x00158d0004466abd_occupancy in workroom
0x00158d000224ba4a_occupancy in kitchen

They all worked flawlessly for 1-2 month after migrating to zigbee2mqtt 1.10, but yesterday for some period of time (like 2 hours) they mostly don't send occupancy true.

I went from Workroom to Hallway. Workroom sends occupamcy True, but Hallway sends signal as well but occupancy False

in logs 

Mar 22 15:50:35 pi4 npm[507]: #033[32mzigbee2mqtt:info #033[39m 2020-03-22 15:50:35: MQTT publish: topic 'zigbee2mqtt/0x00158d0004466abd', payload '{"battery":100,"voltage":3015,"illuminance":8,"illuminance_lux":8,"linkquality":47,"occupancy":false}'
Mar 22 15:50:35 pi4 npm[507]: #033[32mzigbee2mqtt:info #033[39m 2020-03-22 15:50:35: MQTT publish: topic 'zigbee2mqtt/0x00158d0004466abd', payload '{"battery":100,"voltage":3015,"illuminance":8,"illuminance_lux":8,"linkquality":47,"occupancy":true}'
Mar 22 15:50:36 pi4 hass[645]: #033[32m2020-03-22 15:50:36 INFO (MainThread) [homeassistant.components.automation] Executing Workroom Motion#033[0m
Mar 22 15:50:36 pi4 hass[645]: #033[32m2020-03-22 15:50:36 INFO (MainThread) [homeassistant.helpers.script] Script Workroom Motion: Running script#033[0m
Mar 22 15:50:36 pi4 hass[645]: #033[32m2020-03-22 15:50:36 INFO (MainThread) [homeassistant.helpers.script] Script Workroom Motion: Executing step call service#033[0m
Mar 22 15:50:39 pi4 npm[507]: #033[32mzigbee2mqtt:info #033[39m 2020-03-22 15:50:39: MQTT publish: topic 'zigbee2mqtt/0x00158d0002b48e80', payload '{"battery":100,"voltage":3025,"illuminance":0,"illuminance_lux":0,"linkquality":39,"occupancy":false}'

I've updated to zigbee2mqtt 1.12 - and now sensors seems to work Ok. Will see...

jcobreros, 2020-04-03:
I'm having the exact same problem with 2 out of 3 RTCGQ11LM.
I'm using a CC2531 and four OSRAM Lightbulbs as routers.

tpftpf, 2020-04-04:
This problem still exists, even after update to zigbee2mqtt 1.12

jcobreros, 2020-04-04:
Interestingly, the problem solved itself after a couple long button presses and some time for the network to settle (For 5 minutes I was getting data from the RTCGQ11LM every second. It looked like a backlog of data).

madjam002, 2020-04-05:
I was hoping to do some debugging today as I ordered a programmer to flash my zigbee stick with the sniffer firmware but I received the wrong item from the seller :(

I don't know if this information is helpful, TLDR:

- 6 motion sensors stopped working pretty much completely after upgrading from unknown version of dev branch (around May 2019) to 1.11 and subsequently 1.12.
- When going into a room, messages get sent but with occupancy: false, which makes me think it's not a battery issue and instead a software issue
- All 6 sensors are routed through a wall switch with neutral wire (QBKG12LM)
- Repairing 3 of them close to coordinator fixed the issue, they started transmitting occupancy: true reliably
- Repairing 1 of them further away which ended up still going through the wall switch didn't fix the issue, it started rapidly sending occupancy: false.
- Repairing the same one but closer fixed the issue, and the remaining sensors.

Bit more detail:

- 6 motion sensors (RTCGQ11LM), 1 wall switch with neutral wire (QBKG12LM), and 1 door/window sensor (never had issues with this), and 2 cubes (no issues either).
- I was running a `dev` branch version of Zigbee2mqtt, unfortunately I can't remember what version as I didn't backup, I cloned it around mid-May 2019. My motion sensors were working fairly well.
- Upgraded to Zigbee2mqtt 1.11 and they pretty much completely stopped working. occupancy: true messages are being sent but sometimes with a 5 minute delay of entering a room and having since left the room so there is no motion. Most of the time I don't get occupancy: true messages at all. I am now Zigbee2mqtt 1.12 and the problem still persists. What's weird is when I go into a room, it does send a message straight away but with occupancy: false, so there is definitely something being triggered.

Generating a network map shows that all of the occupancy sensors go through the wall switch. Some of them have lines going to both the wall switch AND the coordinator, not sure what this means, but still those sensors are unreliable.

I am in the process of repairing my sensors to see if it makes any difference and I am monitoring the logs.

- Repaired 2 sensors close to the coordinator (before repairing they were erroneously sending occupancy: false). As soon as they repaired, occupancy: true was sent and they seem to be working fine. Generating a network map shows that they are going direct to the coordinator, not the wall switch.

- Repaired 1 sensor further away from the coordinator. Also erroneously sending occupancy: false before repairing. After repairing, it is rapidly sending messages but with occupancy still `false` even though I am in the room and moving around. Generating a network map shows that this sensor is going to the wall switch, _not_ the coordinator. Repairing but closer to the coordinator fixed the issue.

Because of this, I'm fairly certain the issue lies with the wall switch and something to do with the way it is acting as a router/relay (whatever it is called!).
This behaviour is in line with what was experienced further up this thread https://github.com/Koenkk/zigbee2mqtt/issues/2274#issuecomment-590104511 @Koenkk @phplego. Unfortunately I don't have the device to flash the firmware to debug further, I have contacted the seller to ship one but this will take a couple of months to arrive :(

I have had my sensors for around 8 months and they are reporting 91-100% battery but because the messages are still being sent on motion, just with the incorrect occupancy value, it makes me think that it's a software issue? When the sensor is talking direct to the coordinator it appears to be working absolutely fine. I will update if I have any more findings.

Koenkk, 2020-04-06:
> When going into a room, messages get sent but with occupancy: false, which makes me think it's not a battery issue and instead a software issue

Could you provide the zigbee2mqtt debug logging of this?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

bongiozzo, 2020-04-08:
Hi!

I'd like to help
Added to configuration.yaml
advanced:
  log_level: debug

And I've got some question because this problem is very very annoying when house light doesn't work as used to be :))
We noticed that after upgrade and fresh install system works Ok for some time.

Could you provide some example of flush script which I can add to cron or systemctl to return the system to initial state every day?

bongiozzo, 2020-04-08:
Logfile Initialization:

```
info  2020-04-08 08:19:23: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-04-08.08-19-22' filename: log.txt
debug 2020-04-08 08:19:23: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-03-22.16-25-35'
debug 2020-04-08 08:19:23: Loaded state from file /opt/zigbee2mqtt/data/state.json
info  2020-04-08 08:19:23: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
info  2020-04-08 08:19:23: Starting zigbee-herdsman...
debug 2020-04-08 08:19:23: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"
networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigb
ee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2020-04-08 08:19:25: zigbee-herdsman started
info  2020-04-08 08:19:25: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revisi
on":20190608}}'
debug 2020-04-08 08:19:25: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2020-04-08 08:19:25: Currently 11 devices are joined:
info  2020-04-08 08:19:25: 0x00158d0002b8d9ea (0x00158d0002b8d9ea): MFKZQ01LM - Xiaomi Mi/Aqara smart home cube (EndDevice)
info  2020-04-08 08:19:25: 0x00158d0004466abd (0x00158d0004466abd): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2020-04-08 08:19:25: 0x00158d0002b48e80 (0x00158d0002b48e80): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2020-04-08 08:19:25: 0x00158d00012414a2 (0x00158d00012414a2): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
info  2020-04-08 08:19:25: 0x00158d00042c00e8 (0x00158d00042c00e8): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2020-04-08 08:19:25: 0x00158d00027b3ba7 (0x00158d00027b3ba7): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
info  2020-04-08 08:19:25: 0x00158d00044c33eb (0x00158d00044c33eb): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
info  2020-04-08 08:19:25: 0x00158d000224ba4a (0x00158d000224ba4a): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
info  2020-04-08 08:19:25: 0x00158d00020f55fd (0x00158d00020f55fd): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
info  2020-04-08 08:19:25: 0x00158d0002728a26 (0x00158d0002728a26): ZNMS12LM - Xiaomi Aqara S2 lock (EndDevice)
info  2020-04-08 08:19:25: 0x00158d00042d8a97 (0x00158d00042d8a97): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
info  2020-04-08 08:19:25: Zigbee: disabling joining new devices.
info  2020-04-08 08:19:25: Connecting to MQTT server at mqtt://192.168.1.6
info  2020-04-08 08:19:25: Connected to MQTT server
info  2020-04-08 08:19:25: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

```

And just after restart I went to sensor in my workroom
```
info  2020-04-08 08:19:25: MQTT publish: topic 'homeassistant/sensor/0x00158d00042d8a97/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d00042d8a97","json_attributes_topic":"zigbee2mqtt/0x00158d00042d8a97","name":"0x00158d00042d8a97_linkquality","unique_id":"0x00158d00042d8a97_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00042d8a97"],"name":"0x00158d00042d8a97","sw_version":"Zigbee2mqtt 1.12.0","model":"Aqara water leak sensor (SJCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
debug 2020-04-08 08:19:41: Received Zigbee message from '0x00158d00012414a2', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2020-04-08 08:19:41: No converter available for 'ZNCZ02LM' with cluster 'genTime' and type 'read' and data '["time"]'
debug 2020-04-08 08:20:19: Received Zigbee message from '0x00158d0004466abd', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
info  2020-04-08 08:20:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0004466abd', payload '{"battery":100,"voltage":3015,"illuminance":0,"illuminance_lux":0,"linkquality":89,"occupancy":false}'

```

As you can see event was fired but occupancy is false.

Koenkk, 2020-04-08:
@bongiozzo in your log I can see that the sensor only reports illuminance and not occupancy. At what time did you walk into the room, 08:19:41 or 08:20:19?

bongiozzo, 2020-04-08:
It's core problem that sensor sends illuminance and keeps occupancy false.
Of course I should be noticed by sensor 08:19:41 or 08:20:19 - I was right ahead of it.

In normal conditions sensor doesn't sent illuminance changes (AFAIK), only when motion was detected. 
After 1 (or 2?) minutes in case occupancy wasn't detected anymore it should send "occupancy false" state (For turning lights off).

But here we see only illuminance and occupancy false events.

essera, 2020-04-08:
Hi,

@bongiozzo and @Koenkk .
The problem bongiozzo describe is the problem i have. (I wrote a post some weeks ago). 
Meanwhile:
I added an entry in the configuration.yaml  (there was a post from koenkk)
`advanced:
  log_level: debug
  **cache_state: false**
`
The Parameter "cache_state: false" reduce the problem for me but it wasn´t gone.
In the most cases the motions sensor doesn´t switch on/off the light. I checked the log  - there was the wrong Event in the log ( occupancy: false instead of occupancy: true)

I think there is still something wrong there…

Koenkk, 2020-04-08:
@bongiozzo the problem is that it doesn't report occupancy, only illuminance. In order to further investigate it we need to sniff the traffic when trigger the sensor and no occupancy is send. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

bongiozzo, 2020-04-09:
> I was hoping to do some debugging today as I ordered a programmer to flash my zigbee stick with the sniffer firmware but I received the wrong item from the seller :(
> 

As I understood i have to have additional stick for sniffing. Doesn't have it...

@koenkk am I right?

bongiozzo, 2020-04-09:
> cache_state: false

  # Optional: state caching, MQTT message payload will contain all attributes, not only changed ones.
  # Has to be true when integrating via Home Assistant (default: true)
  cache_state: true

I use it with Home Assistant.
Should I try it with false anyway?

essera, 2020-04-09:
@bongiozzo 
1. Yes: you need an additional stick.
2. I have no Home Assistant so i think you must not change it.

Ton1965, 2020-04-09:
I used to have a similar issue for about a month. I have several Xiaomi occupancy sensors and a door sensor. Some time ago they became unreliable and some messages were just lost somewhere. I use CC1352P2 board for coordinator and a couple of cc2531 as routers. Faulty sensors connected to the coordinator through the router. Usually the latest dev version of z2m was used.

I tried to do many different things like log analysis, trying to re-pair, restarting the routers, downgrading z2m but nothing really helped. Finally I lost my patience and re-flashed the coordinator with the latest firmware from the dev branch (previously it used the firmware from mid-October). Since then the network has been very stable (for about three weeks already), I did not notice any single fault.

I do not know what was the exact reason: firmware upgrade or cleaning the coordinator flash memory, or probably something else.

[79 more replies]

closed 2020-09-25 by Koenkk