Koenkk/zigbee2mqtt #2276: Innr SP120 warn message

andreasbrett, 2019-11-04:
I have 2 Innr SP120 power outlets but only one of them is triggering the following warning:

`No converter available for 'SP 120' with cluster 'genLevelCtrl' and type 'attributeReport' and data '{"currentLevel":254}'`

Looks to me as if the pairing did not complete successfully. Is there something I can check within the device database?

Koenkk, 2019-11-05:
Besides this warning does it work as expected?

andreasbrett, 2019-11-05:
Yes. Everything works. I can switch them on and off and get measurement readings regularly.

Koenkk, 2019-11-05:
ok, I guess this message can be ignored (it doesn't make sense for this device). Warning won't be shown anymore in next zigbee2mqtt release. thanks!

closed 2019-11-05 by Koenkk